terjeio/grblHAL #1: GRBL Panel

1Metalguru, 2018-10-06:
Hi Terje:

Since Garret is no longer doing updates to GRBL Panel, and you have worked with him recently, I was wondering if you are interested in taking it over from him.  Note that I am simply a user, and have no affiliation with the program, nor am I a programmer.  However, I do use it, and like it, and supply it with machines that I sell (I am a custom CNC builder).

GRBL panel is a very nice little program, but it could benefit from some more work, and since Gerrit is essentially retired the community needs someone to take up the ball and run with it.

Interested?  If not, know anyone who is?

1Metalguru

terjeio, 2018-10-07:
@1Metalguru 

Well, I am not well versed in VB.Net but perhaps enough to fix bugs and undertake minor changes. I agree that GRBLPanel is a nice program, and I mainly use it to verify that the changes I make to my Grbl port does not break the communication protocol. Other senders I have tested are generally not as well behaved.

What kind of features do you want to add/change? As I am currently developing three different senders for my own use I have put a lot of the code (written in C#) in a common library, it just might be possible some could be utilized in GRBLPanel as well.

 I do not know of anybody else that could be interested in taking over that has the capacity to do so.

Terje

1Metalguru, 2018-10-07:
Hi Terje:

Just a thought, since you seem to have been conversing with Gerrit for some
time.  I'm just sad to see this program support end.

I would like to see a few things added, including tool change capability,
perhaps a visualizer, add camera support, add game pendant support, change
the way some things work like the Homing to make it configurable as to what
quadrant you home to, etc.

Thanks for getting back to me.

I will have a look at some of your G-code sender stuff and see if it meets
my needs.

Regards,
Darrell Norquay - Calgary, Canada
3D Tech Works - 3DTechWorks.ca
403 701 2005 - ndarrell@telus.net

terjeio, 2018-10-08:
@1Metalguru 

I have been looking for a open source visualizer myself but no luck so far, but then this is not on the top of my wishlist - and surely not something that I will try to code in VB.NET.

Camera support is perhaps the easiest bit, I have a basic version running in my sender already.

Game pendant - don't know if any changes are required for that if the keys can be configured for exixting GRBLPanel shortcuts (if there are any - I have not checked).

Homing is configurable in Grbl, $23? There are a few choices for that in the config.h file too but requires a recompile for the Arduino master branch. I have moved quite a few of the options that requires a recompile to the user settings in my port, among them homing behaviour.

Manual tool change - a few days ago I managed to implement that in my Grbl port (cooperating with  the sender), basically send M6 to pause the program, use the cursor keys for jogging, zero or set the Z position and restart the job. This required a Grbl communication protocol change so it is not compatible (and never will be) with the Arduino controllers or existing senders without modification. Easy to do in GRBLPanel? I do not know. ATC is on my todo list and work is started, I (badly) want one for my homemade PCB router - however this is restricted to 1/8" shank bits only.

The sender applications I am currently working on should work with Grbl 1.1 on Arduinos, but are made to work with my ARM port and new features added to that - to satisfy my needs for now. There still are rough edeges to be ironed out so for now I am not ready to publish any code (except the controls library).

My belief is that the 8-bit Atmel processors and the Arduino framework is now a dead end for Grbl. I did my port due to lack of resources in those processors, but in such a way that the core Grbl code is separated from the hardware 100% - that is why you wil see several "drivers" in this repository.

Well, this became a long comment, sorry about that. If you want to try my sender to see what it looks like as it is now I can provide a time limited download link. I do not want it let loose on the internet so you need to keep it for yourself if you so choose.

Regards,
Terje

closed 2018-12-02 by terjeio