terjeio/grblHAL #2: Will GrblHAL have any further development plans?

viewsat, 2018-12-09:
Will GrblHAL have any further development plans?
For example, the 4th Axis or higher pulse...etc.

terjeio, 2018-12-09:
For the common core I am thinking about splitting the gcode parser and executor (gc_execute_block) in two in order to be able add tool diameter compensation and maybe also drag knife support. Perhaps by tokenize the gcode? It is just an idea,  I do not know if it is viable. At least many of the ARM processors has ample RAM for giving it a try... 

The core can already be configured for up to 6 axes, the drivers for MSP432E401Y and TM4C129x has code to support that. The other drivers needs relatively small modifications to support more than three axes. but be aware that not all processors/target boards has enough pins for more than three. I have not run any machines with more than three so it has to be regarded as untestet.

As for higher pulse I assume you mean step pulse rate. I have not done any actual testing but I assume all should go higher than the 30 kHz which is max for the 8-bit arduino version. An indication for what is a possible upper limit may be a measurement of how muct time is spent in the stepper_driver_interrupt_handler function. For the TM4C1294 (running at 120MHz) I measured around 2uS - this with compiler optimizations off (IIRC). So perhaps the upper limit for that processor is around 200 kHz when CPU time for other activities is factored in. IMO in addition to the CPU frequency the availability of FPU may also play a role in the max step rate achievable.  

I do not plan to spend much time maintaining/updating the drivers for the community. Reasonable pull requests will be accepted though so feel free to contribute.

I am using some of the drivers myself (TM4C and MSP432) and for those I have work in the pipeline, some of which may be published. The two most important for me is spindle synced motion for my lathe and ATC support for my mini mill. I have started work on the mechanical part of ATC, but as if now I am unsure if my ideas will work out.

shooter64738, 2019-02-26:
@terjeio - I hate to hijack one of your discussions, feel free to delete this. But I was wondering if you had some time I could bounce some coding ideas (mostly design basics) off of you. I feel like you are better at this than I am and I could use some input on designing a HAL from the ground up.

terjeio, 2019-02-26:
@shooter64738 : I can give it a try but please start a new issue. As you perhaps already know from looking at my code I am quite fond of using function pointers for implementing HALs and state machines, and to keep the HAL implementation (driver code) in separate files from the core funtionality.

shooter64738, 2019-02-26:
Yes, I've seen the function pointers and was the inspiration for the first version of the HAL I created. If you would email me at 6pickles@gmail.com we can discuss it without using issue tracking. I'll send you some examples of what Ive got and you can tell me what problems might come of it.

closed 2020-02-08 by terjeio