terjeio/grblHAL #4: Major rewrite of LPC 176x driver underway

terjeio, 2019-08-07:
I have switched to MCUExpresso IDE v11 and LPCOpen library. Since I only have access to a [OM13085 development board](https://www.nxp.com/design/microcontrollers-developer-resources/lpc-microcontroller-utilities/lpcxpresso-board-for-lpc1769-with-cmsis-dap-probe:OM13085) I am not able to verify code for boards like the Re-ARM or Smoothieboard.  Currently the modified code seems to run as it should via the CMSIS-DAP VCOM port connected to UART0, I have not yet tested comms via USB connected directly to the MCU.

Testing this far has been outside a machine using a oscilloscope to check signalling.

Please note that Re-ARM and Smoothieboard has broken out the limit signals to pins that are not interrupt capable.  If hard limits are to be enabled this has then to be done by polling or perhaps by some extra hardware. Polling could be done by attaching code to the SysTick handler.

I have a Re-ARM board on order targeted for my 3D-printer (running Marlin), I will be able to test grblHAL on that. Input on pin-mapping for this board will be appreciated. If anyone is interested in running grblHAL on the OM1385 board suggestions for pin mappings for that will be welcomed too.

HuubBuis, 2020-05-21:
> If hard limits are to be enabled this has then to be done by polling or perhaps by some extra hardware. Polling could be done by attaching code to the SysTick handler.

I have a simular problem on the Atmega 2560 RAMPS board. I will fix that (some day) by using a polling routine that is triggered by an overflow counter (C) on Timer5. The overflow counter will run at a much lower rate than the Mega 2560 system tic, as slow as possible, probably 1 ms.

terjeio, 2020-05-22:
I have set SysTick timeout to 1ms for all ARM drivers.

The driver has been updated for/checked with a Re-ARM board - serial over USB is working.
I soldered wires on the MCU pins in order to break out the JTAG interface to a pin header (glued to the top of the SD-card socket), this in order to be able to use the debugger.

closed 2020-09-04 by terjeio