terjeio/grblHAL #5: Wall Plotter

CCERocks, 2019-10-30:
Hi, i wonder if this firmware already tested with Wall Plotter setup using ESP32?

any demo that i can learn from..

terjeio, 2019-10-30:
Not tested yet, you can be the first to do so? It is on my todo list to make and test both a wall plotter and a Maslow CNC (similar kinematics but different driver)...

What kind of demo?

CCERocks, 2019-10-30:
owh ok. i do have NodeMCU v3. kinda try to search for demo/sample or guide to setup Wall Plotter with your firmware.

terjeio, 2019-10-30:
I guess it is high time I start a Wiki with more details...

Do you have the ESP IDF up and running? Will this be the your first CNC project? ... 

I can try to guide you via a dialogue here if you want, could be nice input for what to put in a Wiki.

CCERocks, 2019-10-30:
i already have diy Makelangelo V5 ... & i create the unofficial Cartesian CNC base on Makelangelo firmware too.. 

i want to try new Wall Plotter base on ESP32

terjeio, 2019-10-30:
Sounds great! I need to have a look at some of the google hits for Makelangelo.

I just started some pages on the [GrblHAL Wiki](https://github.com/terjeio/grblHAL/wiki) - not much yet but a start. However, It would be nice to know specifically what information you are looking for, then I could try to provide it.

Perhaps the most basic is that you have to uncomment the following line in [config.h](https://github.com/terjeio/grblHAL/blob/master/GRBL/config.h):

`
//#define WALL_PLOTTER // Default disabled. Uncomment to enable.
`

CCERocks, 2019-10-30:
anyway do you knew that BDring inprocess to unviel his Wall Plotter base on ESP32 ? 

https://github.com/luc-github/Marlin/issues/9#issuecomment-508148547 

it looks great too!

terjeio, 2019-10-30:
Did not know about it, only a [pull request](https://github.com/bdring/Grbl_Esp32/pull/94) from some time back that I see now has been recently updated. The main difference from my implementation is the HAL, my implementation has a well defined interface for separating the hardware driver(s) from the core grbl code.

Go for the solution that suits you best :-)

CCERocks, 2019-10-30:
well.. i would like to experiment few settings.. im not a programmer but then i like to try stuff..

i will try to build Wall Plotter base on your code. 

thanks for the sharing.

ithinkido, 2019-11-10:
Could you please tell me a bit more about how the homing routine and setup works on your wall plotter implementation.

terjeio, 2019-11-11:
As I have not tested the code I simply do not know if homing works...

The [wiki](https://github.com/terjeio/grblHAL/wiki/Additional-or-extended-settings) has information on new and changed settings in grblHAL. $44-$45 can be used to set which axes to home and in which sequence. Additionally $22 has to be set (to 1) to enable homing. Default values for the settings can be found in [defaults.h](https://github.com/terjeio/grblHAL/blob/master/GRBL/defaults.h) if you want to set them at compile time.

Maslow CNC, which employs the same kinematics, saves the current position continuously to non-volatile storage. This to manually set the home position once and not perform homing for subsequent sessions. As I understand it this is done to avoid the complexity of homing... However, doing this when non volatile storage is on flas his not advisable - it is slow and will wear out the flash rather quickly. So not a good idea unless adding a EEPROM to the ESP32.

just-jason, 2019-11-11:
Thanks for the explanation. I have seen some useful stuff in the wiki. It seems to be a manual homing, ie you need to position the pen in a set point before you start the plot.

closed 2021-08-21 by terjeio