terjeio/grblHAL #6: ESP32 and WiFi

just-jason, 2019-10-31:
Hi, 
I have compiled the ESP32 driver version with the changes as recommended:
added #include "esp_attr.h" to config.h
changed #define ISR_CODE to #define ISR_CODE IRAM_ATTR in config.h

Two things I noticed:
1. If I do not enable any "extras" in drivers.h such as #define WIFI_ENABLE      1 // Streaming over WiFi.
then I get a compiling error complaining about driver_settings_t;

2. If I do enable wifi with #define WIFI_ENABLE      1, then the compiling goes great, but I do not seem to have a wifi AP after uploading the code.

Could I ask if you have any suggestions for me here.

terjeio, 2019-10-31:
My bad, I did not exclude all code from compilation as I should when no extras was added. Fix comitted: df76046f74478fb2ec50105fb5d41aacb649eed2.

For WiFi only client mode is implemented, perhaps adding AP mode is relatively straightforward? I will have to look into that.

terjeio, 2019-11-01:
I guess I have the SoftAP option ready, but I am not able to test it until I get access to a ESP32 - a few days after november 10th.

just-jason, 2019-11-02:
Sorry if I have overlooked this some where, but I can not see where to define the client credentials. Could you point me in the right direction.

terjeio, 2019-11-02:
You can find info in the [wiki](https://github.com/terjeio/grblHAL/wiki/Additional-or-extended-settings) - $71 and $72.

Perhaps I should provide an option for setting default values for these in driver.h too?

just-jason, 2019-11-02:
It is a good idea I think, otherwise it kind of seems like you need a wired connection to setup a wireless connection. 
How do you plan to do the AP. Will you be able to configure the wifi settings there?

terjeio, 2019-11-02:
OK, will add more #defines for defaults, for AP mode I have already added

````
#if WIFI_SOFTAP
#define WIFI_AP_IP          "192.168.5.1"
#define WIFI_AP_GW          "192.168.5.1"
#define WIFI_AP_NETMASK     "255.255.255.0"
#endif
````
as defaults in driver.h. I will add defaults for SSID and password, and perhaps country settings too - this for regulatory compliance. When no client is connected (via WiFi, Ethernet and/or Bluetooth) the code will always fall back to serial allowing configuration via this interface.

As a temporary workaround you can add your default values in driver_settings_restore() starting at line 1119 in driver.c - by strcpy as is done for bluetooth.

just-jason, 2019-11-02:
I am not sure if the idea fits your roadmap for this project, but when you mentioned AP setup I had something like this in mind. 
https://github.com/zhouhan0126/WIFIMANAGER-ESP32
Would this be an idea for this issue

terjeio, 2019-11-03:
Ok, could be done I guess as I am pondering adding a generic webserver option that can serve files stored both on flash and SD cards. The grbl config subsystem should also be made available as well as an option to connect to the tcp stream IO. How much of a coding effort it would be to create the backend for this I do not know yet.

It would be nice if the backend could be fairly easily put on top of different http daemons, thus allowing its use for other processors than ESP32. Perhaps I am a bit crazy when it comes to layering code but this is the way I like structuring my code...

terjeio, 2019-11-11:
A quick update:

I have verified that AP mode is working and done some initial work on the http backend. Currently this allows reading grbl settings in JSON format via /settings (for all) or /settings/<n> where <n> is the setting number.  Next step is to write some code to read and change settings via a browser, perhaps by some javascript generating a GET/POST messages? If I am lucky the POST handler I added to the backend earlier will work...

I will have to revisit writing browser code, it is a few years since I did that - relearning required. Or could you possibly help me with that?

just-jason, 2019-11-11:
> A quick update:
> I have verified that AP mode is working and done some initial work on the http backend. Currently this allows reading grbl settings in JSON format via /settings (for all) or [...]
 
Wow, that was fast

> I will have to revisit writing browser code, it is a few years since I did that - relearning required. Or could you possibly help me with that?

Sorry, my coding is not very usable.

There is a nice library called ESPdash for http stuff that is more at my level, I do not know if this is something that you consider using, or if it is up to the job.

terjeio, 2019-11-11:
I do not want to be dependent on Arduino libraries for code that may be shared across different drivers so ESPdash does not fit into how I want to proceed. I have written a web-app using ExtJS earlier - IMO using that, or another javascript framework, is more appropriate. Another benefit of using javascript (in combination with Ajax) is this requires less code in the backend.

The first bit was fairly easy, my use of a function pointer based HAL made it straigthforward to parse a settings report and convert it to JSON before pushing it out over http. The remaining bits will take longer...

terjeio, 2019-11-14:
New version committed. AP and APSTA support added. Web-server extended to allow changing settings, get AP list and connect to AP as well as download files from SD card.

Network $-settings has been extended and moved around a bit - will update documentation soon. Until then refer to [settings.h](https://github.com/terjeio/grblHAL/blob/master/GRBL/settings.h) for new $-setting id's. [driver.h](https://github.com/terjeio/grblHAL/blob/master/drivers/ESP32/driver.h) has got a number of new network related options/default values.

More to come later...

just-jason, 2019-11-15:
I think it is now safe to close this issue. Thanks for the great work to get this set up.

closed 2019-11-15 by just-jason