terjeio/grblHAL #7: idf version

jschoch, 2020-01-31:
which version of the idf are you using?  I can't get it to compile with IDF 3.3.1

jschoch, 2020-01-31:
I was able to compile it.

I am running WSL so there are a few tricks needed.

```

git clone https://github.com/terjeio/grblHAL

schoch@xeon:~/dev/esp/test_comp_ghal$ cd grblHAL/
schoch@xeon:~/dev/esp/test_comp_ghal/grblHAL$ cp -R grbl/* drivers/ESP32/grbl

schoch@xeon:~/dev/esp/test_comp_ghal/grblHAL$ cd drivers/ESP32/grbl/
schoch@xeon:~/dev/esp/test_comp_ghal/grblHAL/drivers/ESP32/grbl$ vi config.h

## add the esp include and ISR change  *** this probably should go to the top of the README

schoch@xeon:~/dev/esp/test_comp_ghal/grblHAL/drivers/ESP32/grbl$ cd ..

## this seems to create the makefile
schoch@xeon:~/dev/esp/test_comp_ghal/grblHAL/drivers/ESP32$ cmake .

schoch@xeon:~/dev/esp/test_comp_ghal/grblHAL/drivers/ESP32$ make -j16 all

## WSL specific stuff

#1. WSL is stuck at 115200 baud for the serial ports
#2. you need to chmod the device to use it
#3. you also need to stty -F /dev/ttyS5 -a

sudo chmod 666 /dev/ttyS*
stty -F /dev/ttyS27 -a

# i also think you can set this in menuconfig but I couldn't find it
export ESPPORT=/devttyS27
export ESPBAUD=115200

make flash

```

For some reason my esp board requires me to hold the boot button down to flash.  I also had to unplug it/replug it to get it to connect. 

Finally, it doesn't seem to reset the serial port correctly.  If i connect first to something like arduino serial monitor and then try to connect with your sender it does not connect.

closed 2020-01-31 by jschoch

terjeio, 2020-01-31:
I am using v3.3 and I have to hold down the boot button too, don't know if the Arduino IDE works differently. Some hints here? https://github.com/espressif/esptool/wiki/ESP32-Boot-Mode-Selection

I have tested a bit with PuTTY as terminal. It seems like it has to do with what status the RTS/CTS and/or DSR/DTR signals is left at by the previous connection? Or by grblHAL? Sometimes connecting with PuTTY resets the ESP, sometimes the sender does. Sometimes I cannot connect or I get no response. Etc. Perhaps related to what is described in the link above? Also GPIO 0, 2 12 and 15 are unusable? And 4 and 5 too? Or perhaps only during reset.

I will need to look into this, it is late here so not today...

The sender does have code to reset the ESP via RTS/CTS on connect but I have not yet made that available via configuration. IMO the ESP should only boot on power on or a hard reset and not interfere with with serial communication at all.

terjeio, 2020-02-01:
Found this in the schematic:
```
Auto program
DTR RTS-->EN IO0
1   1     1  1
0   0     1  1
1   0     0  1
0   1     1  0
```
So what happens on connect is definitely dependent on what state DTR and RTS is left at by any previous connection. The ESP may even enter bootloader mode?

I have tried a few changes to the Sender code without any success so far. I am considering adding wires to DTR and RTS signals for checking what is going on (on my logic analyzer).

terjeio, 2020-02-01:
PuTTY and the Arduino serial monitor is both asserts DTR and RTS (sets them low) on connect. PuTTY even as I specify no handshake. Problem is Microsofts serial implementation, it seeminly deasserts DTR on connect regardless of what I specify for DtrEnable before opening the connection - thus triggering a reset.

So, nothing I can do about it short of finding a different serial library that behaves better (by leaving both signals untouched on connect). IMO not worth the effort...

jschoch, 2020-02-03:
i thought i'd add that cmake makefile generator doesn't include the monitor target.  if you want to run monitor to get traces you can run

```
$IDF_PATH/tools/idf_monitor.py --port /dev/ttyS27 --baud 115200 --make make grbl.elf
```

it seems there should be an easier way to do this but too dense to figure it out.