terjeio/grblHAL #8: pins

jschoch, 2020-02-01:
Wondering what blockers there are to trying to map to bart's v4 pinout?  May make it easier for folks who can't fab a board.

Here's what i've collected so far on the mappings:

[image]

jschoch, 2020-02-01:
the reason i mapped this out is that I want to figure out what pin to map to MODE_SWITCH_PIN.

terjeio, 2020-02-01:
MODE_SWITCH_PIN is what? For switching to/from MPG input?

I've not updated the pin mappings for ESP32 since the early days. When looking into the boot/reset problems earlier I decided mapping files should be used instead of having the mappings in driver.h and I was about to change to barts v4 for no boosterpack before other stuff claimed my time.

For the BoostePack I have added an I2C I/O expander as the ESP32 is on the low end for number of usable GPIOs for grbl. I have to say I am not a big fan of ESP32 for a number of reasons, this included.

Tell me how you want the mappings and I will do my best to fix that.

holla2040, 2020-02-01:
I'm very interested in this ESP32 port as well as PSoC5. I have custom boards for each

PSoC5 https://github.com/holla2040/hyatt
ESP32 https://github.com/holla2040/tetrad

For the next ESP32 and PSoC5 boards, I want to move to grblHAL from Bart's grbl ports. Thanks for all the discussions here.

jschoch, 2020-02-01:
If you don't mind I can try to get the v4 mappings working.

I'm also right now looking at the serialSelect stuff.  I'm assuming you'd want to put all that serial switching in esp32-hal-uart.c  vs driver.c correct?

I've only played with mulitple uarts on STM32.  Hope i can get it working.

terjeio, 2020-02-01:
I on my way with v4 mappings, it requires some changes to driver.c as well as there is no mist, coolant or spindle dir pins defined. Serial switching is best done in serial.c except for the mode switch handler, I've put that in driver.c for the other MCU that supports two ports.

I'll commit a v4 map implementation shortly for you to try/verify.

jschoch, 2020-02-01:
no serial.c, the serial.h pulls in "esp32-hal-uart.h

terjeio, 2020-02-01:
sorry about serial.c - flipping too much back and forth...

I will commit a new driver in a few minutes, it will default to v4 - v3.5 is added too. Board selection except for boosterpack has do be done in driver.h for now - not CMakeLists.txt.

The old default mapping is still in driver.h as it may be in use.

terjeio, 2020-02-01:
Commited - be aware that I have not done any testing.

I have not modified the grbl core to fault M8 and M9 due to cooland/mist pins not beeing available. If it is important to do so I'll do that later.

terjeio, 2020-02-01:
@holla2040 Interesting, I tried to open your ESP32 schematic but got a 404. You are using a I2C I/O expander for some of the signals?

holla2040, 2020-02-01:
v1.0 sch is here, https://github.com/holla2040/tetrad/blob/master/release/v1.0/hexad-sch.pdf 

I'm using a mcp23017 but only for LEDs and external switches. I didn't incorporate it into any grbl code for I/O control. 

Next version will include
* stepper motor driver sockets (first version was quick and dirty for step/servo motors I have)
* FFC connector for matrix orbital EVE2 displays

ps. I have a pick and place machine and puhui reflow oven, will sell these boards somewhere for a low cost.

terjeio, 2020-02-02:
@holla2040 Thanks for the link. Have you tried the display interface with CS and DC signals routed through the I2C chip? If so is there any noticeable delay in screen updates compared to wiring them directly? And what clock frequency for I2C are you using?

With GrblHAL it should be possible to add display code at the driver level, I don't think the core grbl code needs any changes. I started tinkering with that a while ago but did not complete anything useful, this as my external MPG/DRO serves me well.

holla2040, 2020-02-02:
Haven't tried the I2C CS and DC driver code yet, I have to rewrite my EVE2 stm32 library for that. I threw the I/O expander on the board at the last moment along with the display connector. The datasheet show 100K, 400K and 1.7M I2C.  

I've been using these EVE2 displays with a STM32 design, they work great (after you figure how the controller works). 

Running over STM32 SPI, the display can easily update 3 large floating numbers at 20fps. SPI data is really small, a few bytes per frame, because the EVE controller draws everything and screen writes internally.  So I2C even at 400K would incur a slight overhead.

https://github.com/holla2040/EVE2-Arduino_STM32

terjeio, 2020-02-08:
@holla2040 EVE2 looks like great displays, will have too take a closer look at them. I like that they have a high-level API compared to the cheap Chinese variants.

jschoch, 2020-05-11:
It looks like SPINDLEPWMPIN and X_LIMIT_PIN both are assigned GPIO_NUM_17 in bdring_v4_map.h

I believe X_LIMIT_PIN should hold 17 and SPINDLEPWMPIN should be GPIO_NUM_2

Also, i'm working out a board I can easily swap between bart's fork and grblHAL.  It looks like the number of pinouts is much larger now.  I'd like to add the MPG_ENABLE_PIN, MPG_RX_PIN, A_STEP, and A_DIR.  I wonder if I should just maintain my own 4 axis layout of if you want to keep one in the main repo.  It may not be easy to stay in sync with bart's variety of board layouts.

Have you had any luck trying to standardize anything collaborating with Bart?

terjeio, 2020-05-12:
I'll check the pin mapping. Hard to keep track of all when I am not able to test.

> Have you had any luck trying to standardize anything collaborating with Bart?

No, but I have not really tried. I used his new  $-commands for handling SD-card streaming which he since changed to his new [ESP xxxx] format. This is surely something I will not adopt...

Ideally a project for handling standardization should be set up to avoid too many ways for implementing extensions. It is not easy for sender developers to support that...

I have tried to stay faithful to the original specification, trying to expand on that instead of inventing too many breaking changes. E.g. I have enhanced the command parser to allow string input, and I still use $-settings.

I have commented on issues so I guess he is aware of what I am up too. It would be nice if he would use the grblHAL core and driver concept - IMO he could then avoid a bit of the #define problems I understand he is sometimes struggling with... ;-)

jschoch, 2020-05-12:
it looks like the ESP XXXX format is driven by the WEBUI.  It also looks like they are playing with an I2S driver to run level shifters and make more pins available.  That may be worth adopting, I think there will be several boards using this approach.

terjeio, 2020-05-13:
> I believe X_LIMIT_PIN should hold 17 and SPINDLEPWMPIN should be GPIO_NUM_2

Yes, looks like I have made a copy/paste error - will commit a fix later.

IMO the ESP32 is not well suited for CNC use: limited number of pins, suboptimal architecture(?), the problematic meditating guru crashing the system...

"suboptimal architecture" - it should be checked out if swapping in program code from the external flash has any effect on step signals. If all processing is suppressed then that may introduce quite a bit of jitter.

The only plus points are wifi and BT. For networking I prefer cabled ethernet, [Teensy 4.1](https://github.com/terjeio/grblHAL/issues/22) looks interesting in that respect.

Your 4-axis version can be added to the repo, a new board file and some #define enabled code in the current driver? Note that for significant driver differences I would rather add a new driver than making it a #define hell..

jschoch, 2020-05-13:
"suboptimal architecture"  I thought all time sensitive stuff (ISR) should be flagged as IRAM_ATTR or DRAM_ATTR and DRAM_STR for const values to prevent flash access and crashes.  I've had crashes but there is no logging and if the sender also crashes you can't tell what happened.

regarding pins and step generation it looks like they can do 250kHz with the i2s stepper code and it also looks like the fault detection is improved.  here's the thread: https://github.com/bdring/Grbl_Esp32/pull/372

After looking at odaki's repo it looks like the pinouts will diverge quite a lot and the expander based board definitions will start with an offset of 128

`// With the I2S I/O expander enabled, you can specify 128..159 as output pins.`

so it looks doubtful board definitions will be reusable since they are also adding a spindle class to manage different spindle types.

terjeio, 2020-05-13:
> I thought all time sensitive stuff (ISR) should be flagged as IRAM_ATTR ...

Does not help if all processing is suspended when swapping in program code from flash. ESP32 documentation is not clear (or was not? - I have not checked in a while) on this and it should be checked. I have seen comments on the ESP32 forum earlier suggesting that processing _is_ suspended.

I do not know if barts port is now complete in flagging all code that can be run in a ISR context, IIRC there was a few places in the core earlier that was missing the IRAM_ATTR decoration. Hopefully I have not missed them in my port...

One issue I see not adressed by the 32-bit ports I have looked at is the use of double version of math functions. IMO it its kind of silly to use double versions when the target and parameters are single precision (float). Note that the 8-bit Arduino floating point library always uses single precision regardless, so no penalty (or precision gain) using double.

[ARM example](https://e2e.ti.com/support/microcontrollers/other/f/908/t/465956):

sqrtf - single precision:
```

        VMLA.F32  S0, S2, S1            ; [DPU_MERLIN_PIPE] |67| 
$C$DW$186	.dwtag  DW_TAG_TI_branch
	.dwattr $C$DW$186, DW_AT_low_pc(0x00)
	.dwattr $C$DW$186, DW_AT_name("sqrtf")
	.dwattr $C$DW$186, DW_AT_TI_call

        BL        sqrtf                 ; [DPU_V7M3_PIPE] |67| 
        ; CALL OCCURS {sqrtf }           ; [] |67| 
	.dwpsn	file "../GRBL/gcode.c",line 68,column 1,is_stmt,isa 1
$C$DW$187	.dwtag  DW_TAG_TI_branch
	.dwattr $C$DW$187, DW_AT_low_pc(0x00)
	.dwattr $C$DW$187, DW_AT_TI_return

        POP       {A2, A3, A4, PC}      ; [DPU_V7M3_PIPE] 
```

sqrt - double precision, two extra calls - I guess for converting float <> double:

```
        VMLA.F32  S0, S2, S1            ; [DPU_MERLIN_PIPE] |67| 
        VMOV      A1, S0                ; [DPU_MERLIN_PIPE] |67| 
$C$DW$188	.dwtag  DW_TAG_TI_branch
	.dwattr $C$DW$188, DW_AT_low_pc(0x00)
	.dwattr $C$DW$188, DW_AT_name("__aeabi_f2d")
	.dwattr $C$DW$188, DW_AT_TI_call

        BL        __aeabi_f2d           ; [DPU_V7M3_PIPE] |67| 
        ; CALL OCCURS {__aeabi_f2d }     ; [] |67| 
        VMOV      D0, A1,A2             ; [DPU_MERLIN_PIPE] |67| 
$C$DW$189	.dwtag  DW_TAG_TI_branch
	.dwattr $C$DW$189, DW_AT_low_pc(0x00)
	.dwattr $C$DW$189, DW_AT_name("sqrt")
	.dwattr $C$DW$189, DW_AT_TI_call

        BL        sqrt                  ; [DPU_V7M3_PIPE] |67| 
        ; CALL OCCURS {sqrt }            ; [] |67| 
        VMOV      A1,A2, D0             ; [DPU_MERLIN_PIPE] |67| 
$C$DW$190	.dwtag  DW_TAG_TI_branch
	.dwattr $C$DW$190, DW_AT_low_pc(0x00)
	.dwattr $C$DW$190, DW_AT_name("__aeabi_d2f")
	.dwattr $C$DW$190, DW_AT_TI_call

        BL        __aeabi_d2f           ; [DPU_V7M3_PIPE] |67| 
        ; CALL OCCURS {__aeabi_d2f }     ; [] |67| 
        VMOV      S0, A1                ; [DPU_MERLIN_PIPE] |67| 
	.dwpsn	file "../GRBL/gcode.c",line 68,column 1,is_stmt,isa 1
$C$DW$191	.dwtag  DW_TAG_TI_branch
	.dwattr $C$DW$191, DW_AT_low_pc(0x00)
	.dwattr $C$DW$191, DW_AT_TI_return

        POP       {A2, A3, A4, PC}      ; [DPU_V7M3_PIPE] 

```

Sorry for the ranting... ;-)

terjeio, 2020-09-04:
> It also looks like they are playing with an I2S driver to run level shifters and make more pins available. That may be worth adopting, I think there will be several boards using this approach.

Now in the master branch - as of now untested, I will order shift registers later for that.

closed 2021-08-21 by terjeio