terjeio/grblHAL #10: settings reset

jschoch, 2020-02-07:
i sent `$121=400` and the esp32 crashed and all the settings have reverted to default.  reset the settings and no crash so I can't reproduce.  This might be really bad if something really needed to be set though your defaults seem very conservative.  I don't know of a good way to generate a warning that defaults are set but it may be worth considering.

terjeio, 2020-02-07:
Strange, must be a crash during writing settings to flash and ending up with a mismatching checksum. This should result in error 7 beeing reported on the next boot. My test setup is with a BoosterPack, this has an EEPROM that I use for settings, never seen any crashes with that. I will switch to storing settings in flash, programaticallly corrupt a checksum and see what happens.

By the way, did this happen when changing the setting from a terminal window or a sender? If a sender, which one?

terjeio, 2020-02-07:
I can manage to provoke a panic `Cache disabled but cached memory region accessed` if I try to set more than one variable by pasting them to a terminal window. You got the same error?

Most likely this is due to an interrupt occuring during flash erase or write accessing code that is currently paged out. I have pinpointed the crash to happen during erase or write, but not the exact reason. The likely culprit that comes to mind is the uart serial RX interrupt, however the relevant functions are already decorated with the required IRAM_ATTR attribute...

When using the BoosterPack I2C EEPROM for settings I am not able to provoke a crash.

Note: grblHAL caches the settings in memory - they will only be commited to flash on changes, delayed to the next IDLE state beeing processed in the main protocol loop.

jschoch, 2020-02-07:
my mpg was a "man in the middle", and I sent the command through it via an arduino ide serial monotor window so it went through the stm32 to uart1.  i don't believe anything else was happening but my apartment is noisy and when the heat kicks on sometimes signals appear.

terjeio, 2020-02-07:
Thanks for feedback. I've solved the crash I was able to provoke, it was due to the uart interrupt handlers accessing constants in flash. By moving some to DRAM and switching off real time command enqueueing during flash erase/write it no longer crashes. With luck the crash you experienced was due to this too, perhaps likely if your "man in the middle" constantly requests real time status reports?

jschoch, 2020-02-07:
what is a sane update rate for real time status?

terjeio, 2020-02-07:
I am using 200 ms, what a sane lower limit is depends on MCU speed and baud rate - and also on the capabilities of the receiving end...

terjeio, 2020-02-08:
A likely fix has been commited.

closed 2020-02-08 by terjeio