terjeio/grblHAL #11: Inconsistent code style

nythil, 2020-02-07:
Out of curiosity I started looking around the code and noticed inconsistency in indentation (tabs vs spaces) and code style (mostly spacing around for/if/while and functions). This makes some parts of the code [hard to read](https://github.com/terjeio/grblHAL/blob/master/grbl/grbllib.c#L187-L246), possibly scaring away potential contributors to what seems the most inventive successor to grbl :)

terjeio, 2020-02-07:
I guess I have to walk through this again - I recently did. Perhaps my editors are messing up again... 

Line feed for line endings and four spaces indentation is the style I use. I guess need to create a validation script... As for my coding style generally: I try to keep it consistent but quite a bit code is inherited from previous programmers (and in a style that I do not like). I have put some effort into making the style consistent throughout the codebase - but have to admit that it has not been a top priority.

terjeio, 2020-02-08:
Fixed indentation in the grbl core, a couple of drivers still has a few tabs lingering - will commit fix for those later. Made script for verifying code before committing.

Closing this issue for now, reopen if other aspects needs discussion.

closed 2020-02-08 by terjeio

nythil, 2020-02-08:
That's a lot better. Thanks! Are you going to commit the script too? A clang-format script would be useful for contributors too.

terjeio, 2020-02-08:
I do not use clang and as my in-house repository is subversion I made a quick-and-dirty grep one-liner to spit out the offending code lines to a file. It is not perfect but serves its purpose.

[checktabs.zip](https://github.com/terjeio/grblHAL/files/4174749/checktabs.zip)

nythil, 2020-02-09:
I can prepare a .clang-format config for you. Which styles do you prefer in the following cases?
1) `static void debug_out (bool on);` vs `static void debug_out(bool on);`
2) `if(!driver_ok) {` vs `if (!driver_ok) {`

terjeio, 2020-02-11:
I prefer:

`static void debug_out (bool on)` and static `void debug_out (void)` not `static void debug_out ()` as the function name is then delimited by a space and functions taking no parameters are clearly marked as such.

`if(!driver_ok)` as I see no point in adding a space after `if` and similar language specific statements. IMO if a space is to be added it should also be added for function calls...

Oh, and I generally dislike editors that are modifying my code when writing.