terjeio/grblHAL #12: esp32 nvs.c

jschoch, 2020-02-11:
need to add 

```
#include "esp_system.h"
```

with the current commit I get this without it

```
/home/schoch/dev/esp/terj/grblHAL/drivers/ESP32/nvs.c:35:24: error: expected '=', ',', ';', 'asm' or '__attribute__' before 'char'
 static const DRAM_ATTR char ESP_SPACE_CHAR = ' ';
                        ^
/home/schoch/dev/esp/terj/grblHAL/drivers/ESP32/nvs.c:36:24: error: expected '=', ',', ';', 'asm' or '__attribute__' before 'char'
 static const DRAM_ATTR char ESP_DEL_CHAR = 0x7F;
                        ^
/home/schoch/dev/esp/terj/grblHAL/drivers/ESP32/nvs.c:37:24: error: expected '=', ',', ';', 'asm' or '__attribute__' before 'char'
 static const DRAM_ATTR char ESP_CR = ASCII_CR;
                        ^
/home/schoch/dev/esp/terj/grblHAL/drivers/ESP32/nvs.c:38:24: error: expected '=', ',', ';', 'asm' or '__attribute__' before 'char'
 static const DRAM_ATTR char ESP_LF = ASCII_CR;
                        ^
/home/schoch/dev/esp/terj/grblHAL/drivers/ESP32/nvs.c:39:24: error: expected '=', ',', ';', 'asm' or '__attribute__' before 'char'
 static const DRAM_ATTR char ESP_QUESTION_MARK = '?';
                        ^
In file included from /home/schoch/esp/esp-idf/components/spi_flash/include/esp_partition.h:19:0,
                 from /home/schoch/dev/esp/terj/grblHAL/drivers/ESP32/nvs.c:24:
/home/schoch/dev/esp/terj/grblHAL/drivers/ESP32/nvs.c:53:18: error: expected '=', ',', ';', 'asm' or '__attribute__' before '_Bool'
 static IRAM_ATTR bool nvs_enqueue_realtime_command (char c)
                  ^
/home/schoch/dev/esp/terj/grblHAL/drivers/ESP32/nvs.c: In function 'nvsWrite':
/home/schoch/dev/esp/terj/grblHAL/drivers/ESP32/nvs.c:73:43: error: 'nvs_enqueue_realtime_command' undeclared (first use in this function)
     hal.stream.enqueue_realtime_command = nvs_enqueue_realtime_command;
                                           ^
/home/schoch/dev/esp/terj/grblHAL/drivers/ESP32/nvs.c:73:43: note: each undeclared identifier is reported only once for each function it appears in
CMakeFiles/grbl.elf.dir/build.make:125: recipe for target 'CMakeFiles/grbl.elf.dir/nvs.c.obj' failed
make[2]: *** [CMakeFiles/grbl.elf.dir/nvs.c.obj] Error 1
make[2]: *** Waiting for unfinished jobs....
CMakeFiles/Makefile2:396: recipe for target 'CMakeFiles/grbl.elf.dir/all' failed
make[1]: *** [CMakeFiles/grbl.elf.dir/all] Error 2
Makefile:127: recipe for target 'all' failed
make: *** [all] Error 2
```

terjeio, 2020-02-11:
Have you modified grbl/config.h (at the top) per instructions?

Add: `#include "esp_attr.h"`
Change: `#define ISR_CODE` to `#define ISR_CODE IRAM_ATTR`

I see this information is at the bottom at the driver page, I need to move it to the top. Sorry about that.

I have tried to detect if this has been done, but without luck, will change to a simple `#define ESP32` that needs to be commented in, as this can be detected by the preprocessor.

The reason behind this is that it is important to mark all code that can be run in a interrupt context with `IRAM_ATTR` to avoid a large hit to latency if the code needs to be swapped in from flash. Some of the code in the grbl core can be, I use the `#define ISR_CODE` directive to achieve that in a generic fashion.

terjeio, 2020-02-11:
Update commited, should fix your compile problem. I have added a define to CMakeLists.txt so configuration for ESP32 should now be taken care of automatically.

closed 2020-02-11 by terjeio

jschoch, 2020-02-11:
i had done a fresh pull and forgot to make that change, sorry.