terjeio/grblHAL #13: bluetooth MAC

jschoch, 2020-02-11:
might be handy to print out the bluetooth and or wifi mac addrs with a '$' command or on boot.

Also wondering what client you are using to test BT?

jschoch, 2020-02-11:
oh, duh!  i see the sender works great with windows BT!!!!  You should put that in the readme!!!

Also you should put the driver capabilities matrix in the readme!

terjeio, 2020-02-11:
Since BT just adds a virtual serial port any sender or terminal supporting serial can be used.

I will add information to the `$I` command. Outputting such information at boot is not a good alternative for a number of reasons, for ESP32 the main one is that connection over BT or wifi cannot reset the processor as connection will be lost. Another is if in MPG mode a hard or soft reset is IMO usually not wanted. Better then to provide this information on request, and `$I` fits the bill.

Will look into how to add driver capabilities to the driver readme files.

terjeio, 2020-02-11:
Added device and client (if available) MAC to the `$I` report in the latest commit . Will add this information to the About... dialog in my sender later.

jschoch, 2020-02-11:
Thanks.  I did some testing with bluetooth and the MPG function last night and it worked great!!!!  I finally have decently ramped jogging based on the MPG wheel speed.  Have you tested BLE?  I have several machines and it would be nice to share a MPG between them since the MPG hardware likely costs more than the grbl controller setup.

jschoch, 2020-02-11:
another quick thought, it might be nice to have a single place to name the SSID or bluetooth device since you may have more than one.  I can do this if you want help with it.

terjeio, 2020-02-11:
I am not sure what you mean by your latest comment. 

The previous one is about having the MPG communicate over Bluetooth I assume? I have to admint I do not comprehed what all the acronyms _really_ implies, BLE is one of those. I am an old man and slow to pick up new stuff...

terjeio, 2020-02-11:
I do not know if it is related to your latest comment, but I just found out that the sender settings file was an old version with wrong names. Use this instead:

[hal_setting_codes_en_US.txt](https://github.com/terjeio/grblHAL/files/4186930/hal_setting_codes_en_US.txt)

jschoch, 2020-02-11:
The device shows up as "GRBL" in bluetooth which is set in bluetooth.c

`strcpy(driver_settings.bluetooth.device_name, "GRBL");`

the WIFI SSID is set in driver.h

`#define WIFI_AP_SSID        "GRBL"`

I'm suggesting to put the "name" in a single location like driver.h and use the same name for both bluetooth and wifi.

BLE is bluetooth low energy.  I was wondering if you had tested using it yet.  BLE would be nice for a battery powered MPG pendant.

terjeio, 2020-02-11:
Ok, I see that I have not provided defines for the default BT settings in driver.h, will add that.
Note that these can be overridden at run time by new [$-settings](https://github.com/terjeio/grblHAL/wiki/Additional-or-extended-settings) 71 and 72, or via the `Settings: Grbl` tab in my sender.

GrblHAL supports string settings, including the characters used for printable real-time commands if $39=0 (_Enable legacy real-time commands_ is disabled).

I know what BLE acronym means, but not what it takes to implement it. So no, I have not tested it. Do you know if it has to be added to the driver in a separate file or is it a simple modification to the exisiting bluetooth.c?

jschoch, 2020-02-11:
thanks for the pointer on the settings string

from my dabbling BLE is quite a different beast than bt classic.  I would put it in a separate file with a separate config and config option in CMakeLists.

terjeio, 2020-02-11:
Has your dabbling been with ESP32 and do you have some code that can be used as a starting point, or even better, do you want to give it a try yourself?

jschoch, 2020-02-11:
Yes, i've hacked a bit of both on the esp32.  I'll give it a try!!!

closed 2022-01-27 by terjeio