terjeio/grblHAL #14: esp32 spindle sync

jschoch, 2020-02-11:
I wonder if you've had a crack at spindle sync in the esp32 driver?  I may take a crack at it since I finally will get access to my other lathe soon.  Any tips on your implementation?  Did you base your code on the grblMEGA implementations?  

I also recall you are trying to PID control both the motor speed and the Z position, have you tried just focusing on Z positioning and letting the spindle motor do what whatever it wants?

terjeio, 2020-02-11:
I am not so sure I want to add spindle sync to ESP32. My three main reasons are lack of GPIO pins, a FPU that  crashes the system when accessed in an ISR and the overly(?) sensitive guru that is lurking behind the scenes.

My implementation is not derived from any grbl codebase,  I specifically enhanced the HAL API in order to move the spindle sync algorithm to the driver layer. I will stick with the MSP432 MCU for now, I see no good reason for me to spend time implementing this on other MCUs before I am happy with the MSP432 version.

Yes, I have switched to Z-position PID only - the VFD motor allows that. However, I will not spend much time on the code before the new Z ballscrew is in place. It can be noted that the latest test I did was promising, the PID error was within ±0.005 mm when cruising and ±0.01 mm when accelerating/decelerating. What the physical error was I do not know, my current Z axis has 1.5mm of backlash... The nut went on smoothly so maybe not too bad.

jschoch, 2020-02-11:
ouch, that is a lot of backlash!!!!   cutting steel must be a real bear!  I lucked out and picked up 8 ballscrews on ebay for $70 bucks and i'm planning on converting both of my lathe cross slides to ballscrew.  The leadscrews have fairly low backlash but I plan to convert the small C1 size lathe to z ballscrew.

regarding the esp32, i think you can do the math without FPU.  I've looked through several ELS implementations which avoid FPU.  If i get it working maybe you can merge it.

closed 2020-02-11 by jschoch

terjeio, 2020-02-11:
Integer math (with scaling?) or fixed point math can be used if you go for PID, but there may be other options as well. I am lazy and opted for floats for MSP432.

@BassMatiFreecad may have some useful ideas, see issue gnea/grbl#809.

It would be nice if you can get spindle sync to work with ESP32.

I envy you those cheap ballscrews, I guess the one I am waiting for will end up costing me around $250...

BassMatiFreecad, 2020-02-12:
I see that you have the ESP32 going - which is at this very moment  hooked up to my machine and blinking :-D 
  At the moment, I am reading through https://github.com/bdring/Grbl_Esp32 in order to get a better understanding of grbl internals. After that I should be able to create a viable high-level design and put it to implementation. The aim is to tear grbl apart into one (G-code interpreter, motion planner and motion blocks sender) and multiple (single-axis motion block receiver, motion profile generator and step/dir pulse generator (i. e. single axis controller)), running on different MCUs.
  An idea is to supply the timers in the axis controllers with a clock signal generated in the motion controller so that the axis controllers keep running in sync to each other. If the clock frequency is controlled by the spindle speed, we could easily (at least I hope so) establish a rather fixed speed relationship between a spindle and one or more other axes.
  The division into multiple small and cheap MCUs should also eliminate the "not enough pins" problem.

jschoch, 2020-02-12:
@BassMatiFreecad  are you doing this with bart's port or grblHAL?

BassMatiFreecad, 2020-02-13:
Not decided yet. First I have to understand the structure of Grbl. This is currently done with https://github.com/bdring/Grbl_Esp32 because I have ESP32s lying around for testing. If I can use a JTag debugger to step through the code, I'm sure that helps a lot. 

Second step is to look into other repositories. Maybe there are some GEMs I want to integrate. Closed-loop steppers and DC Servos would be fine.

Third step is deciding on the hardware (MCUs mostly). All common stepper motor driver ICs should be supported, although I am somewhat inclined towards the Trinamics. They seem to deliver a decent job, so I guess I'll start with them first. 

Step 4 implements the motion planner MCU, the axis driver MCU, communication channel, step clock and sync. This verifies that I'm able to master the necessary hardware features. This is where I also look for a way to control the stepper clock from outside, either a potentiometer, software, or a rotary encoder mounted on a spindle.

Step 5 adds more axis MCUs for some dry-run tests.

Step 6 adds an IO module to get digital inputs and outputs available. Should be a no-brainer, based on SPI or I2C.

Step 7 integrates the electronics and mechanics (a design inspired by the Mostly Printed CNC project, perhaps I'll call it Mostly Bolted CNC...) and perform some real-life tests.

Step 8 adds additional power outputs to the design, such as Power PWMs, things that are required for 3D printers.

Everything should follow the Lego approach, so that a user can build a custom system for his very own needs, and extend it later when required. Although this requires several PCB types, they should be very small and thus very inexpensive.

terjeio, 2020-02-15:
@BassMatiFreecad : Maybe [this project](https://github.com/shooter64738/Talos-CNC) is closer to what you want?

BassMatiFreecad, 2020-02-15:
@terjeio: Thank you for the link. I'll have a close look. Like the fact that it runs on a PC with Visual Studio. 
Digging through the sources and trying to document and understand is always a hard job for me. When a viable concept is ready, I would like to find a team to bundle forces and knowledge.

shooter64738, 2020-02-22:
@BassMatiFreecad if you are looking for a control thats broken into basic parts (gcode larse, motion plan, trajectories, etc) I have a project alreadt broken out. A good portion of the motion planner is at its heart grbl, but the gcode processor is much expanded to support parameters and macros. The hardware that drives the motors is also broken into its own module.