terjeio/grblHAL #15: eeprom for Teensy4 board?

phil-barrett, 2020-02-16:
Is there an i2c eeprom you particularly like to use. Or at least, one that you know works well with your code? Designing a breakout board for the teensy4 version and have room for the IC.

terjeio, 2020-02-17:
Yesterday I made a "motherboard" for my [CNC BoosterPack](https://github.com/terjeio/CNC_Boosterpack), the BoosterPack has a 24LC16B (2K) on board so this is the first EEPROM I will add support for. It will be easy to add support for higher capacity EEPROMs if this should be required, many are pin compatible so no need to redo your board. FRAM (such as FM24C16B) may also be used, no need to wait for write cycles to complete for these. FM24C16B is pin compatible with 24LC16B.

The "motherboard" I made has an optional opto coupled/opto isolated PWM to DC converter circuit for spindle speed control, if you have room on your board you may consider adding one too.

phil-barrett, 2020-02-17:
Thanks. I've got my board mostly done. Optos for input, drivers for the stepper optos (the teensy4 can't drive much above 6 ma at 3.3v), opamp for 0-10v spindle, a set of relay drivers and a 24LCxxx  footprint. A few things to verify still but close.

terjeio, 2020-02-18:
I've just got initial code up and running for the EEPROM connected to pin 18 and 19. To which pins are you routing yours?

May I ask if you are using pins on the underside? If so, how are you going to connect to them? Via pogo pins? Pin headers? Some other means?

phil-barrett, 2020-02-18:
I'm using backside pins 24 and 25.   Here's my working doc.
[teensy 4 pinouts.pdf](https://github.com/terjeio/grblHAL/files/4221516/teensy.4.pinouts.pdf)

phil-barrett, 2020-02-18:
A little more on the SMT headers. Here is [the data sheet](https://www.mouser.com/datasheet/2/418/NG_CD_146131_H4-1238434.pdf) for ones I have used in the past. You have to take a little care to get them lined up when soldering.  Usually do one corner and verify alignment. Then finish with the others.

phil-barrett, 2020-02-18:
I'm a github rookie but created a repository for the GRBL Header file.  I'm about to send it off to get fab'd - just need to verify a couple of parts.
https://github.com/phil-barrett/grbl-teensy-4

Note, I'll make it open source once I know it works correctly (or close enough...).

terjeio, 2020-02-19:
Looks good, the KiCad importer worked well so was able to open the Eagle project . FYI the schematic files in .png-format are huge - lots of white space.

You want to add a pin-header for I2C? GrblHAL has a plugin code for a I2C keypad that can be used for jogging etc. This [I2C keyboard](https://github.com/terjeio/GRBL_MPG_DRO_BoosterPack/tree/master/Firmware/Keypad%20processor) code is my current implementation of one, can handle up to 25 keys in a 5x5 matrix. I am using this pinout for my BoosterPacks:

<pre>
Pin 1: GND
Pin 2: SCL
Pin 3: SDA
Pin 3: Keypad strobe 
Pin 5: 3V3 (or 5V as I have an I2C level shifter with selectable drive voltage on my board)
</pre>

Can I link to your repository from the Teensy driver when you have verified the design?

Thanks for the datasheet link, will try to hack together a SMT pin header from an angled through hole one for my prototype board...

phil-barrett, 2020-02-19:
That's funny, I added an I2C header after your last message.  Have to rework it though since the I didn't realize a strobe was needed.  I assume pin 4 (not 3) is the strobe.  I assigned all the back side pins but some of them are rather unlikely (7th relay or or 3 rd step enable) so I'll multiplex one, probably relay.

On level shifting, I'm still a bit unsure I absolutely need the HCT541s to interface with the stepper drivers. I looked at a lot of different drivers and they are vague or silent about actual current consumption and some say 5V or higher. Also the i.MX RT1060 datasheet is maddenly silent on GPIO current limits. I did see one note in the PJRC (teensy) forum that implied the limit is around 6 mA which puts it on the edge for decent Opto interfaces.  Cheaper stepper drivers may not be so friendly.  I guess if it turns out unnecessary it's a wasted $1 per board.

By the way, on soldering the SMD pin header - I remembered how I did it last. If you install the side pin headers first, you can use a spare socket to hold the SMD ones in alignment while you hit them with the iron.  Tight but doable.

terjeio, 2020-02-19:
Oops, fixed the pin number...

I managed to hack a through hole pin header and solder it while aligning it as you did. Used soldering paste. Running a test file with the basic BoosterPack now, code for the Trinamic version will have to wait for later.

For buffering I will use a SN74LVC07 - open-drain is what I want for the BoosterPack (running without buffering on the prototype). As for the GPIO current limits I saw a formula mentioned somewhere that did not make any sense to me. I suspect the limit may be related to maximum power dissipation, which is dependent on many factors in addition to the GPIO current. I note the MCU feels slightly warm to the touch while running the test...

I have modified some DM420A drivers to run with 3.3V by reducing the internal drop resistors to 120R, cannot remember now what the original value was - current was around 10mA?

IMO the buffers will act as a protection for the MCU as well as providing drive current - $1 is not much unless you are going to make thousands of copies...

phil-barrett, 2020-02-19:
Yeah, I was guessing that most stepper driver optos like 10 mA - kind of a standard engineering number though the LTV486 I have spec'd looks like it has a good CTR down to 4 mA or so.

By the way, the HCT logic interfaces nicely with 3.3V while running at 5V. Thus allowing 5V external interfaces. (except the I2C hdr).

I'm fine with you linking once I'm done.  I'll probably redo the repository and put up my other BoBs. 

I like LV logic, great drive current, except for the Ioh number - TTL logic levels rocks there.

phil-barrett, 2020-02-19:
pushed the changes with i2c header.

phil-barrett, 2020-02-20:
Hi, I made a change to the pin assignments to put the Spindle PWM pin on the Teensy LED pin. This is so it forms a simple diagnostic for spindles.  I've seen a lot of people flailing over VFDs and this will at least let one know that there is a pwm signal going to it.  So, Spindle PWM is on 13 and Spindle enable is on 12.  I updated the PDF doc to show this plus also changed the pin assignment tables to show alternate assignments more readily. Hope this isn't a problem.

closed 2020-06-19 by terjeio