terjeio/grblHAL #16: Errors building iMXRT1062

phil-barrett, 2020-02-20:
Trying to build iMXRT1062 driver.

I'm getting I2C port undefined error.
> i2c.c: In function 'i2c_init':

> i2c.c:195: error: #error "I2C port is undefined!"
>  #error "I2C port is undefined!"

Very early in the build I get this. Could be related? 
> Error while detecting libraries included by C:\Users\philba\AppData\Local\Temp\arduino_build_641586\sketch\i2c.c
> Using cached library dependencies for file: C:\Users\philba\AppData\Local\Temp\arduino_build_641586\sketch\uart.c

phil-barrett, 2020-02-20:
By the way, was able to compile the SAM3X8 version. No board to test it on, though.

phil-barrett, 2020-02-20:
Not sure what happened but that error went away and got  include file errors. That occurs because in eeprom_24LC16B.c, lines 33, 34 read:
`#include "../../grbl/grbl.h"
#include "../../grbl/plugins.h"`
but should read:
`#include "../grbl/grbl.h"
#include "../grbl/plugins.h"`

Making the change allows it to compile.  

Some warnings in i2c.c though:
> warning: "USB_SERIAL" redefined
> warning: 'rxbuffer' is static but used in inline function 'serialRxCount' which is not static

terjeio, 2020-02-20:
The errors (there are three) in eeprom_24LC16B.c is likely due to me not keeping source trees in sync. I use a local subversion repository for my code, and I copy from a "master project" for all shared code that goes to github.

Moving spindle PWM to pin 13 will require some code changes, a different timer will have to be used. I will keep pin 12 as an option as this is how I have routed my prototype board. Pin options for PWM will be limited to 12 or 13 though - I am not prepared to make this generic.

Re warnings:

The redefinition of USB_SERIAL is perhaps ok? I cannot find a menu item from where compiler flags are set in the Arduino IDE, perhaps they are hidden so that users cannot mess with them?

I'll add inline to serialRxCount().

No trace needed - would rather have a decent IDE and a debugger... It is bad that many Arduino boards does not bring out the JTAG or SWD pins...

terjeio, 2020-02-20:
Update comitted.

I had to change` hal.f_step_timer` in driver.c line 959 from 24Mhz to to 33Mhz for feed rate timing to be correct - why I do not know because I am near 100% sure I verified 24Mhz to be correct earlier. I would like to know what the correct setting is for you before spending time investigating. The clock tree for this processor is not easy to comprehend so I would rather avoid digging into that...

phil-barrett, 2020-02-20:
Good morning! thanks for the update.   What time zone are you in?

Not knowing GRBL code beyond changing options, I have no clue if the redefinition is ok. Looked at it, probably ok.

Thanks for adding 13 to the SPWM pins.  I think the led will help with diags.  I'm fine with locking it down.  Are there any other pins that can't be (easily) moved? I looked through your booster pack mappings and 12 was the only one with such a caution.

Will pull the update in just a bit and try.  I enabled USB serial so we'll see how that goes in a bit. 

Next, I'll write pin mappings include file for my board. 

Yes, my biggest complaint about Paul (Mr Teensy) is that he has steadfastly refused to support HW debugging. If he did that (even with a debug version of the teensy) it would make his stuff a slam dunk. I had  some code that still has a lingering error that needs HW debugging.

terjeio, 2020-02-20:
I am at usually at GMT-1 (CET), so 16:12 now.

All GPIO pins can be remapped, however there are a limited set of I2C and UART pins available so restrictions are in place for those. The GPIO handling in the driver is very flexible so no restrictions for these (as I know of). To avoid code complexity (and making testing more time consuming) I want to restrict the use of PWM, UART and I2C pins to as few ports as possible.

Your relay outputs will need extra code, IMO best done by using the [ioports template](https://github.com/terjeio/grblHAL/tree/master/templates) as a starting point.

The default pin mappings in driver.h is currently for your board, perhaps the default should be for the "simple machine": 3 axes, no pins on the underside, option for I2C.

phil-barrett, 2020-02-20:
I'm at GMT+8, west coast USA.  

I have some thoughts on simple machine but will start a new issue to keep the data in it's proper place.

closed 2020-06-19 by terjeio