terjeio/grblHAL #17: Simple Machine Definition

phil-barrett, 2020-02-20:
Trying to build iMXRT1062 driver.

I'm getting I2C port undefined error.
> i2c.c: In function 'i2c_init':

> i2c.c:195: error: #error "I2C port is undefined!"
>  #error "I2C port is undefined!"

Very early in the build I get this. Could be related? 
> Error while detecting libraries included by C:\Users\philba\AppData\Local\Temp\arduino_build_641586\sketch\i2c.c
> Using cached library dependencies for file: C:\Users\philba\AppData\Local\Temp\arduino_build_641586\sketch\uart.c

phil-barrett, 2020-02-20:
Still thinking about these issues.  I'd like to address the question of I2C vs other pins in the simple machine definition. I'm willing to go along with I2C on the front side but that does cause either coolant or door to be pushed to the back. 

So, the question is what is the priority order for I2C, Door and Coolant?

terjeio, 2020-02-20:
> So, the question is what is the priority order for I2C, Door and Coolant?

Select some pins for dual purposes and add config options so the end user can decide? If USB is used then the UART pins may be reconfigured.

phil-barrett, 2020-02-21:
Good point about RX and TX. I had assumed they were sacred but that's old school arduino thinking. USB will be pretty common use case for a simple machine. 

I wonder about how a network interface would work for teensy based system. if that's possible at all.  Certainly it's my goal to move to ethernet connection on my next machine.  But then perhaps that's not so simple...

terjeio, 2020-02-21:
There is an Ethernet MAC on the MCU but I belive I read somewhere that the pins were not brought out. If true the only option is an external controller with a serial interface?

Two of the grblHAL drivers supports ethernet and one wifi so a fair bit of the needed codebase is already present.

closed 2020-09-04 by terjeio