terjeio/grblHAL #18: Slave A Axis

phil-barrett, 2020-02-20:
I'm not clear on how a Axis A works in GRBL. I see two distinct uses for it but am not sure both are handled.  Case 1) is a true 4th axis - horizonatal or vertical.  I think that's handled.  

Case 2) is when A is slaved to Y.  In some cases, it is as simple as wiring two stepper controllers into the Y outputs. sub case 2A is where the output of Y is mirrored on A. Neither are independently controllable. I think that is handled but haven't seen how.   But sub case 2B is where you can control either Y or A to be able to "jog" the Y beam (X axis) into square.  I don't think that is handled in GRBL. It's important for cases where either Y or A misses steps and you need to return the gantry or table to square.

terjeio, 2020-02-20:
A, B and C axes in grbl are separate axes.

Slaved axes should not (IMO) be confused with these.

Three modes of operation is possible for slaved axes:

1. Electrically connected step and dir signals.
2. Separate step and dir signals - separate in that they are assigned their own GPIO pins. 
3. Same as 2. but with the option of auto-squaring. This requires an addtional limit switch and code for handling running only one motor at a time during (final) homing. Squaring by jogging is new to me - you really want that?

GrblHAL has core support for 3., however this is currently untested. The driver has to be enhanced for this, I am waiting for user feedback for my first attempt at that (for the SAM3X8E driver - code not yet published).

The nomenclature I chose for the SAM3X8E is the prefix `X2...`, `Y2...` etc. for the step and dir signal definitions, and I add the postfix `_Max `to the additional limit signal definitions. The IMXRT1062 driver already has entries for the required limit signals in driver.c.

I do not own a machine wih slaved axes so I am not able to test this myself (unless I hack my 3D printer for the purpose - it has a slaved Z-axis electrically connected).

phil-barrett, 2020-02-20:
OK, good to know. Thanks.  I do expect my next CNC to be a moving gantry with 2 Y Axis motors.  I like the idea of using Y1 and Y2 for those instead of Y and A. Logical.

phil-barrett, 2020-10-04:
I believe this has been resolved to everyone's satisfaction.

closed 2020-10-04 by phil-barrett

Lopezno68, 2021-04-23:
The dual axis for LPC17xx is working? I'm building a CNC with dual motors in the Y axe and I use the MKS SBASE board. There is information about configuring the auto-square?

terjeio, 2021-04-23:
> The dual axis for LPC17xx is working?

No, I have not added driver support (yet) for this processor. Option 2 as explained above is easy to add, option 3 is more work. 

>  There is information about configuring the auto-square?

Not really, however the [iMXRT1062 driver implementation](https://github.com/grblHAL/iMXRT1062/blob/master/grblHAL_Teensy4/src/driver.c) can be used as a reference of how-to for option 3. Most of the required code is within `#if SQUARING_ENABLED ... #else ... #endif` blocks if you want to give it try.

It is my plan to come up with a uniform way of setting up dual-axis/auto-square in _my_machine.h_ for all drivers. This may take some time though.

Lopezno68, 2021-04-24:
Thanks for the information, from what I have seen in the iMXRT1062 software, it is only necessary to modify the driver.c file according to your instructions and in my case the mks_sbase_map.h file with the definitions. Do I just have to modify these files?
The truth is that I am not very fluent in grblhal software but I will try to do it myself.
Anyway if you need me to test some of your software on my mks sbase board, tell me and I'll try ...

terjeio, 2021-04-26:
> Do I just have to modify these files?

Correct.

> The truth is that I am not very fluent in grblhal software but I will try to do it myself.

Great, I have plenty on my todo list already.

> Anyway if you need me to test some of your software on my mks sbase board, tell me and I'll try ...

The board map file needs to be verified, and also the changes needed for auto squaring.