terjeio/grblHAL #19: iMXRT driver with GRBL panel?

phil-barrett, 2020-02-20:
I was under the impression that GRBL/HAL worked with GRBLPanel. It connects to my Teensy build but does nothing (doesn't allow config params, no DRO positions, no jogging...).  I can connect with the Arduino com monitor and can read/write config and such.  I do quit out of the arduino monitor to run GRBLPanel. Other com programs work ok.

I am using USB Serial but other than that just the straight dist from github.

And, GRBLPanel on my machine works fine with standard GRBL/GNEA.

Anything I should look at?

terjeio, 2020-02-20:
Version? 1.0.9.17 works for me - both with UART and USB comms.

Note that some senders are not happy with extensions to grblHAL, setting [COMPATIBILITY_LEVEL](https://github.com/terjeio/grblHAL/wiki/Changes-from-grbl-1.1) should help for those.

Can you test with [my sender](https://github.com/terjeio/Grbl-GCode-Sender/releases/tag/v0.6-alpha), it should work with all extensions.

phil-barrett, 2020-02-20:
I tried downloading the sender zip file but get a disallowed - "zip is not commonly downloaded and may be dangerous" message. Weird, I dl zips all the time never a message like that. Anyway got it and it shows alarm state. It shows error 18.

Using a terminal program, it looks like grbl/hal comes up in an alarm state. Door Open. I moved door to pin 8 and pulled it to ground.  Still comes up in an alarm state but throws error 18 when I send a $X. I don't see that in the code and existing grbl/gnea wiki doesn't show 18 in it's error list. grbl/hal specific?  Still working to understand the error reporting code.

So, I'll try pulling all the limit and input pins to 0. Starting with limits.  could be that.  I should be able to invert limit sense but not while locked out.

terjeio, 2020-02-20:
Error codes can be found in grbl/gcode.h -` Status_Reset = 18` 

This is a grblHAL specific error code - grbl allows you to continue if the reset signal is asserted, grblHAL does not. Perhaps I should remove this check for one of the compatibility levels?

Also, asserting reset seemingly kills USB comms - looking into how to fix that now. Reason is that Teensy serial USB requires constant polling by grblHAL for real-time commands to work - I would like to have a callback or interrupt to hook into instead.

phil-barrett, 2020-02-20:
Ah, things are clearer now.  I was getting some flakiness and inconsistent behavior but it was likely due to killing USB comm via reset.  Abort was the state that was killing my sessions, by the way.

Paul is pretty active on the PJRC forum, he wrote most of the Teensyduino code and will probably help out if you ask. Plus I think this is something they would really like to see working well.

terjeio, 2020-02-20:
Can you try with adding `usb_execute_realtime(0)` to `systick_isr `at the end of driver.c?

```
// Interrupt handler for 1 ms interval timer
static void systick_isr (void)
{
    systick_isr_org();
    usb_execute_realtime(0);
    if(grbl_delay.ms && !(--grbl_delay.ms)) {
        if(grbl_delay.callback) {
            grbl_delay.callback();
            grbl_delay.callback = NULL;
        }
    }
}
```

terjeio, 2020-02-20:
It seems it was/is PJRCs plan to add a hook as they put this comment in the code a few places: `TODO: trigger serialEvent`.

The best solution would be to avoid double (or is it triple now?) buffering of RX data, I need to take a closer look at how that can be done. For the LPC1769 driver I added hooks into the USB stack at a lower level - it should be possible to do the same for this stack as well.

Anyway, it is early days for this driver - cannot expect everything to be perfect yet...

phil-barrett, 2020-02-20:
Added the code (replaced the prior systick_isr).  It works but with one problem.  If I start your gcode sender after a power cycle (or code upload), it (gcode sender) hangs. If you run a serial terminal app and talk to grbl, it works fine. Then if you quite the terminal app and run your gcode sender it works fine.

phil-barrett, 2020-02-20:
uh, wait. I backed out the change and gcode sender still won't start until something else has talked to grbl.

terjeio, 2020-02-20:
It seems to me that when USB is enabled booting (of grblHAL) is delayed until the first connection... Arduino insanity?

phil-barrett, 2020-02-20:
I've had issues with the teensyduino stuff on startup in the past. Not sure if this is the same.  GCode sender hangs pretty hard.  Windows task manager can't kill it.  Though clicking on the close box while it's hung get you there.  Then I have to power cycke the teensy because it can't find the com port.

phil-barrett, 2020-02-20:
t be clear, I'm now running with usb_execute_realtime(0); in the code.

terjeio, 2020-02-20:
I enabled DTR pulsing on startup, then it works! So it does not wait for connect, it waits for DTR to be pulsed. I really hate this...

For my sender I have to add config options for this (ESP32 needs RTS to be pulsed for a reset, but works happily without). Not sure what GRBLPanel does or if it is configurable.

phil-barrett, 2020-02-20:
I'm giving up on GRBLPanel.  It's seems to have other issues and I'd prefer to focus on the hal port.

terjeio, 2020-02-20:
I can confirm that when USB is enabled application boot is delayed until DTR is pulsed. I assume this is due to the bootloader somehow claiming the USB port for programming until so happens. I believe there is nothing I can do about that on the firmware side...

I will add config options to my sender tomorrow as it is just past mindnight here, time for some sleep.

phil-barrett, 2020-02-21:
Thanks for all you are doing. I'm having fun with this.

terjeio, 2020-02-21:
New version of my [sender ](https://github.com/terjeio/Grbl-GCode-Sender/releases/tag/v0.7-alpha) is now up.

Thanks for taking time to test and provide feedback.

terjeio, 2020-03-03:
Here is the patched version of Grbl-Panel that I made:

[Grbl-Panel.zip](https://github.com/terjeio/grblHAL/files/4280836/Grbl-Panel.zip)

Only minor changes made to stop it crashing with grblHAL, only usable for testing?

closed 2020-03-03 by terjeio