terjeio/grblHAL #21: IOPorts wait on input

phil-barrett, 2020-02-24:
I've been working on the ioport code for my breakout board and think I've got it understood except for wait on input.  What is that for?

phil-barrett, 2020-02-24:
ah, I went and read Unix CNC M66 description.  There doesn't seem to be a mechanism for defining inputs.  Work in progress or is it expected to be implementation dependent. I don't suppose it's as simple as spinning on a digitalRead().

terjeio, 2020-02-24:
Wait on input is not (yet?) usable with gcode as there are no variables and no conditional execution. I added it anyway since an implementation may be user defined (any ioport code is by definition).

I have not tested any wait code but a possible implementation could be something like this:

If not `IMMEDIATE` mode set up interrupt handling for the pin and in the isr set a flag when the `Mode` condition is met.

Poll this flag while in a delay loop, perhaps by calling `delay_sec` in nuts_bolts.c as this keeps real time processing alive. `hal.ms_delay` may also be used for the delay, perhaps with a callback.

Depending on whether the read timed out or not (or what the input is when `IMMEDIATE` mode is used) take appropriate action (in the `wait_on_input` function itself).

Hmm, looking at this now I believe I should have added a  return value (`status_code_t`) to this function...

phil-barrett, 2020-02-24:
OK.  glad I wasn't missing something obvious. Let me know what you do in the template code. 

This also has made me think about the output side of ioports (M62-M65 outputs).  The unix cnc docs are vague about how things like hold, errors and termination interact with the outputs.  I assume that the reset code means that you set the M62-M65 outputs to some default state. I am also assuming that M62-M65 outputs are not affected by hold/lock/error states - since the user turned them on, they can turn them off.  Is that correct? Though I can imagine a situation where you want things shut down but don't want to hit a full reset. (as is surely obvious, I'm still a little unclear about the various GRBL states)

A question about analog output M67, 68.  I presume the valid range of values (Q in the Unix CNC docs) is implementation dependent.  I'm thinking 0-10V is typical.  So, analog_out() is responsible for mapping the float value input by the MCode into something sensible on the output port.  The 1062 doesn't have an ADC but PWM and a good filter could suffice. I presume that Arduino analogWrite is not reasonable here.  I noticed that AXBB-E has 3 analog outs.  Easy enough to add one more to my breakout - there's an unused opamp on it and I could move down to smaller Rs and Cs in the 2nd order filter to get it all in.  (I use 1206s because they are easy to hand solder but I can still do 803s.)

Finally, just for confirmation, Unix CNC talks about 3 digital and analog outputs each max. I presume that's just an implementation detail. I've defined 6 digital in mine.

By the way, my breakout PCB has shipped from China. Should get it around mid-week.

terjeio, 2020-02-24:
Whan happens in ioports.c is completely implementation dependent and should be tied to the selected board definition. State changes does not change any outputs by default, however any implementation can hook itself into `hal.execute_realtime` if this is required, the current state is passed as a parameter for that entry point.

Reset handling is not required, again implementation defined. Up to you...

The [SD card plugin](https://github.com/terjeio/grblHAL/tree/master/plugins/sdcard) does some fancy stuff with the HAL pointers, amongst them detecting when a program ends, if you need outputs to change on program end it can be done in the same way.

What the analog output values translates to in the real word is also implementation defined.

As max for number of I/O signals allowed for each category I decided to put the grblHAL limit at 256 (since that fits an uint8_t), it is an extension after all.

So, you are pretty free to do what you want - and the pointer based HAL allows you to do it without messing with the grblHAL core and with only minor changes to driver.c (if only outputs are implemented a call to` ioports_init()` will do).

Note that input interrupt handling is done in driver.c, do not use the Arduino framework!

phil-barrett, 2020-02-24:
Where does ioports_init need to be called in driver.c?

terjeio, 2020-02-24:
I would put it at the end of `driver_setup`, here is an example from another driver (for the keypad plugin):

```
...

    hal.stepper_go_idle(true);
    hal.spindle_set_state((spindle_state_t){0}, 0.0f);
    hal.coolant_set_state((coolant_state_t){0});

#if KEYPAD_ENABLE
    keypad_init();
#endif

...
```

Perhaps we should standardize `IOPORTS_ENABLE` as a #define to test for?

phil-barrett, 2020-02-25:
OK. Thank you.

I'm fine with IOPORTS_ENABLE.  Defined in ioports.h I assume. Set to 0 if no ioports, 1 if there are?

terjeio, 2020-02-25:
Perhaps best place will be in your board map file? And yes set to 0 and 1 as you indicate.
Since ioports.c should be tied to your board map it would be a good idea to have a folder for your map file and your ioports.c. The code in your ioports.c should then be compiled only if your board is selected in driver.h.

ioports.h should be the same for all boards, and pin definitions related to ioports should be in the board map file?

phil-barrett, 2020-02-25:
OK.  I did some defs in .h but will move. I created an ioports directory under src, should probably be in main?

By the way, I get unsupported command when I try to use M62-M65.  How does that get enabled?

terjeio, 2020-02-25:
Your map file and ioports.c should be in a folder in main as they are a part of, and specific, to the Teensy driver and your board, folders in src are for the grbl core and plugins shared across drivers.

Oops, I keep forgetting that the Arduino IDE is rather primitive, IIRC any subfolders has to be in src...
Perhaps myboard_map.h and myboard_ioports.c in main is the best way to name board specific files then? ioports.h can keep its name as is.

Calling ioports_init() should automagically enable M62-M65 for the pins declared in your ioports_init() function. Upload your ioprts.c here if you want me to take a look (will have to wait until tomorrow as I am way past my bedtime now).

terjeio, 2020-02-25:
I did a simple test by just adding ` ioports_init()` to driver.c and code for turning the LED on and off in `digital_out()` - and it worked as it should. Perhaps your ` ioports_init()` never get executed?

A tip for debugging is to uncomment `DEBUGOUT` in grbl/config.h - then you can use the `hal.debug_out(bool on)` call to turn off/on the led connected to pin 13 (provided it has been set up as an output and is not set/reset elsewhere in the code).

phil-barrett, 2020-02-25:
It turns out my changes to driver.c didn't get built so of course it didn't work. Using an external editor with adruino and that's where I went wrong. sorry to waste your time. 

thanks for the debugout info. will use that.  running the USB serial test now. Got my teensy order so will build up a second to debug at ioports.  I'll try not to send false alarms.

closed 2020-06-19 by terjeio