terjeio/grblHAL #23: EEPROM support

phil-barrett, 2020-03-06:
I'm testing eeprom support with the teensy4 version. Or, I should say I'm trying to test it.  My board uses backside pins 24 and 25 for i2c.  The code has I2C_PORT 4 selected so should be correct port.

I set EEPROM_ENABLE to 1 in driver.h in the #else clause of #if CNC_BOOSTERPACK
CNC_BOOSTERPACK is set to 0.

Anyway, I have a 24LC16 installed and 2 pullup resistors (2.2K - I did not have 2.4Ks).  HAL compiles and uploads but does not start up.  With EEPROM_ENABLE set to 0, it compiles, uploads and run

I've verified that my pinout for the 24LC is correct and that I have pin connectivity between the traces and the IC. 

Not sure how to proceed. I'll figure out how to test that the pins are actually connected (back side pins). Anything else I can do on my end?

phil-barrett, 2020-03-06:
Verified pin connectivity on the SDA/SCL pins from the teensy 4 to the 24LC pins with a program to toggle the pins. Is it possible I need higher value pull-ups?

terjeio, 2020-03-06:
My fault, bad copy of definitions. Replace `i2c4_hardware` definition in i2C (around line 73) with this:

```
static const i2c_hardware_t i2c4_hardware = {
    .clock_gate_register = &CCM_CCGR6,
    .clock_gate_mask = CCM_CCGR6_LPI2C4_SERIAL(CCM_CCGR_ON),
    .port = &IMXRT_LPI2C4,
    .irq = IRQ_LPI2C4,
    .sda_pin = {
        .pin = 25,
        .mux_val = 0 | 0x10,
        .select_reg = &IOMUXC_LPI2C4_SDA_SELECT_INPUT,
        .select_val = 1
    },
    .scl_pin = {
        .pin = 24,
        .mux_val = 0 | 0x10,
        .select_reg = &IOMUXC_LPI2C4_SCL_SELECT_INPUT,
        .select_val = 1
    }
};

```

My pin 25 had a bad solder joint as well, perhaps soldering the connector on the underside is not going to be easy for some users.

phil-barrett, 2020-03-06:
That works better. Now, how do I know if the eeprom is working? Tried writing a stand alone program to test it but it's not working yet.

It appears my soldering on the backside was good but it is really hard because of all the caps next to the pads. I managed to hit a 402 cap with the iron and knocked it off.  one dead teensy for lack of a cap.  This is why I think the Teensy 4.1 can't come soon enough.

My V1 board appears to be working quite well. All inputs, outputs and relay drivers work. I have a few small mods for the next version but it's pretty much ready to go.  Go ahead and link to it. Are you ok with me placing a pre-built hex file in my repository?

terjeio, 2020-03-06:
> Now, how do I know if the eeprom is working?

Change a setting and/or set the info string to something with the $I command. Then compile with EEPROM disabled and check if changes are still there (should not), compile again with EEPROM enabled and changes should reappear.

---

Great that your board works ok, I'll add a link. And yes it is ok that you add a hex-file in your repository.

I am still fighting with the serial over USB problem, using `Check` mode now for faster testing. A STM Redpill board completes the 10hrs test in 3 minutes(!), no failures yet. All Arduino versions fails randomly. About to run the same test with a ReArm now.

phil-barrett, 2020-03-06:
OK.  Verified that GRBL settings changed with EEPROM enabled disappear with it disabled and then reappear once it is enabled again.

Autodetecting EEPROM would be a good thing but probably not worth the effort.

Ah, so it's not just a Teensy problem (there's an American joke in there!). If there is anything I can do to help, let me know.

On a somewhat different topic, in testing my board, I discovered how bad some Chinese electronics are.  I have 3 relay boards that I bought for my CNC rewiring project and all 3 are utter garbage. A 12V one was delivered with a 5VDC relay that doesn't fire at 5 and overheats at 12.  Another 4 channel one has 2 bad channels. I wound up laying out and ordering my own relay PCB so I can control the quality. Had to test the HAL board with bare relays.

terjeio, 2020-03-06:
Autodetecting an EEPROM is, IMO, not a good idea. For that to work all I2C ports needs to be scanned and then the pins for these ports must be reserved(?). Somehow the autodetect routine would need to find out if is a 2K EEPROM that is found as well since this uses a different command set. So I agree, not worth the effort.

I will soon commit some updates for the communication issue for you to test. I have slowed down the Teensy a bit when it comes to reading data off the USB RX input buffer, initial testing looks promising for that. If I not do that the tsender seems to freeze in `Check` mode, but that is likely due to the Teensy responding so fast that the application message pump hardly gets any time to run.
Time to completion when using agressive buffering goes from 3 minutes to 4 minutes (for the 10 hr job). Without agressive buffering time to completion is arount 8.5 minutes.

Tsender seems to be quite robust now, both the Redpill and the ReArm has completed quite a number of tests without any failures. Notably grblHAL drivers for these uses vendor provided USB drivers, and I have hooked the grblHAL RX routine in at a fairly low level. This unlike the Arduino USB drivers where the RX data is polled on systick events.

As for Chinese electronics I generally have had few problems, but then I tend not to buy stuff that is known to be often faked, such as semiconductors. Cheap can end up expensive...

closed 2020-06-19 by terjeio