terjeio/grblHAL #24: Power Fail support

phil-barrett, 2020-03-06:
I'm sure this a can of worms but thought it a good idea to at least write it down. A guy on the GNEA ARM thread mentioned it and I have to agree it would be a good feature that separates HAL from the others.

While power failures aren't that common, some people have panic E-Stop switches that cut power. In at least some of the E-Stop situations, a resume could be possible. It would also be a nice feature to be able to pause (hold) a job, save state and turn off the machine for maintenance, repair, sleep or what ever. Then resume the job where left off.  Given that the Teensy and others have some sort of battery backup capability, it is quite possible to do.  I'm sure there are plenty of issues. Probably control application interactions. A pin to indicate power failure could be useful here and pretty easy to do.

terjeio, 2020-03-06:
Yep, I saw that comment.... It would require an UPS to start with, and use the power failure signal from that to start a controlled shutdown (a hold, or perhaps better a parking sequence) followed by copying the current state to non-volatile storage. And then there is the issue of powered down stepper motors holding their position even without any significant load. Even with microstepping enabled?

An example from the real world: my router spindle needs the z-axis motor to be powered to avoid it dropping down by its own weight. It would have to be parked in a position where it could not move on its own. Or pehaps some kind of axis locking could be employed? 

So IMO the only practical solution is to use a UPS, perhaps in combination with backup generator for long power failures. Starting the generator on a power failure signal from the UPS, perhaps after some delay, should be straightforward to implement.

Can of worms indeed.

phil-barrett, 2020-03-06:
OK, makes sense.

closed 2020-06-19 by terjeio