terjeio/grblHAL #25: Big test case.

phil-barrett, 2020-03-09:
67.5K lines.  Concentric circles with arcs. Running time for me is 7:23 on a machine with X and Y max set to 5000 mm/min, 100 mm/s^2 acc and 200% over ride.

[concentric circles as arcs fr5000.zip](https://github.com/terjeio/grblHAL/files/4309205/concentric.circles.as.arcs.fr5000.zip)

phil-barrett, 2020-03-10:
Ran this test case today using TSender alpha 10. It ran for  7:05 when I stupidly pressed hold.  It would not restart so the full 7:23 was not reached (18 minute short). I will restart for an overnight run.  However, I believe GRBL/HAL using USB_SERIAL_GRBL 2 is solid. Will run some of my other test cases against in the next few days. Are there any areas you would like me to focus on? I will report the TSender issues on it's repository issues list.

phil-barrett, 2020-03-10:
This test ran to completion. 7:23:37 USB_SERIAL_GRBL 2.  Alpha 11 TSender, GCode Viewer On, Aggressive Buffering Off.  Will retry with AB On.

We probably need to start using build or release numbers and some way of summarizing options.

terjeio, 2020-03-10:
Great, getting there slowly...

I have added a build number already,  see _File > About_ in tsender (`DRIVER VERSION:nnnnnn`). I need to remember to update this on changes. The HAL interface has gotten a pointer for an options string as well, on todo list for adding one for all drivers.

I believe I have fixed the feed hold issue, I may post a patch here depending how a split of the gcode list and related buttons (Start, Hold, ...) pans out. Split is nearly complete - but I want to add the gcode list to a tab view with additional tabs for a grbl response list and a small 3D view.

terjeio, 2020-03-10:
I have now run tests with Due and MKRZERO to completion, both with AB on. Running one (overnight) with `USB_SERIAL_GRBL 1` for T4, AB on and foreground polling of USB input stream.

There is an oversight on my part in the core when it comes to foreground polling of the USB input stream, it blocks exit from Hold state by the cycle start real-time command. Asserting the cycle start pin works. Foreground polling is currently only used by the Due and MKZERO drivers available on GitHub.

Will commit fixes/updates tomorrow after testing is completed.

phil-barrett, 2020-03-11:
Successfully finished another concentric circles as arcs fr5000 run with Tsender Alpha 11.  USB_SERIAL_GRBL 2. AB on, GCode Viewer on - 7:23:35 (same time as with AB off)

[VER:1.1f(Teensy 4.0).20200306:]
[OPT:VNMSL,35,1024,3,0]
[NEWOPT:TC]
[DRIVER VERSION:200303]

As a side note, $I sent in TSender doesn't display anywhere.

terjeio, 2020-03-11:
Good, I ran 3 successful tests overnight (T4, Due, MKRZERO) - all with AB on. GCode Viewer on/off should not have any influence, there is no live update of the plot yet (add a animated tool?).

> As a side note, $I sent in TSender doesn't display anywhere.

I am aware of that, a list control for displaying responses has been made but needs to be added somewhere.

phil-barrett, 2020-03-11:
Successfully finished concentric circles as arcs fr5000 run with TSender Alpha 11. 
USB_SERIAL_GRBL 1. AB On  7:59:55 No feed rate override so slower.

[VER:1.1f(Teensy 4.0).20200306:]
[OPT:VNMSL,35,1024,3,0]
[NEWOPT:TC]
[DRIVER VERSION:200303]

closed 2020-06-19 by terjeio