terjeio/grblHAL #26: Thoughts on options string

phil-barrett, 2020-03-10:
You've probably already thought through all of this but can't hurt to write it down. 

The options string should capture all (as many as possible, anyway) configurable options. Certainly from config.h options.  I don't know if it's easy to capture plugins. For example, keypad doesn't seem to have any way to register itself other than KEYPAD_ENABLE in config.h. Maybe each option should also have an options string that gets tacked onto the main options string. For diagnostics and debugging purposes, more info is better.  Certainly for testing and bug reports it's important.

terjeio, 2020-03-10:
There are several different categories of options and they have different use.

1. Basic options, these typically has to do with the operation of the core. Reported as part of the OPT: string - this is defined in the 1.1f spec and may be used by senders for configuration. grblHAL has added a few options to this.

2. Extended options, grblHAL specific - reported as part of the NEWOPT: string. Currently only used by tsender for configuration.

3. Driver specific extensions, grblHAL specific - reported as part of the DRIVER OPTIONS: string. This is intended for options not used by the core or senders. May be useful for the user as a part of system configuration (e.g. IP- and MAC-addresses may be shown here for networking capable drivers) - or for debugging.

All these are reported on a `$I` request, I do not want them to be reported automatically on boot.

Options from config.h: I have reduced the number of options in config.h quite a bit by moving them to `$<n>` settings. Some of the remaining are reported as part of 1 or 2 above. The rest are fairly special and unlikely to be changed?

Example `$I` report from the MKRZERO driver:

```
[VER:1.1f(SAMD21).20200306:CNC Boosterpack]
[OPT:VNMSL,35,1024,3,0]
[NEWOPT:SD,TC]
[DRIVER VERSION:200307]
ok
```
Reported in _File > About_ in tsender like this:

[image]

In the Settings: Grbl tab in tsender a backup can be made of the `$<n>` -settings, will add output from `$I` as comments at the start of this file.

PS: If you provide a binary for your breakout boards you may add a default `$I` string and lock it.

phil-barrett, 2020-10-04:
I believe this has been resolved to everyone's satisfaction.

closed 2020-10-04 by phil-barrett