terjeio/grblHAL #27: Noob question Teensy 4 setup

einencool, 2020-03-11:
Hello and sorry for this noob question.
I’m going to setup a new cnc router and wanted to go with a teensy 4.0.
I got the board today and wanted to flash grblhal with arduino ide. 
I installed the teensy flasher.
Then i copied the files from the grbl and eeprom folder to the driver/xxx (teensy) folder.

When i flash the Teensy the status Message finishes without errors.
When i plugin the board and try to connect to the board lets say with „UGS“ or bCNC or Lasergrbl
Then nothing happens. There is no error, but i can’t get any Status or something like this....

Can someone please guide me, what I’m doing wrong?
Thank you very much and one more sorry for my stupid questions

terjeio, 2020-03-12:
What is your setting for `#define USB_SERIAL_GRBL` in [driver.h](https://github.com/terjeio/grblHAL/blob/master/drivers/IMXRT1062/main/driver.h)? The default is currently for an external USB to serial breakout board connected to pin 0 and 1. Change this to 1 or 2 (try with 2 first) for native USB.

If you enabled EEPROM support in driver.h then try with it disabled. If there is a problem with the EEPROM the program is likely to hang.

Be aware that some senders are not happy with grblHAL extensions and may crash, either try changing the [compatitibility level](https://github.com/terjeio/grblHAL/wiki/Changes-from-grbl-1.1) or try [my sender](https://github.com/terjeio/Grbl-GCode-Sender) that is made to be compatible.

You may also try to connect with a terminal program such as [PuTTY](https://www.putty.org/) as a first step.

einencool, 2020-03-12:
Hello Terjeio,

thank you for your fast reply, I didn't know, that I have to change the "driver.h" but now I was able to get a connection to the software. It was full of alerts, but it connects, and thats a first good sign *g* 
It's 2 years ago when I set up my CNC with an Arduino Mega and my little Laser with an Arduino Uno and CNC shield... So I think I have to read a lot, how to set up and wire up my machine...

Is it possible to set up the Teensy 4 with a Arduino CNC Shield and route the right Pins together? Because I have a Shield lying around on my desk... Or will this destroy the Teensy? 

The following points I need for the CNC:
- XYZ + Y2 or "A" with external units for Nema 23 with 36V
- Endstops XYZ min + max
- Probe Pin for Z-height
- PWM for 2.2kw China Spindle (5V PWM signal)

Or would it be better to go with another Mainboard? Well, I think i should have asked the question before *g*
Greets Chris

terjeio, 2020-03-12:
Teensy 4 is a 3.3V device, 8 bit Arduinos are 5V - 5V will kill the Teensy. It _is_ possible to modify a board, ref [this thread](https://reprap.org/forum/read.php?219,479626). 

A better option will be to use a dedicated breakout board like [this one](https://github.com/phil-barrett/grbl-teensy-4) made by @phil-barrett - it seems to me to be a good match for your machine. 
The only issue may be your limit switches - do you need the min/max switches to be wired to separate inputs for auto squaring?

phil-barrett, 2020-03-12:
Einencool, take a look at my breakout board at https://github.com/phil-barrett/grbl-teensy-4. It shows how to interface 3.3V logic with the outside world.  Also, most of the Stepper Driver documents are pretty vague about interface requirements. Some say 5V but most are silent.  This is why I went to 5V outputs.  One other thing, if you are going to interfacing the Teensy4 directly with external devices be aware that electronics with 5V pullup resistors will kill the Teensy4. There are a number of stories on the pjrc forum about that. You can check for this by measuring the voltage on the device's input. If it is 5V, don't connect.

My board is working pretty well with the default pin assignments in the T4 driver.h. If you can solder SMDs, it's pretty easy to build up.  I will be putting the gerbers up for people who want to get the PCB made.  There's even a pretty picture of it in the repository now.

Also, I have a few extra copies of the bare V1.0 PCB and will give them out on a first come basis. Terje, if you want one let me know.

phil-barrett, 2020-03-12:
On the button's - as long as you stick to 3.3V, they will be ok.  But understand that by not isolating the inputs, you are opening yourself up to electrical noise and potentially "glitching" any job you are running.  Better to use either a low pass filter or an opto-isolator between the buttons and the Teensy.  My board uses optos.

einencool, 2020-03-13:
@phil-barrett I've seen your board and it looks really nice. What would be the cost for this board?
The soldering wouldn't be the big problem. I've got the TS100 with a fine tip, and my skills aren't sooooo bad :-)
But I haven't the time and the possibility to make my own board. 
Would it be possible to get one of your boards (maybe with seperate parts)? I'm from Germany.

So your Board can be connected to 5V and then I could handle my machine like before with the Arduino Mega? This is also only connected to USB Power...

In my actual machine I'm using direct connetcted Endstops with shielded cables and a low pass filter. Also the Steppers and the spindle is driven with 4x0,75mm2 shielded cables.

phil-barrett, 2020-03-13:
Cost for a bare V1.0 board is postage to Germany.  It looks like it is $2.45 based on the the US Post Office calculator assuming 2 oz (< 54g). You will need to find the parts locally.  I have a BOM that lists parts with some part numbers from Mouser (a US distributor) so you can know what to order. Though, there is nothing exotic on the list.

The board draws 5V from USB.  Worst case draw for the fully built up board is an additional 100 mA without relays (though likely a lot less). The Teensy is supposed to draw 100 mA max. At 200 mA total, it would be under the 250 mA. It is easy to switch the 5V to an off-board supply if USB supply is a problem.

Take a look at the schematics - all the external switches are isolated via opto-isolators. So, if your Endstops can drive an opto (they probably can), you are good.

By the way, the V1.01 schematics are in the repository.  The board was made from V1.00 schematic so there are a few minor changes that I can walk you through. Same build of GRBL/HAL works on both version.

einencool, 2020-03-14:
Hello Phil,

Then I would like to purchase a circuit board from you if it is possible to ship it to Germany. :-)
What changes have been made from V1.00 to V1.01? Anything that would be important for my project?

Then I‘ll start my new CNC with the old Arduino Mega ans then switch to the new one, when it is finished :-)

Di you put a Heatsink on the Main CPU of the Teensy? I have noticed that the temperature of the chip is rising quite a bit. I would guess 50-60 degrees.

Greets Chris

phil-barrett, 2020-03-15:
OK, send me your address - phillip.lyne.barrett  at gmail dot com.  We will use email to figure out payment. Probably paypal if you have access.

No heat sink on the Teensy.  While it may feel warm to the touch, it isn't that hot. I've been monitoring the temperature with an IR thermometer and the rise over ambient temperature is about 12 C. I've never seen it above 32C even after running a 9 hour test. One of the advantages of my design is that there is minimal load on the I/O pins of the IMXRT1062.  At some point I will play with dropping CPU speed to see if that affects the temperature.

You don't absolutely need an eeprom. Settings can be saved in flash. I'm keeping it in the design because there is room on the PCB and it is cheap to add. Less than $1 in quantity one prices. Terje can better answer why one would prefer an eeprom.

V1.00 to V1.01 differences are minor and probably will not affect you at all.  
- silk screen error on V1.00 - relay 5 and 7 labels are switched.
- changed LEDs - TH on V1.00, SMD on V1.01.
- Serial connection header changed - V1.00 only has RX and TX, added a GND pin to V1.01.
- V1.01 adds a pinheader for a switch to control the dust collector relay (on both, it's also slaved to the spindle relay). If you use a shop vac (Staubsauger? ich weiß es nicht) for dust control, the switch is convenient for clean up later.
- various traces have been cleaned up to look nicer. 

Have you ever soldered TSSOP packages?  These take a little care but I can talk you through it.

terjeio, 2020-03-15:
I prefer EEPROM (or FRAM) in order to reduce the risk of wearing out the flash. Most users will likely have a very low risk of doing so. From my use the most frequent updates are from zeroing the machine.

Typically flash may wear out after 10^5 writes, EEPROM after 10^6 and FRAM after 10^14.
For the T4 I am using a PJRC EEPROM emulation library that uses wear levelling so lifetime for that would be higher than the specified lifetime (I have not checked the datasheet).

Write speed is also a factor, FRAM is fastest requiring no delay, EEPROM typically needs a 5ms+ delay after each page written. I have not checked flash write times, but I would guess it is slower than EEPROM, page sizes are typically larger than the 1 or 2KB needed for grbl and the whole page often needs to be erased before writing even a single byte.

As a side note it can be mentioned that by default grblHAL defer updating non-volatile storage until in `Idle` or `Alarm` mode.

I have added support for [Maslow CNC](https://www.maslowcnc.com/) kinematics to the grblHAL core (currently untested), for this I would go for FRAM as this constantly writes the current position to non-volatile storage.

Finally, I consider having an I2C port available for other expansions a plus.

@phil-barrett : Thanks for the pcb offer. Currently I do not need one so perhaps better to let somebody else get it? I use TI processors/boards for my own machines and have 20+ BoosterPack PCBs available...

phil-barrett, 2020-03-15:
Yes, I've wondered about the flash wearout issue. Certainly an issue for a development board.

Maslow seems to be fading away. The web sites look pretty abandoned. Are there any active groups working on it?

Yes, I2C port stays and will get more love. I've been looking at the keypad code and it looks pretty straightforward.  I am thinking about how to use another teensy to interface to a USB pendant and talk via the I2C interface to the  teensy running grbl. It looks like only the T3.6 and T4.0 can be as a USB master.  It would need 5V though so the current interface on my board won't work. There may be other solutions that would work as well.  I'm looking at other uses of i2c to see best practices. Sparkfun Qwiic only puts the i2c signals on the bus but uses configurable address jumpers on target boards and an easy way to disable the pullups if needed. Also, their connector looks fairly rugged. 3.3V only simplifies the interface.

Also, there are some transceivers that allow for fairly long distance I2C - 10 meters or more though speed is slower (100KHz).

terjeio, 2020-03-16:
I do not now if there is any work currently going on for Maslow, there is a [Due port](https://github.com/ldocull/MaslowDue) available but no updates for a while. There seems to be a bit of activity on their [forum](https://forums.maslowcnc.com/).

Why 5V for the I2C interface when using second t4 or other 3.3V processor as a USB pendant middleman?

For my BoosterPacks I am using a level shifter (TCA9406) with jumper selectable interface voltage for flexibility. As for I2C usability: in my [CO2 laser build](http://www.buildlog.net/forum/viewtopic.php?f=16&t=2625) I have 4 microprocessors talking to each other via I2C - in what is a very noisy environment.

phil-barrett, 2020-03-16:
Because if I'm going to use something to drive a USB pendant, it will need 5V.  I would try to make the i2c level be 3.3V, though.

einencool, 2020-03-22:
Hello, 
today I've got a little bit of time to test some minutes, but I've got some strange problems, I don't know what I've done wrong and it's only the Teensy liing on the desk :-)

I flashed GRBLHAL onto the Teensy and it is recognized as grblHAL 1.1.
Then I always got the Problem (#13) about the Safety Door, so I uncommented 
"//#define SAFETY_DOOR_PIN     (29u)" in driver.h 

Then I got the Error Message #18 when I press the "unlock" Button and about this Error I can't find anything in the WWW.
I also tried the "#define COMPATIBILITY_LEVEL 1" in config.h

I tried it with bCNC / UGS Platform and your GCode Sender and every time I got the same problems.

And I can't change any values like $22=16, but I think that's why I didn't enable EEProm...

Here is the code from the Console:
`**** Connected to COM4 @ 115200 baud ****
Grbl 1.1f ['$' for help]
>>> $$
$0 = 10    (Step pulse time, microseconds)
$1 = 25    (Step idle delay, milliseconds)
$2 = 0    (Step pulse invert, mask)
$3 = 0    (Step direction invert, mask)
$4 = 0    (Invert step enable pin, boolean)
$5 = 0    (Invert limit pins, boolean)
$6 = 0    (Invert probe pin, boolean)
$10 = 382    (Status report options, mask)
$11 = 0.010    (Junction deviation, millimeters)
$12 = 0.002    (Arc tolerance, millimeters)
$13 = 0    (Report in inches, boolean)
$14 = 0   
$15 = 0   
$16 = 0   
$17 = 0   
$18 = 0   
$19 = 0   
$20 = 0    (Soft limits enable, boolean)
$21 = 0    (Hard limits enable, boolean)
$22 = 16    (Homing cycle enable, boolean)
$23 = 0    (Homing direction invert, mask)
$24 = 25.000    (Homing locate feed rate, mm/min)
$25 = 500.000    (Homing search seek rate, mm/min)
$26 = 250    (Homing switch debounce delay, milliseconds)
$27 = 1.000    (Homing switch pull-off distance, millimeters)
$28 = 0.100   
$29 = 0   
$30 = 1000.    (Maximum spindle speed, RPM)
$31 = 0.    (Minimum spindle speed, RPM)
$32 = 0    (Laser-mode enable, boolean)
$33 = 5000.000   
$34 = 0.000   
$35 = 0.000   
$36 = 100.000   
$37 = 0   
$39 = 1   
$40 = 0   
$41 = 0   
$42 = 2   
$43 = 1   
$44 = 0   
$45 = 0   
$46 = 0   
$56 = 5.000   
$57 = 100.000   
$58 = -5.000   
$59 = 500.000   
$60 = 0   
$61 = 0   
$62 = 0   
$63 = 3   
$64 = 0   
$65 = 0   
$100 = 250.000    (X-axis travel resolution, step/mm)
$101 = 250.000    (Y-axis travel resolution, step/mm)
$102 = 250.000    (Z-axis travel resolution, step/mm)
$110 = 500.000    (X-axis maximum rate, mm/min)
$111 = 500.000    (Y-axis maximum rate, mm/min)
$112 = 500.000    (Z-axis maximum rate, mm/min)
$120 = 10.000    (X-axis acceleration, mm/sec^2)
$121 = 10.000    (Y-axis acceleration, mm/sec^2)
$122 = 10.000    (Z-axis acceleration, mm/sec^2)
$130 = 200.000    (X-axis maximum travel, millimeters)
$131 = 200.000    (Y-axis maximum travel, millimeters)
$132 = 200.000    (Z-axis maximum travel, millimeters)
ok
>>> $G
[GC:G0 G54 G17 G21 G90 G94 G98 G50 M5 M9 T0 F0 S0.]
ok
>>> $X
[Error] An error was detected while sending '$X': (error:18) An unknown error has occurred. Streaming has been paused.
>>> $J=G21G91Y10F508
[Error] An error was detected while sending '$J=G21G91Y10F508': (error:8) Grbl '$' command cannot be used unless Grbl is IDLE. Ensures smooth operation during a job. Streaming has been paused.
`

terjeio, 2020-03-22:
@einencool Try with  `COMPATIBILITY_LEVEL 10` (the wiki incorrectly stated this is not yet published) for other senders than mine . For safety reasons grblHAL defaults to normally closed (NC) switches - set `$14=31` if using nomally open (NO). `Error 18` is because the reset pin is asserted, indicating it is either not connected or connected to a NO switch. Setting `$14=31` should fix that. Again, this is a safety feature in grblHAL.

You may check the output from a real-time status request (`?`), the `Pn:` part show which switches are detected as asserted. If using my sender asserted switches are displayed as red "LEDs" in the Signals box:

[image]

IMO none should be lit during normal operation.

einencool, 2020-03-22:
So when I’ll invert all pins , it should also work?
Like i said, in the moment only the teensy is on the benchtop, no wires , nothing connected. 
In the moment I’m searching for a new software. 
I like your lightweight software and will have a try with it. Is it possible to customize it? I would like to map the jogging and the macros to some keypads...

terjeio, 2020-03-22:
> So when I’ll invert all pins , it should also work?

yes.

$14=31 inverts the pins in the firmware. If possible it is better to use normally closed switches. 

> Is it possible to customize it?

Currently only for what is available from the Settings: App tab. For additional customizations then you may either add an issue describing what you want or try to modify the source yourself.

phil-barrett, 2020-03-22:
It can be very confusing.  Try using the default build settings and pull door, halt, cyc/start, and hold to gnd. That should get you going.  Then I would experiment with inverting individual control pins if you want to NO control switches.  Typically the Door and Halt switches are NC.  

Like Terje said, NC switches are preferable. This is because it is somewhat less noise susceptible than a NO configuration. I also have all my limit switches NC.

einencool, 2020-04-11:
Hello together,

today I tried just a little bit, and wanted to compile the newest Version from Github of your GrblHal.
I tried it with the `COMPATIBILITY_LEVEL 10` but it doesn't compile.
Then I tried with Level 2 which also doesn't work. When I switch back to Level 1 it works, but with my GRBL Sender it doesn't work.
I alos tried it with the GRBLHAL Version from mid of March, but here is the same behavior.

This is the Error message i got from "Atom"
Maybe you can tell me, what I'm doing wrong with this, because I only changed the Compatibility Level...

C:\Users\me\AppData\Local\Temp\arduino_build_533784\sketch\src\grbl\report.c:423:19: error: 'Setting_LinearSpindlePiece4' undeclared (first use in this function)

for(idx = Setting_LinearSpindlePiece4 + 1; idx < Setting_SpindlePGain; idx++)

^

C:\Users\me\AppData\Local\Temp\arduino_build_533784\sketch\src\grbl\report.c:423:19: note: each undeclared identifier is reported only once for each function it appears in

Fehler beim Kompilieren für das Board Teensy 4.0.

terjeio, 2020-04-11:
@einencool : it is my fault - from not testing for all config combinations after changes...
Fix committed: f83484a - please report back if you still have problems with your sender.

einencool, 2020-04-12:
@terjeio 
Thank you for your fast help, this fixes the Combile Bug.
But I have a problem with I think all versions, and I don't really know where I make the mistakes...
I always get the `Error:50` when I click the "Unlock" Button. It doesn't matter which sender I'm using. 
The Error is on `COMPATIBILITY_LEVEL 1 / 2 and 10`
I also changed the `#define USB_SERIAL_GRBL    1 // Set to 1 for Arduino class library, 2 for PJRC C library` to "1" but this doesn't change anything.
I can't find anything in the Internet about this error, so you are my last help :-)

I inverted the Pins and endstops, because I only got the Teensy plugged in into my Laptop...
[image]
[image]

terjeio, 2020-04-12:
Can you try to short the RESET_PIN (pin 14) to ground or try with `#define ESTOP_ENABLE  0` in driver.h? When set to 1 the reset pin is redefined as E-Stop. Ref the comment there:
_When enabled it only real-time report requests will be executed when the reset pin is asserted._

It could also be that inverting the Reset pin will resolve this. Will check tomorrow, I do not have a Teensy connected and it is late here...

Sorry about the confusion this is causing - I will have to improve how this is handled in my sender and make it clear in the documentation.

einencool, 2020-04-12:
Yeah, that was the Problem, I chosed the `#define ESTOP_ENABLE 0` version, and now it works with your sender, but with bCNC I didn't get it to work. 

With the Compatibility Mode 2 and 10 and can't invert the logik pins because $14 is not availible and when I choose `#define INVERT_CONTROL_PIN_MASK SIGNAL_MASK // Default disabled. Uncomment to enable.` in default.h I get an Error while compiling.

Well, I don't know, what I can do to get it to work with bCNC, and so I give it one more try in the next days when I get all the missing parts for the board for the Teensy, and when it doesn't wotk, I have to go with the old arduino and work with this on my new CNC. 

Maybe you have a idea what I'm doing all wrong, because I don't :-)

phil-barrett, 2020-04-12:
You should be able to send $14=15 in the console.  (I think it's 15 assuming the 4 pins are represented by the lowest 4 bits.)

terjeio, 2020-04-13:
I have installed bCNC to check, a bit confusing for me in the beginning but now I have a GCode program running!

There are a few remaining bugs regarding compatibility level handling, $10 setting is one of them.
Also, what is important is that a `$RST=*` command should be issued after compatibility level is changed - it is required if an EEPROM is used to store the settings.

I will commit an update after I have done some more testing, and also fixed the new compiler error you get.

I am sure we will eventually get the Teensy working with bCNC, hopefully with my next commit.

einencool, 2020-04-13:
I can't really say how much I want to thank you :-)
You are so helpful I can't believe it 

The reset I've done every time after I reflashed the teensy. In the moment I'm not using a eeprom,because the breakout board isn't ready. I ordered some wrong chips and because of Covid-19 they didn't send them in the moment (it's around 10 days ago)...
I think I'll try another store and order them there..

terjeio, 2020-04-13:
@einencool : updated core and Teensy driver comitted, I hope this works better. IMO bCNC is acting a bit strange, but I guess this it how it is designed to work. Among the strange things is that it reports "Not connected" even if connected - seemingly when it is not happy with the responses it gets from grbl, eg. when $10 is not set to an odd value. A file I tested with that contains an error just stops executing and status goes to Idle. I have to click on the status bar to find out that something went wrong.

Anyway, I appreciate your feedback - without feedback bugs will not get fixed unless I stumble over them myself. So thank you as well.

Some details:
When `COMPATIBILITY_LEVEL` is set > 1 `ESTOP_ENABLE` in driver.h will be disabled.
$14 is not supported by vanilla grbl - you have to use #defines in defaults.h as you found out, the compiler bug related to that should now be fixed too.

@phil-barrett : I have changed the `ESTOP_ENABLE` implementation. It will now use the correct inversion and pullup disable flag (bit 6) when enabled. The next tsender release will be updated for that.

terjeio, 2020-04-13:
@einencool : I did not notice your edit, another bug to be fixed... Will do that later, it is not critical.

einencool, 2020-04-13:
Hi @terjeio ,
thank you, now I can compile it, but as you said, bcnc is a little bit strange...
With your Version (any Versions from the beginning) I don't have any DRO Values, not the MPOS nor the WPOS. So I didn't ever try to start a GCode File, but like you said, now I tried it, and it seems to run on the PC Display, but I don't get any messages from the Positions. And I don't know if I can move my gantry or something like this...
With your Sender (downloaded Version 15), I can move and I see the values. Thats not the case with bCNC. I also startet the UGS Platform, here works everything fine. So I think it's not your fault / problem.  

I really think, I have to go with another Sender than bCNC, and I'll have a closer look to your sender 👍 
Your Updates look good, and I will post some thoughts for individualisiation in the thread there :-)

Like I said, thank you so muc for your superb support 🥇

closed 2020-05-13 by einencool