terjeio/grblHAL #29: SAMD21 - diskio.h?

phil-barrett, 2020-03-15:
Trying to build the SAMD21 version for MKRZERO.  Get diskio.h missing error.  Search of tree shows no such file.

phil-barrett, 2020-03-15:
By the way, here's the MKRZERO Board.  All the I/O checks out so hoping to run grbl/HAL on it soon. [image]

terjeio, 2020-03-15:
Try with deleting disko.c and setting SDCARD_ENABLE to 0 in driver.h. Will commit a fix later, I need to finish an update to tsender first - getting closer to a beta release.

Your boards looks great!

phil-barrett, 2020-03-15:
Removing the diskio.c file did it - stupid arduino assumes any file in directory needs compiling. No other edits - driver.h was fine.  It seems to run OK (at least TSender comes up in an alarm state as expected) but I need to wire up the inputs and test.  I don't expect any surprises.

Looking forward to tsender beta. 

By the way, from here on out, I'm calling the boards motion controllers because that is what they are. We should talk about "branding" and such - I want to be consistent with the general grbl/HAL naming and terminology. If this is going to take off, we should be in sync.

Ciao from Covid-19 ground zero, USA.  Stay safe!

phil-barrett, 2020-03-15:
Looking good.  Running concentric circles as arcs now.  Aggressive buffering really gets ahead on the SAMD21.

phil-barrett, 2020-03-16:
concentric circles as arcs fr5000 completed in 7:31:09 on the MKRZERO.  8 minutes slower than on the Teensy 4.0 version.

terjeio, 2020-03-16:
Arduino behaves like most other IDEs regarding which files to compile, some that I am using has an option for flagging which to exclude. But still a stupid IDE...

In my notes I see I have clocked Due, MKRZERO and T4 to within 30 seconds of each other for your test file. Running new tests for Due and T4 now to make sure.

I needed to modify the Due driver slightly for it to behave correctly with native USB streaming, one test file had random pauses of a few minutes each - possibly from the main stepper timer losing interrupts.

terjeio, 2020-03-16:
Naming/terminology can be tricky. Are you sure you want to call your boards motion controllers? I think the correct term is breakout board, this since the T4 or MKRZERO is the actual controller. If you solder a processor directly to the board then it becomes a controller. Maybe I am wrong but this is how I am thinking...

Also, I see you are using grbl/HAL as name for my port, I am calling it grblHAL. Should I change too?

phil-barrett, 2020-03-16:
My thinking is that the package is a motion controller - processor+grblHAL+connectors/interfaces. I agree that the heart is the processor+grblHAL.  Though, pretty much useless without the breakout portion. Not sure how important it is to split that hair, though.  I am recommending that people actually solder the Teensy to the board for reliability, by the way.  I'll think about it some. Not trying to over promise.

I think I've seen/used grbl/HAL and GrblHAL and grbl-HAL.  I'm not set on any one particular name but rather on consistent use. I defer to you on this because, well, you are the man! People get confused seeing different names for the same thing so consistency is paramount.  Using lower case grbl is different from the "official" upper case GRBL. But GRBLHAL seems like shouting. The use of lower case softens it.  A lower case name makes starting a sentence with it look visually odd to me.  

If this was a company, we'd probably do focus groups and test a number of options and waste, er invest, a lot of money on a name...

Unless you think differently, I will change everything to grblHAL and use that going forward. Not sure if I can change the github name, though.

closed 2020-06-19 by terjeio