terjeio/grblHAL #30: Did you have plan to add G5 for grbl?

yuyue2013, 2020-03-17:
Did you have plan to add G5? I find linuxcnc support it.

terjeio, 2020-03-17:
Looked around a bit for code, should not be too hard to add. Are you willing to test?

Teilax, 2020-03-19:
http://linuxcnc.org/docs/2.6/html/gcode/gcode.html

terjeio, 2020-03-19:
@Teilax Modified code from the linuxCNC doc:

```
G90 G17
G0 X0 Y0
G5 I0 J30 P0 Q-30 X10 Y10 F500
P0 Q-30 X20 Y20
```
Renders like this in [my sender](https://github.com/terjeio/Grbl-GCode-Sender):

[image]

and runs ok.

It would be nice if someone needing G5 support can help with real test files/testing.

Please note that the current code on gitHub has not been updated for this!

closed 2020-05-24 by terjeio

ithinkido, 2020-05-25:
Just saw this now, I would be interested in using G05 bezier curves for a pen plotter. Can you tell me what the current status is. I would like to use them in a pen plotter setup to reduce the number of segments and hopefully improve the plot speed. I have spoken to the maker of juicyGcode https://github.com/domoszlai/juicy-gcode and we will be happy to send a few test files with Bezier G05 curves to try out.

terjeio, 2020-05-25:
> Can you tell me what the current status is.

It is ready for you to test and has been for a while. G5 only!

ithinkido, 2020-05-25:
Is the G05 handled by your sender or on the GrblHAL boards? ( I could not find any mention of G05 in the supported commands list of GrblHAL)

terjeio, 2020-05-25:
> Is the G05 handled by your sender or on the GrblHAL boards? 

G5 is handled by both. However, I do not want to add G5 to the supported list yet because I have only one small file available for testing. Other senders _may_ be usable too but I do not believe any has support for graphical rendering of G5 paths.

It would be nice if you can source test files, test and report back.

ithinkido, 2020-05-25:
[Bezier_test.zip](https://github.com/terjeio/grblHAL/files/4677272/Bezier_test.zip)
I have some files to test on. I have tried them in the sender , but there seems to be a  error. I do not see any mistakes at first glance, could you tell me what needs to changed

terjeio, 2020-05-25:
1. G4 P2000.000000 is a half hour delay (2000 seconds), you really want that?
2. Feedrate is slow at F10?
3. Grbl do not accept negative P-values.

The rendering in the sender seems to be ok, but is upside down?

A quick fix for 3. is to remove the check for negative values:

https://github.com/terjeio/grblHAL/blob/73604aa13c0b748a760d6f2e49c77830f38ce4f3/grbl/gcode.c#L756

Remove `bit(Word_P)` from the line above:

`if ((bit(word_bit.parameter) & (bit(Word_D)|bit(Word_F)|bit(Word_H)|bit(Word_N)|bit(Word_T)|bit(Word_S))) && value < 0.0f)`

and voila - it seems to be working :-)

I need to check for negative P-values elsewhere in the gcode parser, so I am not yet ready to commit a fix. However, the above change should enable you to start testing.

ithinkido, 2020-05-25:
Ok, I will try that out in the next days. One thing I am still a bit confused by , when I tried to open the file in the sender to take a look at the simulation, I was getting an error on every G05 ( probably the neg values) how did you load the file into the sender?

terjeio, 2020-05-25:
> how did you load the file into the sender?

I drag&drop into the program list or 3D View.

Is [COMPATIBILITY_LEVEL](https://github.com/terjeio/grblHAL/wiki/Changes-from-grbl-1.1#workaround) set to > 0 in your build? If so, that is a likely reason it won't load.

I changed the feed rate to 300 and your file is still running here, at 1:56:25 now and about 3/4 finished:

[image]

ithinkido, 2020-05-25:
Ahh.....TBH, I am strugling with the ESP IDF , I had opened the sender on a stock esp32-grbl ( I should have guessed that there would be compatabilty issues) I thought the sender would at read the file but have issues when sending to the stock esp32-grbl . As soon as I have managed to get a working grblHAL build flashed I will givit a try. 
From what it sounds like though, this is just what I was looking for to try out bezier curves. It would be great to have a way so improve the motion on such curvy plotts. The current slow down is not ideal, ( although not that bad either ) and with a pen plot you can go faster than with a mill....

terjeio, 2020-05-25:
> this plot is almost hypnotic to watch in real life !

Ok, I guess I have to try to plot it when I get my plotter operational. I 've got a C.ITOH CX-6000 for which I will swap the electronics with one of my grblHAL drivers/boards. One rainy day...

ithinkido, 2020-05-25:
Is the wall plotter project plans still actual?

terjeio, 2020-05-25:
> dc motors with encoders

Not supported yet - but I have tested closed loop control with some old printer parts, it seems to be doable with the current HAL interface. Perhaps I will build a Maslow router too - I have a partly finished driver for that also.

> Is the wall plotter project plans still actual?

Yep - still is. But I have too many unfinished projects so it may take some time... At least the lathe CNC conversion is nearing completion now.

ithinkido, 2020-05-25:
>
> At least the lathe CNC conversion is nearing completion now.
>
Thread milling is going to fun, in the video it looks great. NIce work.

ithinkido, 2020-05-26:
OK, I have the board setup and the sender is simulating correctly. I will run a few more tests, but it looks perfect. 
>
>The rendering in the sender seems to be ok, but is upside down?
>
This was my mistake, and is easy to fix.