terjeio/grblHAL #32: GPIO_MAP not implemented

shaise, 2020-03-24:
@terjeio Hi and thank you for this excellent HAL-ed port!
I have decided to move to your port after [The discussion here](https://github.com/gnea/grbl/issues/832)  
There are some minor bugs I fixed but I think the one in the title should be fixed at your port as well.  
I can not issue a PR as I made some custom changes. But [Here is the commit](https://github.com/shaise/grblHAL_CNC3040/commit/c14661a29802554494bfb65bffe916e7ad9017bf) That made this option work.  
By the way, the Jog problem exists in this port as well, so I plan to fix it as well.  
I plan to add some more features so it will work with my LinuxCNC clone for windows. (To be published soon)
Thanks again!
shai

terjeio, 2020-03-24:
Thanks for the comment about the port. I will add your patch to the STM32F1xx driver, thanks for that too. I do a bit of copy/paste when writing a new driver and sometimes I stop when the basics are working - the end result beeing some may be a bit lacking in features.

I am curious about the the jog problem still existing in grblHAL - it has proven to be robust for me, at least with [my own sender](https://github.com/terjeio/Grbl-GCode-Sender), for a long time. You get the same intermittent failure as for vanilla grbl?

Looking forward to check out your LinuxCNC clone.

shaise, 2020-03-24:
@terjeio 
Thank you for your response. See the issue [Here](https://youtu.be/nzQHpMUJlKo)
Hard to see exactly but when I stop pressing on the button, the jogs slows to a stop, then jumps several millimeters away. I believe its the same issue as the standard Grbl. I will try to apply the same fix and get back to you.
shai

terjeio, 2020-03-24:
I have never seen behaviour like that. Are you sending a lot of short distance commands continually or single long ones? I send long distance commands [to jog](https://github.com/terjeio/Grbl-GCode-Sender/wiki/Usage-tips) (except for step jogging) and cancel on key up.

It is definitely not the same issue as for vanilla grbl, this since I clear both the serial and block input buffers on jog cancel. Could be due to another rare edge case in grblHAL or issues in your sender?

shaise, 2020-03-24:
Hi, @terjeio 
Found the issue and fixed it. Indeed it will not cause any issues when issuing long steps then canceling. This issue happens ONLY when you issue many commands till the planner buffer is full. In this case it goes to busy waiting for the buffer to be free, and as soon as you clean everything, it exits the loop and initiates a new plan...
Its the same bug I fixed for Grbl. Fortunately the fix was simpler in your code. [See diff here](https://github.com/shaise/grblHAL_CNC3040/commit/33072f19373ba6358993dcf484e6a53db21c4878).
shai

shaise, 2020-04-11:
Thanks for the fix!

closed 2020-04-11 by shaise