terjeio/grblHAL #33: Feed Rate issues with STM32F1xx driver

shaise, 2020-03-27:
Hi,

When setting very low feed rates (F <= 20 mm/min) The step speed calculation goes wrong sending the CNC to very high/random speeds. (happens on a 400 steps/mm axis with AMASS = 3)
Solution: use stepperCyclesPerTickPrescaled always (instead of stepperCyclesPerTick) as even when AMASS is used, cycles_per_tick can be higher then 65535.

shai

terjeio, 2020-03-27:
Thanks, will have to look into this as I may have messed up code in the core (in stepper.c).

shaise, 2020-03-27:
From what I investigated, I think the stepper.c code is ok, Its just a very large (> 16bit) wait between pulses, and the timer is only 16 bit. 
Unless of course you planed to handle it with extra interrupts between pulses.

shaise, 2020-04-11:
Works great now, thanks!

closed 2020-04-11 by shaise