terjeio/grblHAL #34: Adding a pendant to grblHAL

phil-barrett, 2020-03-29:
Hello, I'm starting work on adding a pendant to grblHAL.  My current thinking is to use the keypad interface and a small micro to format up the pendant commands and hand them off the the grbl processor via I2C. 

I've been looking at the keypad code and don't understand how it gets run in the context of grbl. For example, looking at keypad_process_keypress, I don't see where it gets called anywhere in the grblHAL source code tree. Could you kindly point me to example code? Once I understand that, it should be fairly easy to format up grbl commands and push them into the queue.

terjeio, 2020-03-29:
It gets called indirectly via a call to `hal.execute_realtime`  in [grbl/protocol.c](https://github.com/terjeio/grblHAL/blob/cdab733cae58b598ecc849a01008a5157ad93bcd/grbl/protocol.c#L405). The `hal.execute_realtime` [function pointer](https://www.cprogramming.com/tutorial/function-pointers.html) is currently set in the Teensy 4 driver like this:

https://github.com/terjeio/grblHAL/blob/cdab733cae58b598ecc849a01008a5157ad93bcd/drivers/IMXRT1062/main/driver.c#L1048-L1050

I will change this in the next commit to be similar to the MKRZERO driver:

https://github.com/terjeio/grblHAL/blob/cdab733cae58b598ecc849a01008a5157ad93bcd/drivers/SAMD21/main/driver.c#L969-L979

and

https://github.com/terjeio/grblHAL/blob/cdab733cae58b598ecc849a01008a5157ad93bcd/drivers/SAMD21/main/driver.c#L1110-L1112

HuubBuis, 2020-07-01:
I am also thinking some time about making a pendant. My choice for the processor will be the esp32 because it is cheap, easy to program for novice users (when using arduino) and has serial, Bluetooth and WiFi on board. This will make it also possible to use Modbus to interface gui's like mach3 or even a VFD.
The pendant can send commands to the controller or gui (gcode sender). For me,  a command is just a short ascii string.
The gui can send information about the max speed, selected axis, jog step, etc to the pendant.
I have looked at the web gui for the grbl_esp32. This could be an easy to use gui to setup the pendant besides a setup like $x=yy.
A few buttons,  2 joysticks,  a rotary knob and a small display can do the job. This gives users freedom of choice for the hardware to use.

terjeio, 2020-07-02:
You may want to take a look at my [MPG/DRO](https://github.com/terjeio/GRBL_MPG_DRO_BoosterPack) for code/ideas. Some drivers already has support for the secondary input stream and MPG mode switch input required.

closed 2022-01-27 by terjeio