terjeio/grblHAL #35: CMD_STOP and spindle reporting

shaise, 2020-04-11:
Hi,

I'm trying to use CMD_STOP to stop gcode run job.
First of all it abruptly stops all motors, so I issue a pause, wait for motors to stop, then issue CMD_STOP

When i issue the CMD_STOP, indeed everything is cleared and the Spindle physically stops moving, however the status still reports the spindle turning ("FS:0,10000")
Is this the intended behavior? 

shai

terjeio, 2020-04-11:
Certainly not intended - fix commited: f5588af

By issuing a pause you mean a feed hold command? Some senders just stop sending when pressing a pause button and waits for the motion to stop. I do not like that...

shaise, 2020-04-11:
Yes, Pause = feed hold. It stops immediately, but uses deceleration as opposed to abruptly stopping, which might cause a step slip.
And thanks for the fix!

closed 2020-04-15 by terjeio