terjeio/grblHAL #36: softlimit deaktivate only for the rotationsaxes, a-achses

S2000Stefan, 2020-04-15:
Hello terjeio,

it is possible when the soft limit $20=1 is switched on that the rotationaxis (a-axis) 
is deactivated by entering 0 in $133=0?
The background is that I mostly work in a helical milling. Machine in one direction with a lot of rotations and that is not possible with soft limits on. But like to use soft limits in xyz  ;)

BTW i cannot invert the probe pin $6=0/1.

Greeting
Stefan

terjeio, 2020-04-15:
It should be possible, will have to look into that.

Which driver are you using? I need to know since pin inversion is not done in the common core.

S2000Stefan, 2020-04-15:
Thank you.
My driver is: STM32F1xx Bluepill

terjeio, 2020-04-15:
Fix committed for driver.
Added skip of 0 max travel axes in soft limits check.
I have a TODO note in the source about adding check for homed axes as well, I think I should add that too since grblHAL keeps track of which axes are homed. May I ask if you home the A-axis?

S2000Stefan, 2020-04-16:
Many many Thanks,
Although I was only able to do short tests today, everything is working as desired. :)
No i don't home the A-axis, see no sense in it for me.
Many thanks again.
Cheers
Stefan

closed 2020-05-01 by S2000Stefan