terjeio/grblHAL #37: STM32F103 Version with only 64kb flash supported?

hdo, 2020-04-21:
I'm wondering why the STM32F1 version does need 128kb of flash. Most of the blue pill boards i own only have 64kb flash. Also most of the blue bill boards on ali**** have 64kb flash.

Is it possible to size that down?

I noticed that @sheisa's fork support the blue pill board with only 64kb flash.

Thanks.

terjeio, 2020-04-21:
I managed to reduce the size down a quite a bit by disabling the USB serial code, SD Card support etc. and compile optimized for size:
```
arm-none-eabi-size   GRBL\ Driver\ STM32F103C8.elf 
   text	   data	    bss	    dec	    hex	filename
  66676	    140	  10952	  77768	  12fc8	GRBL Driver STM32F103C8.elf
Finished building: default.size.stdout
```
Still over 1000 bytes too long...

I guess you need to remove code from both the driver and the core grbl to make it fit into 64kb. I am not too keen to do that, as it shouldn't be to hard to get a board that has 128kb flash.

BTW, have you tried flashing your boards already? The few I have has 128kb even though they are marked as C8T6 variants.

The fork you mentioned is @shaise's [fork of grblHAL](https://github.com/shaise/grblHAL_CNC3040) or the other [vanilla grbl fork](https://github.com/shaise/grbl_cnc3040)? The latter may well fit into 64kb, grblHAL not so because there is quite a few of additional G- and M-codes supported as well as many other changes.

shaise, 2020-04-21:
I am using a 64K STM32F103 original from ST, and have no issues to flash code larger then 64K. The word around says, it the same chip as the 128K, just not fully tested.

hdo, 2020-04-23:
@terjeio @shaise 

Thank you both for your feedback. To be honest i havent' tried it yet.

I have a couple of blue pills lying around so will try that ;-)

hdo, 2020-04-27:
Update:

I flashed grblHAL on my blue pill board with STM32F103C8T6 without any issues ;-)

Haven't tried it on my machine though.

hdo, 2020-04-28:
@terjeio 

Would you mind sharing your '.ioc' project file for the STM32 driver?

I'm about doing a port to Nucleo STM32F411RE and the '.ioc' would be helpful
as a template.

Thanks.

terjeio, 2020-04-28:
@hdo : download it from [here](http://svn.io-engineering.com/GRBL%20Driver%20STM32F103C8.ioc). I'll leave this link active only for a short time - please post a comment when downloaded.

NOTE: do not use this to rebuild the STM32 driver as the generated files has been modified by me, and is the reason I did not provide it as a part of the driver.

I hope you will share your driver when finished - it will be the first I know of written by somebody else than me.

hdo, 2020-04-29:
@terjeio Thanks!

No worries i won't use this to generate the code only for reference ;-)

I'm preparing a little port for STM32 Nucleo with STM32F11RE and later on for 
the black pill once i got it here.

Of course once the port is done i'll submit a pull request for integration.

hdo

closed 2020-04-29 by hdo