terjeio/grblHAL #39: STM32F103 GPIO_MAP broken again

shaise, 2020-04-23:
Hi @terjeio 
GPIO_MAP is still broken (and causes memory overrun as well)
problems in driver.c
Line 97 : missing "B_STEP_BIT |" at the beginning
Line 511 and 519, change uint16_t to uint32_t. (or change the type of step_outmap and dir_outmap)
Thanks a lot!

shai

terjeio, 2020-04-23:
Line 124 was incorrect too - copied from your commit...
Changed all to uint32_t since the GPIO registers are 32-bit wide.

Commit aff8367 adds manual homing and parser status report, but slighly differently from your version. For the parser status report I decided against adding this to the real-time report as it requires fewer changes to the sender side if it is sent separately.

Comments appreciated.

shaise, 2020-04-23:
Hi @terjeio ,
Looks like you did lots of changes. Hopefully I will not need to hold a version of my own.
I will try to adapt to your manual homing system.
I did not understand though regarding the parser status. What do you mean by fewer changes? All i wanted is to save trafic, so instead of polling the status all the time, I get it automatically only when there are changes.
Thanks for the big support! Today was the first time I operated the CNC with grblHal and my sender
shai

shaise, 2020-04-23:
Oh! I see now. You trigger showing the standard $G parser state. Much better!!!

closed 2020-04-23 by shaise

terjeio, 2020-04-24:
> Hopefully I will not need to hold a version of my own.

If possible the grblHAL core should be kept as a common single version. The HAL drivers not necessarily so - but it would be nice to keep a single, but configurable version, of these too if the main difference is the pin mappings. For the STM32 driver I have added your pin mappings in a [separate file](https://github.com/terjeio/grblHAL/blob/master/drivers/STM32F1xx/Inc/cnc3040_map.h), and it is a matter of  removing a comment in [driver.h](https://github.com/terjeio/grblHAL/blob/master/drivers/STM32F1xx/Inc/cnc3040_map.h) (at line 38) to activate it.

For manual homing I have added the supporting code to motion_control.c:

https://github.com/terjeio/grblHAL/blob/aff83670ec0d4b68f0a86a5ebc4051a064999f7e/grbl/motion_control.c#L726-L736

This allows both automatic and manual homing to be active at the same time. Also, if only manual homing is enabled all axes can be homed with `$H`.

grblHAL already has a status message for homing status, `|H:<0|1>` or `|H<axis flags>` depending on the configuration:

https://github.com/terjeio/grblHAL/blob/aff83670ec0d4b68f0a86a5ebc4051a064999f7e/grbl/report.c#L1108-L1109

I am not entirely happy about doing it like this as it ambiguous, and I want the boolean when only automatic homing is enabled not the axisflags. IMO this could be solved by changing the homing status to `|H:<0|1>,<axis flags>`. I think this should remove the need for the `Hs:<axis flags>` status you added. Your opinion on this?

FYI some of the status elements are only sent on changes, this posed a problem for me when I started writing my sender and is complicated by the fact that many of my drivers (luckily) does not reboot when the sender connect. To work around this I have added a new real-time report command, `CMD_STATUS_REPORT_ALL 0x87`, that outputs the complete report. I request this on sender startup.

> Today was the first time I operated the CNC with grblHal and my sender

:+1:

shaise, 2020-04-24:
@terjeio,
I have already removed my Hs system, and also removed the in-realtime-status Parser state. In fact (and I need to double check) it looks like I can work directly with you latest version!
Regarding home reporting I am entirely in favor of `|H:<0|1>,<axis flags>`. It can also help with Issue #40 suggestion.
Thanks again!
shai

shaise, 2020-04-24:
oh and `CMD_STATUS_REPORT_ALL 0x87` is great, I really needed that...

shaise, 2020-04-24:
I have reforked grblHAL, to start clean. Indeed almost all i needed to change was the inclusion of the CNC_3040 map file, however, I also had to re-add the jog slipping issue fix. The motors do stop when i lift the finger from the jog key, but the location registration skips several millimeters. There is also a small homing issue which I'm looking into now.

terjeio, 2020-04-24:
The jog slipping issue you see is that the same you [made a video](https://www.youtube.com/watch?v=nzQHpMUJlKo&feature=youtu.be) about? I do not see that with my sender and neither yours so perhaps something different?

shaise, 2020-04-24:
yes indeed. here is the parameters for x axis I use:
[image]
I use 1200mm/m jog speed.
It not happening all the time but quite often. move X to ~ +30mm using on screen button, then move back about -40mm and release the button. I guess its not happening in your sender as you jog differently (I think you use long segment jogs)
When I managed to debug the issue, it looked from the code that indeed such thing can happen.

shaise, 2020-04-24:
Ohh...
It looks like its happening with the fix as well.... let me look into it.

shaise, 2020-04-24:
OK, Forget it. Its some crazy thing with report rate. most likely not on grbl side.

shaise, 2020-04-24:
This is one of the craziest things I ever saw. after I jog quite a bit until i see the problem. I just move the y axis to -10 (in local coordinates) and from this moment on, I get status report only after 10 times sending the '?' (and i get all 10 reports together). I can reproduce it in terminal window, its not sender issue. And as soon as Y goes above -10 say -9.5, everything is back to normal. It can not be grbl, as it has a short output buffer, and certainly can not hold 10 messages. Have no idea how to approach this. This is why you see the jump in the video. because the report messages were held.