terjeio/grblHAL #41: Issue with home state reporting.

shaise, 2020-04-24:
I keep get H:1 even if nothing is homed yet.
Report.c:1109 - perhaps it should be "== sys.homing.mask" instead of "== sys.homed.mask" ?

shai.

terjeio, 2020-04-24:
Looking into this again, and still arguing with myself...

Try with changing

https://github.com/terjeio/grblHAL/blob/aff83670ec0d4b68f0a86a5ebc4051a064999f7e/grbl/report.c#L1109

to

` hal.stream.write_all(appendbuf(4, "|H:", sys.homing.mask != 0 && (sys.homing.mask & sys.homed.mask) == sys.homing.mask ? "1" : "0", ",", uitoa(sys.homed.mask))); `

If only manual homing is enabled and all axes are homed (`sys.homed.mask == AXES_BITMASK`) should the first parameter be set to `1`?

Also
https://github.com/terjeio/grblHAL/blob/aff83670ec0d4b68f0a86a5ebc4051a064999f7e/grbl/motion_control.c#L728

should be
```
    if(settings.homing.flags.manual && (cycle.mask == 0 ? sys.homing.mask == 0 : (cycle.mask ^ sys.homing.mask))) {
```

And perhaps change line 
https://github.com/terjeio/grblHAL/blob/aff83670ec0d4b68f0a86a5ebc4051a064999f7e/grbl/motion_control.c#L770
to
`            sys.homed.mask &= ~sys.homing.mask;`

to only clear homed status for automatically homed axes on `$H`...

shaise, 2020-04-24:
Regarding reporting, I think is should be as simple as H:<1|0> and if "Single axis homing" is selected, add the ",<mask>"
Because even if you are in manual homing, if single axis is not selected yo can not home single axis. Furthermore if you are in automatic mode, but still use the single axis option, it will be nice to know which axis are not homed yet (as issue #40)
And yes, It will be nice if in manual mode all axis are homed you will get "H:1" as well.
Thanks!
shai

terjeio, 2020-04-24:
Ok - you want the second parameter inverted - like this:
```
        if(sys.report.homed && (sys.homing.mask || settings.homing.flags.manual)) {
            axes_signals_t homing = {sys.homing.mask ? sys.homing.mask : AXES_BITMASK};
            hal.stream.write_all(appendbuf(2, "|H:", (homing.mask & sys.homed.mask) == homing.mask ? "1" : "0"));
            if(settings.homing.flags.manual && AXES_BITMASK ^ sys.homed.mask)
                hal.stream.write_all(appendbuf(2, ",", uitoa(AXES_BITMASK ^ sys.homed.mask)));
        }
```
or this:
```
        if(sys.report.homed && (sys.homing.mask || settings.homing.flags.manual)) {
            axes_signals_t homing = {sys.homing.mask ? sys.homing.mask : AXES_BITMASK};
            hal.stream.write_all(appendbuf(2, "|H:", (homing.mask & sys.homed.mask) == homing.mask ? "1" : "0"));
            if(settings.homing.flags.manual)
                hal.stream.write_all(appendbuf(2, ",", uitoa(AXES_BITMASK ^ sys.homed.mask)));
        }
```
?

shaise, 2020-04-24:
for me, the best will be if the last 2 lines are:
```
if(settings.homing.flags.single_axis_commands)
    hal.stream.write_all(appendbuf(2, ",", uitoa(sys.homed.mask)));
```
But if you think it will cause compatibility issues, any other option is good.

terjeio, 2020-04-24:
No compatibility issues that I am aware of as this is a grblHAL extension. Since you wrote

> which axis is not homed yet

then using `uitoa(AXES_BITMASK ^ sys.homed.mask)`  will report the _not_ homed axes.
Perhaps this is easier to handle than a value for homed axes? 0 or no value means all axes homed...

Note that grblHAL returns the number of axes defined as part of the `$I` response, e.g:

[VER:1.1f(MSP432).20200416:]
[OPT:VNMZHSL+,35,1024,_**3**_,8]

shaise, 2020-04-24:
No  - for me it is better to get the home status and not the xor one. Sorry for way I said it.
Regarding the compatibility, its because i wanted to use the `flags.single_axis_commands` instead of the `flags.manual` in the if statement, as this is an existing flag.

And regarding the step  - or rather - report skipping, this is absolutely amazing  - If the message is exactly 64 bytes, it gets buffered and not sent. don't know if its the usb-serial issue or windows driver issue. anyway as a horrible hack, I add extra char to any message that is 64 byte long, now everything works flawlessly...

terjeio, 2020-04-25:
Ok, I will use your version.

As for the 64 bytes issue - perhaps checking for a multiple of 64 bytes is needed/should be done?

shaise, 2020-04-25:
yes, of course, on multiples as well, but I saw you already did so... Thanks again!

shaise, 2020-04-26:
Motion_control.c line 728, you suggested to change to:
`if(settings.homing.flags.manual && (cycle.mask == 0 ? sys.homing.mask == 0 : (cycle.mask ^ sys.homing.mask))) {`
I think maybe it should be: `cycle.mask & sys.homing.mask` ? (`&` instead of `^') otherwise if say X and Y are automatic, and you stage a `$HX`, it will manually set X to 0.

terjeio, 2020-04-27:
This seems to work, that is allow automatic homing of single axes configured for that:

`if(settings.homing.flags.manual && (cycle.mask == 0 ? sys.homing.mask : (cycle.mask & sys.homing.mask)) == 0)`

If `$H<axis>` should always set the axis manually then my suggested version works. Is that more useful? Perhaps we need another settings flag or an `$H` extension? Or not?

shaise, 2020-04-27:
In your version if there is only one axis in the cycle.homing.mask, (say X) then $HX will still do an automatic home, even if set axis manually is selected. if there are more then one (say X and Z) then it will set X manually. So it is not consistent. That's why I suggested the second version.
Do you plan to commit this to your repository, or is it only for my version?

terjeio, 2020-04-27:
I'll commit when we are in agreement about the solution.

`if(settings.homing.flags.manual && (cycle.mask == 0 ? sys.homing.mask : (cycle.mask & sys.homing.mask)) == 0)`

 seems to be the same as your version but with one less compare?

Your version is:
`if(settings.homing.flags.manual && (cycle.mask == 0 ? sys.homing.mask == 0 : (cycle.mask & sys.homing.mask) == 0))`

I did not read your version correctly at first because the modification was out of context - I am an old hand and thus a bit slow sometimes...

terjeio, 2020-04-27:
This description is then correct?

[image]

I need to update the [advanced settings file](https://github.com/terjeio/grblHAL/blob/master/doc/csv/hal_setting_codes_en_US.txt), it can be used instead of hardcoding settings specs in the sender. However, the format should be changed (to xml or json) and grouping added. No need to update the sender on setting changes if loaded at startup and used for configuring the UI.

shaise, 2020-04-27:
OK, I think I also missed something, I was referring to the ^ version... 
Anyway here is what i think (maybe its simpler), let me know your thoughts:
1. manual selected and home all (mask = 0): auto home all axis in homingmask, manually home all others.
2. manual selected, home specific axis: if this axis in homing mask, auto home it, otherwise manual home it.
3. manual not selected: only auto home-s allowed, homing others will report error.

Regarding description I would separate between single axis, and manual:

`When Enable single axis commands is checked, single axis homing can be performed by $H<axis letter> command`

`When allow manual is enabled, axes not using limit switches may be homed manually by $H or $H<axis>` 

BTW I also use a [setting description file](https://github.com/shaise/GrblCNC/blob/master/Software/GrblCNC/Resources/GrblParamDescription.txt) to generate the GUI. Perhaps I should use yours...

shaise, 2020-05-03:
Here is what I finally did in my version:
Instead of:
```
bool home_all = cycle.mask == 0;

if(settings.homing.flags.manual && (home_all ? sys.homing.mask : (cycle.mask & sys.homing.mask)) == 0) {

    if(home_all)
        cycle.mask = AXES_BITMASK;

    sys.homed.mask |= cycle.mask;
    limits_set_machine_positions(cycle, false);
} else {
```
It is now:
```	
if(settings.homing.flags.manual)
{
    axes_signals_t manual_cycle = cycle;
    //&& (cycle.mask == 0 ? sys.homing.mask == 0 : ((cycle.mask & sys.homing.mask) == 0)))

    if(manual_cycle.mask == 0)
        manual_cycle.mask = AXES_BITMASK;
    manual_cycle.mask &= ~sys.homing.mask;
    if (manual_cycle.mask != 0)
    {
        sys.homed.mask |= manual_cycle.mask;
        limits_set_machine_positions(manual_cycle, false);
    }
}

if (cycle.mask == 0 || (cycle.mask & sys.homing.mask) != 0)
{
    cycle.mask &= sys.homing.mask;
````

This way manual / automatic homing can work side by side. If an axis is in the homing mask, it will be homed automatically. If not, it will be homed manually. $H will home all homing mask axis automatically, and all others manually.

Let me know what you think. Thanks!

terjeio, 2020-05-03:
IMO same as my version, but I have delayed manual homing of the remaining axes until after error checking for automatic has been done...

shaise, 2020-05-03:
Sounds good to me.
Is your version commited, can i try it?

terjeio, 2020-05-03:
Yes, it is committed.

shaise, 2020-05-03:
Works exactly like advertised! 
Thanks! (sorry for not noticing the commit)
regarding the safety door disabled by default, I had to change the config.h file to re-enable it. Maybe it should be a define in the map file? so no need to touch grbl directory?

terjeio, 2020-05-03:
Some options may  be added to the compiler command line, eg:

[image]

 No need to touch the sources...

shaise, 2020-05-03:
Ahh! Great idea, thanks!

closed 2020-05-04 by shaise