terjeio/grblHAL #42: STM32FXX USB to Serial problem with 64 byte messages

shaise, 2020-04-24:
Apparently this is a real issue!
64 byte messages are exactly the size of the usb endpoint. On the PC side if a packet is 64 bytes in length, it assumes this packet is a part of a bigger message, and will not pass the packet on, until a shorter then 64 bytes packet arrives.
This must be fixed in the grbl side. when a message is 64 byte or multiple of it in length, a zero message must be sent after it to close the packet. 
[See explanation here](https://community.cypress.com/docs/DOC-10941)
Let me know if you want me to try and fix it.

terjeio, 2020-04-25:
This seems to fix it:

```
//
// Writes a null terminated string to the USB output stream, blocks if buffer full
// Buffers string up to EOL (LF) before transmitting
//
void usbWriteS (const char *s)
{
    size_t length = strlen(s);

    if(length + txbuf.length < USB_TXLEN) {
        memcpy(txbuf.s, s, length);
        txbuf.length += length;
        txbuf.s += length;
        if(s[length - 1] == '\n') {
            length = txbuf.length;
            txbuf.length = 0;
            txbuf.s = txbuf.data;
            while(CDC_Transmit_FS((uint8_t *)txbuf.data, length) == USBD_BUSY) {
                if(!hal.stream_blocking_callback())
                    return;
            }
            if(length == 64)
            	CDC_Transmit_FS((uint8_t *)txbuf.data, 0);
        }
    }
//  while(CDC_Transmit_FS((uint8_t*)s, strlen(s)) == USBD_BUSY);
}
```

I have never seen this behaviour before, so tested with MSP432 and Teensy 4 - they do not fail. Will have to check all the other drivers offering a USB option too. However, since MSP432 and Teensy 4 do not fail it seems to me that this is a flaw in the STM32 USB code - IMO this should have been handled at a lower level...

MSP432 has an onboard serial > USB converter so no local buffering in grblHAL, for Teensy 4 I do similar local buffering as for the STM32.

terjeio, 2020-04-25:
Oops - the debugger masked a problem, with this change it seems too work.

```
            if(length % 64 == 0) {
            	while(CDC_Transmit_FS((uint8_t *)txbuf.data, 0) == USBD_BUSY) {
                    if(!hal.stream_blocking_callback())
                        return;
                }
            }
```

shaise, 2020-04-25:
Yap, that was my plan too. And it definitely solves the problem, no more "jog slips", sorry for the initial wild goose chase...

And I definitely agree it should have been done by STM in the lower level. 
Thank you for your great support!
shai

shaise, 2020-04-26:
Works good. Thanks.

closed 2020-04-26 by shaise