terjeio/grblHAL #44: Issue with Feed Hold, Cycle Start

S2000Stefan, 2020-05-01:
Hello terjeio,
yesterday I noticed the following after real tests on my milling machine with your programs.
When I press the feed hold and cycle start buttons in the program window, everything is OK.
Refers to spindle on, off.
But if I press my feed hold switch on the milling machine and switch off the spindle and then press my cycle start button on the milling machine, the milling program starts again without waiting for the spindle start.
The fact that you can home the axes manually is a very good thing if you have 4 axis (a-axis). I usually mill with a helix with a lot of rotations in one direction, so I can finally use g28 and g30 commands.    ;)
Thanks a lot for this.
Cheers
Stefan

terjeio, 2020-05-01:
Signalling works as expected for me, perhaps your $63 setting is not correct?

[image]

When restore is enabled there will be a few seconds delay before program starts up again.

S2000Stefan, 2020-05-02:
You misunderstood me, sorry for my bad english. 
Signalling works and $63 settings are correct but when i press feed hold and start in the program window everything is ok, the spindle starts and the program with a delay.
If I stop and start the milling program with a button on the signal pins, the spindle starts but the milling program continues directly without delay for the spindel.

terjeio, 2020-05-02:
Strange, both works in the same way for me. And should so because either way just sets the cycle start flag to be picked up by the foreground process that initiates the restore. Can you post your grbl settings? In my sender select _File > About_ and _To clipboard_ to get a copy that can be pasted into a comment.

BTW: I see now that the flag is set differently by the physical signal, but that should not matter. I'll change that (to "atomic" mode) since ARM processors supports nested interrupts unlike 8-bit Arduinos.

terjeio, 2020-05-02:
Oh, I guess I found it - if you have a "bouncy" switch that triggers the cycle start signal more than once it will restart the program prematurely. In itself easy to work around but what if a feed hold is issued during the delay phase...

I'll sleep on that.

S2000Stefan, 2020-05-03:
Yes i have "bouncy" switches for feed hold and start that is correctly. Maybe I can change something on the hardware side, solder a capacitor?
It's not a bad mistake, who turns off the spindle at a feedhold, I just noticed it.

Here my grbl settings if it still helps:
%
; grblHAL
; 1.1f(STM32F103C8).20200416
; [OPT:VNMHTS+,35,1024,4,0]
; [NEWOPT:TC]
 ;[DRIVER VERSION:200417]
;
$0=4
$1=25
$2=0
$3=4
$4=0
$5=15
$6=1
$10=1023
$11=0.010
$12=0.002
$13=0
$14=15
$15=1
$16=0
$17=0
$18=0
$19=0
$20=1
$21=1
$22=23
$23=0
$24=20.000
$25=400.000
$26=25
$27=2.000
$28=0.100
$29=0
$30=24000.000
$31=0.000
$32=0
$33=5000
$34=0.0
$35=0.0
$36=100.0
$37=0
$39=1
$40=1
$41=0
$42=2
$43=1
$44=4
$45=3
$46=0
$47=0
$56=5.0
$57=100.0
$58=-5.0
$59=500.0
$60=1
$61=0
$62=0
$63=2
$64=0
$65=0
$100=640.000
$101=640.000
$102=640.000
$103=66.667
$110=3500.000
$111=3500.000
$112=3500.000
$113=18000.000
$120=500.000
$121=500.000
$122=500.000
$123=3500.000
$130=513.000
$131=539.000
$132=160.000
$133=0.000
%

Thank you for your fantastic work.

terjeio, 2020-05-03:
I have fixed this in code as the issue is a bug, will commit a change soon. This will handle a feed hold issued during the delay as well.

S2000Stefan, 2020-05-04:
Thanks for the fast support!
Works wonderful.

closed 2020-05-04 by S2000Stefan