terjeio/grblHAL #46: Homing do not finde Endstop Z & Y

Korowah, 2020-05-10:
hey
I have used and tested an Arduino Due. So far so good. I can move the Jogen and Manuel axes. the endstops are also recognized and trigger an alarm.
if I want to activate the homing function do I have to do this in the default.h? because it is not activated via the UGS software.

if he then has to drive the homing cycle, the z axis moves 2 x 5 mm away from the end stop and stops there then he drives the y axis 2x 5 mm away from the end stop and also stops. only the X axis searches for the end stop in the correct direction and finds the homing cycle correctly. after that the machine is idle without an error message although the z and y axes are nowhere in the middle of nowhere.

is this a bug or do I overlook a setting?

my setting:

$0 | 10 | Step pulse time
-- | -- | --
$1 | 25 | Step idle delay
$2 | 0 | Step pulse invert
$3 | 0 | Step direction invert
$4 | 7 | Invert step enable pin
$5 | 7 | Invert limit pins
$6 | 1 | Invert probe pin
$10 | 382 | Status report options
$11 | 0.010 | Junction deviation
$12 | 0.002 | Arc tolerance
$13 | 0 | Report in inches
$14 | 0 |  
$15 | 0 |  
$16 | 0 |  
$17 | 0 |  
$18 | 0 |  
$19 | 0 |  
$20 | 1 | Soft limits enable
$21 | 1 | Hard limits enable
$22 | 1 | Homing cycle enable
$23 | 0 | Homing direction invert
$24 | 25.000 | Homing locate feed rate
$25 | 500.000 | Homing search seek rate
$26 | 250 | Homing switch debounce delay
$27 | 1.000 | Homing switch pull-off distance
$28 | 0.100 |  
$29 | 0 |  
$30 | 1000. | Maximum spindle speed
$31 | 0. | Minimum spindle speed
$32 | 0 | Laser-mode enable
$33 | 5000.000 |  
$34 | 0.000 |  
$35 | 0.000 |  
$36 | 100.000 |  
$37 | 0 |  
$39 | 1 |  
$40 | 0 |  
$41 | 0 |  
$42 | 2 |  
$43 | 2 |  
$44 | 4 |  
$45 | 2 |  
$46 | 0 |  
$56 | 5.000 |  
$57 | 100.000 |  
$58 | -5.000 |  
$59 | 500.000 |  
$60 | 0 |  
$61 | 0 |  
$62 | 0 |  
$63 | 3 |  
$64 | 0 |  
$65 | 0 |  
$100 | 1600.000 | X-axis travel resolution
$101 | 1600.000 | Y-axis travel resolution
$102 | 255.100 | Z-axis travel resolution
$110 | 1000.000 | X-axis maximum rate
$111 | 1000.000 | Y-axis maximum rate
$112 | 1000.000 | Z-axis maximum rate
$120 | 10.000 | X-axis acceleration
$121 | 10.000 | Y-axis acceleration
$122 | 10.000 | Z-axis acceleration
$130 | 800.000 | X-axis maximum travel
$131 | 400.000 | Y-axis maximum travel
$132 | 90.000 | Z-axis maximum travel

Thanks for your Great work ;)

terjeio, 2020-05-10:
If an axis moves away from the limit switch then the homing direction needs to be changed, this can be done with the $23 setting.

grblHAL has quite a number of [new settings](https://github.com/terjeio/grblHAL/wiki/Additional-or-extended-settings) that most senders does not recognize. $44, $45 and $46 are for the homing masks - if you can use them there is no need to change defaults in code. Typical values are 4 for $44 (Z-axis first) , 3 for $45 (X- and Y-axis second) and 0 for $46 (no action).

If you are using Windows you may try [my sender](https://github.com/terjeio/Grbl-GCode-Sender) that supports the new settings, and other additional features of grblHAL. At least you may use it for the settings as it understands the new settings and presents all in the UI in (IMO) a more user-friendly way than most.

Korowah, 2020-05-10:
Yes, I've tried all of these settings, even if I invert the axes, I'm not interested in the homing cycle. he then drives in the right direction but not to the end stop .... and does not issue the error message that he cannot find it.

terjeio, 2020-05-10:
Ok, then I need a bit more info in order to check this out.
Which board map file are you using (may be enabled in [driver.h](https://github.com/terjeio/grblHAL/blob/master/drivers/SAM3X8E/main/driver.h) if not using the default map)?
Have you modified other files?
Are you using NO (normally open) or NC (normally closed limit switches)?
Do you have two limit switches per axis connected to separate input pins?

Yo also wrote "even if I invert the axes the homing cycle run in the right direction" - does this mean that inverting the axes does not change the direction?

I do not have a Due connected to a machine so my testing capabilities is a bit limited.

Korowah, 2020-05-10:
The driver is the SAM3X8E with the cmcgrath_rev3_map an no other change here. In the grblhal config.h I define:  
COMPATIBILITY_LEVEL = 1

all other files are untouched ;)

the Physical setup :

Arduino Due + DRV8825 
Endstops are Halleffect sensors with additional pullup (NO) 
i put an LED in parallel to the pull up so that i can verify if there are trigger ;) 
so i can say that the x axis produce an error if the homing cycle can't find the magnet.
the Y and Z axis try to find the magnet ... stops at time out .... and report a Homing done?!

so single Axis Homing goes well on the X Axis on, the Y an Z Finish in the middle of noway and report successfull Done.

terjeio, 2020-05-11:
Try this from a terminal (such as puTTY) with homing enabled for Y:
Send a ? in order to get a status report.
Send $H and wait for a result.
Send a ? in order to get a status report.

Post the result.

I have done this twice with homing enabled for Y, once with the limit switch left open, next with it left closed. First time takes relatively long time, second quickly and I get the following results:

```
GrblHAL 1.1f ['$' for help]
<Idle|WPos:0.000,0.000,0.000|Bf:35,1023|FS:0,0|Pn:P|Ov:100,100,100|Sc:>
<Home|WPos:0.000,0.000,0.000|Bf:35,1023|FS:0,0|Pn:P>
ALARM:9
ok

GrblHAL 1.1f ['$' for help]
[MSG:'$H'|'$X' to unlock]
<Alarm|WPos:0.000,300.000,0.000|Bf:35,1023|FS:0,0|Pn:P|Ov:100,100,100>
<Home|WPos:0.000,300.000,0.000|Bf:35,1023|FS:0,0|Pn:PY>
ALARM:8
ok

GrblHAL 1.1f ['$' for help]
[MSG:'$H'|'$X' to unlock]
<Alarm|WPos:0.000,-1.000,0.000|Bf:35,1023|FS:0,0|Pn:PY|Ov:100,100,100>
```
This is as expected. Note that grbl resets after a failed homing sequence which may cause senders to not display the error number.

If I do this one and briefely close the limit switch I get:

```
GrblHAL 1.1f ['$' for help]
<Idle|WPos:0.000,0.000,0.000|Bf:35,1023|FS:0,0|Pn:P|Ov:100,100,100|Sc:>
<Home|WPos:0.000,0.000,0.000|Bf:35,1023|FS:0,0|Pn:P>
ALARM:8
ok

GrblHAL 1.1f ['$' for help]
[MSG:'$H'|'$X' to unlock]
<Alarm|WPos:0.000,-1.000,0.000|Bf:35,1023|FS:0,0|Pn:PY|Ov:100,100,100>
```

Which is also as expected because it is very hard to simulate a homing sequence outside a machine.
If the homing move never reaches the limit switch and results in alarm 8 beeing issued I suspect electrical noise in your system is triggering the input.

You may see error 59 in your reports, this is a minor bug that I will commit a fix for soon. 

I wonder @norru has homing enabled on his machine and working as it should...

norru, 2020-05-11:
Yes, homing works perfectly on my machine. I had to configure the homing axis sequence though (z, then y, then x in my case).

Korowah, 2020-05-11:
So a night of hardware debuging brings the answer ;) it seems to be a problem with the due it self.
i figure out that:

if i drive the DRV8825 with 3.3v direkt on the due I can jog all axis normally and the homing for Z & Y fails with out errors.

if i drive the DRV8825 with 5v and level shifter everything gos well and as expected.

so i will deliver my putty results as soon as possible and will close this issues.

Thanks for your wonderful work it is awesome :)

closed 2020-05-11 by Korowah

norru, 2020-05-11:
> if i drive the DRV8825 with 5v and level shifter everything gos well and as expected.

I'm also driving my TB6600 with a level shifter- 3.3v didn't work even if the TB6600 vendor claimed it would. Maybe not enough total current available from the Due board?

Korowah, 2020-05-11:
the due deliver 180mA per output and 800mA at the 3.3vOut ...... i think this could be the problem.

but why it jog the axis as expected and fail only the homing? so with 3.3v i can drive all axis simultaneously .... only the homing cycle was the problem :/

terjeio, 2020-05-11:
> the due deliver 180mA per output

According to Table 45-2. DC Characteristics in the datasheet not so.
The max (in "relaxed mode" whatever that is) with VDDIO > 3V are: when sourcing -24/-9mA, when sinking 14/9mA - depending on the pin.

Korowah, 2020-05-11:
is my fault .... 130 mA total on all outputs....... :(

so there is still the question why there was no alarm message that he dosn´t find the stop in expected direction .....

Korowah, 2020-05-11:
Thats what i get..... 
`14:29:06.503 -> Grbl 1.1f ['$' for help]
14:29:06.503 -> [MSG:'$H'|'$X' to unlock]
14:29:50.022 -> 
14:29:50.022 -> Grbl 1.1f ['$' for help]
14:30:52.145 -> <Idle|WPos:0.000,0.000,0.000|Bf:35,1022|FS:0,0|Pn:X|Ov:100,100,100|Sc:>
14:30:52.145 -> error:2
14:31:09.997 -> <Home|WPos:0.000,0.000,0.000|Bf:35,1023|FS:0,0>
14:31:26.829 -> ALARM:8
14:31:27.327 -> error:59
14:31:27.327 -> 
14:31:27.327 -> Grbl 1.1f ['$' for help]
14:31:27.327 -> [MSG:'$H'|'$X' to unlock]
`

yes the x didn't homed because the 1mm pulloff wasn't big enough ^^ but she searched and find the ends top ..... Y&Z search and doesn't find :(

Korowah, 2020-05-11:
so that's with correct pulloff..... an i can't see an axis error :(

14:59:18.708 -> <Idle|WPos:0.000,0.000,0.000|Bf:35,1022|FS:0,0>
14:59:18.708 -> ok
14:59:21.454 -> <Idle|WPos:0.000,0.000,0.000|Bf:35,1022|FS:0,0>
14:59:21.454 -> error:2
14:59:31.573 -> <Home|WPos:0.000,0.000,0.000|Bf:35,1023|FS:0,0>
15:01:06.780 -> ok
15:01:18.159 -> <Idle|WPos:0.000,0.000,0.000|Bf:35,1022|FS:0,0|A:|H:1,7>
15:01:18.159 -> error:2

terjeio, 2020-05-11:
What is the output from a $N command? The error:2 after homing is complete at 15:01:18.159 may be from corrupted settings.

Korowah, 2020-05-11:
15:30:55.568 -> Grbl 1.1f ['$' for help]
15:31:02.687 -> <Idle|WPos:0.000,0.000,0.000|Bf:35,1022|FS:0,0|Ov:100,100,100|Sc:>
15:31:02.687 -> ok
15:31:07.897 -> $N0=
15:31:07.897 -> $N1=
15:31:07.897 -> ok
15:31:19.512 -> <Home|WPos:0.000,0.000,0.000|Bf:35,1023|FS:0,0>
15:32:54.912 -> ok
15:35:57.369 -> <Idle|WPos:0.000,0.000,0.000|Bf:35,1022|FS:0,0|A:|H:1,7>
15:35:57.369 -> error:2
15:36:01.717 -> $N0=
15:36:01.717 -> $N1=
15:36:01.717 -> ok

terjeio, 2020-05-12:
Strange, are you using the USB port for communication (`#define USB_SERIAL 1` in driver.h)? If so, can you try with 0 for communication over the programming port?
The last sequence you posted seems ok to me - if that failed I believe you have issues with electrical interference. Adding debug output may help to check for that.
E.g. add:

`if(limit_state) hal.stream.write("[MSG:limit trig]\r\n");`

at line 182 in limits.c to output a message when a switch is triggered.

Korowah, 2020-05-13:
The Serial  output was with `#define USB_SERIAL 0` 

So at the last days i called a friend who's professional working on ic´s an he explained that: 

at the moment the homing cycle starts running the limit inputs registers are near the driver controll´s..... in addition with the poor electrical characteristics of the due ..... there is a massive voltage drop inside the due chip an there are unexpected register changes ......

so  NEVER USE A DUE WITHOUT AN LEVELSHIFTER or OPAMPLIFIER ;) 

done as said an your code runes pretty well ...... i love it :)

norru, 2020-05-13:
@terjeio I've attempted to use `USB_SERIAL 1` but it doesn't work for me. Control flow and alarms don't seem to be reliable.