terjeio/grblHAL #47: Compiling to DUE

qiwalker, 2020-05-11:
Hi, please help
i want to compile using SAM3X8E and use tinyg2_due_map.h
programming using native USB

where i can find each setting in grbl file

Thanks
Rizki

qiwalker, 2020-05-11:
after enable corexy than arduino IDE show below error 

--------------------------------------------------------------------------------------------------------------

sketch\src\grbl\motion_control.c.o: In function `mc_homing_cycle':

sketch\src\grbl/motion_control.c:736: undefined reference to `limits_set_machine_positions'

sketch\src\grbl/motion_control.c:798: undefined reference to `limits_set_machine_positions'

collect2.exe: error: ld returned 1 exit status

exit status 1
Error compiling for board Arduino Due (Programming Port)

qiwalker, 2020-05-11:
anyone can help :( ?

terjeio, 2020-05-11:
I will, but give me some time. I am just out of bed and at least I want my breakfast before looking into this and another Due issue reported yesterday.

qiwalker, 2020-05-11:
hi @terjeio im sorry hehee
oke thankyou very much

anyway this is amazing job.....

qiwalker, 2020-05-11:
oke it can compile now

at config.h 
if i use due board should i change this Line ?
#ifdef GRBL_ESP32
#include "esp_attr.h" (where to find this file ?)

terjeio, 2020-05-11:
> if i use due board should i change this Line ?

No, this is for the ESP32 controller.

I am not sure what you mean by "due board" - if you have managed to compile for the Due already you have done it correctly. However, there are a couple of different pin mappings available for[ Due shields](https://github.com/terjeio/grblHAL/tree/master/drivers/SAM3X8E/Arduino%20Due%20shield%20pin%20mappings), if it is one of those you want to select then this can be set in the start of the [driver.h](https://github.com/terjeio/grblHAL/blob/master/drivers/SAM3X8E/main/driver.h).

qiwalker, 2020-05-11:
hi, oke my machine is moving perfectly now 
thankyou very much for ur support, 

what is maximum pulse can it around 200kHz ?
i see ar config.h it have backlash compensation how to use it ?

qiwalker, 2020-05-11:
due board i mean i use arduino DUE hehee

terjeio, 2020-05-12:
I do not know what the upper limit for the pulse rate is, but it surely is significantly higher than what an 8-bit Arduino is capable of.

When enabled $-settings are available for the backlash distance, $106, $206... I have not done any testing in a machine so great if you report back if you try it.

Can you confirm that you are using coreXY? Nice to know since I have refactored the code from the original codebase.

qiwalker, 2020-05-13:
Ok i will try to use backlash, and send report back.
Yes i use coreXY, it working perfectly.

btw can i ask u a favor, i need to get trigger from sensor if sensor high and send signal to available pin and then send output to the terminal, just give an output the terminal, no need to do such as reset, free hold, or cycle start.

ie : if it triggered then send data to terminal  "in1"

can u help me please...

qiwalker, 2020-05-13:
"_Can you confirm that you are using coreXY? Nice to know since I have refactored the code from the original codebase_."

WoW you're very smart and awesome.....

qiwalker, 2020-05-14:
Hi @terjeio 

i already try to enable backlash compensation, error found
if it enable cannot connect top Arduino DUE port

if disable it can connect normally
Grbl 1.1f ['$' for help]
[MSG:'$H'|'$X' to unlock]

terjeio, 2020-05-14:
I was wrong with the $-settings, they are $160, $260 etc. And reporting them is broken, set `AXIS_N_SETTINGS `in settings.h to `10` - I'll commit a fix for that later.

Not beeing able to connect could be due to settings beeing reset to default on enabling backlash comp, can you connect after a power cycle? I can connect to both ports, but it failed once for me too - a reset/power cycle fixed that.

It is preferable to use the programming port for communication, less load on the processor. The native USB code [seems to be unreliable](https://github.com/terjeio/grblHAL/issues/46#issuecomment-627894098)...

qiwalker, 2020-05-20:
hi @terjeio 
i will compiling again after u fix it, and give u report. 
may be if u have a time u can also take a look for _"S curve 3rd order motion planning"_

qiwalker, 2020-07-03:
hi 
can i use ramps1.4 with arduino due ?

thanks

terjeio, 2020-07-03:
Should be possible if you are able to take care of the 3.3V (Due) vs 5V (8-bit Arduino) logic level difference.
Googling "ramps 1.4 due" will find some info of what to be aware of like [this post](https://reprap.org/forum/read.php?219,479626).

qiwalker, 2020-07-03:
ouh i see, thankyou very much for your answer
btw ifound this board on internet
can it compatible with grblHal
it use ARM Cortex m3 for cpu

https://www.aliexpress.com/i/33010260342.html

terjeio, 2020-07-03:
It has a STM32F103 MCU and grblHAL already has a driver that should be usable. The RCT6 variant has plenty of flash so that is not a problem, this unlike some Red/Blue Pill boards. However, it is likely that a board map file must be created to match the pinout and the driver code might need some adjustment as well.

Is there a schematic available? Or perhaps the [Marlin codebase](https://github.com/MarlinFirmware/Marlin/tree/2.0.x/Marlin/src/HAL/STM32F1) can be used to find the pin mappings?

qiwalker, 2020-07-03:
https://github.com/bigtreetech/BIGTREETECH-SKR-MINI-V1.1
here's the link i found for this board

from sam3x8e i found it can use ramps1.6
is it mean arduino due is compatible using ramps1.6 ? (no need hardware modificatio?

terjeio, 2020-07-03:
Thanks for the link, I see from the schematic that a new map file as well as some code changes is required for the Bigtreetech board. Notably the control signals (reset, feed hold, cycle start) are missing, repurposing the + or - limit switch inputs is one option, another is using the thermistor inputs after removing the large filter caps.

> from sam3x8e i found it can use ramps1.6

Again possibly but with care not to expose any pins to 5V, see [this link](https://reprap.org/wiki/Arduino_Due_Based_Electronics) as well. Shorting the termistor inputs to ground would be enough?

qiwalker, 2020-07-04:
yes, i already figure it out not to use this board, always thanks for your explanation

and about ramps 1.6 so i just need to connect input to thermistor to gnd to make it work with arduino due ? actually i have tmc2130 driver, and on grbl hal have driver setting for it, thats why i need to use ramps which is already provide socket for the external driver

btw your firmware will be profesional use it very good firmware which is can be able to almost 32bit cpu, good job bro.

closed 2020-12-23 by qiwalker

weed2all, 2020-12-23:
I have an old arduino due with radds 1.6 board used in 3d printers...the radds is well contructed piece of hardware...could add support for it?

terjeio, 2020-12-23:
> I have an old arduino due with radds 1.6 board used in 3d printers...the radds is well constructed piece of hardware...could you add support for it?

Yes, but I have no information about that board. Can you submit a mapping file or provide the needed info (which pin is used for what)?

And please open a new issue as this has been closed by the OP.

weed2all, 2020-12-23:
[Radds photo and pin as in marlin.zip](https://github.com/terjeio/grblHAL/files/5734705/Radds.photo.and.pin.as.in.marlin.zip)

Is all i have for this board...I took the pinout from marlin firmware as it is

phil-barrett, 2020-12-23:
I know I should wait until the new issue is opened but...

A suggestion  - set up to build for the Due and then make your own map file.  There is a mega_2560_map.h that is probably a good starting point.  One you have to build set up, it will be pretty easy.