terjeio/grblHAL #48: issues with safety door and cooland pins

S2000Stefan, 2020-05-11:
Good morning,
When i activate the safety door pin and open the safety door during a milling program, the spindle moves to the parking position and switches off. After closing the door, the spindle moves down and starts again and the milling program continues.
If the door is opened several times, the spindle moves downwards and starts but the milling program does not continue even though the axis has reached the start value but the Z axis moves slowly upwards, on and on.
Usually happens after two or three times open the door.

If i invert the mist pin in grblhal and open the safety door the spindle moves to the parking position but does NOT switch off.
The milling program continues to run normally when the door is closed again unless the error I described above occurs.

BTW the mist and cooland pin does not work for me neither with M7 M8 M9 in the milling program or when clicking on the control box.

terjeio, 2020-05-12:
Coolant outputs are not working due to the pins beeing used for the JTAG interface. I see that I have commented out a line that disables this but leaves the SWJ interface active.

https://github.com/terjeio/grblHAL/blob/b6e74c1011e4b716d58668ec55eec9eae6b0d7e5/drivers/STM32F1xx/Src/driver.c#L915

Depending on how you program the device it may be ok to remove the comment. I program via a ST-Link connected to the four pin header at the end (I do not connect the reset pin), this is working for me - the device is not bricked by uncommenting the line.

I did add a piece of code to reenable JTAG, but only if the SD-card plugin is active:

https://github.com/terjeio/grblHAL/blob/b6e74c1011e4b716d58668ec55eec9eae6b0d7e5/drivers/STM32F1xx/Src/driver.c#L707-L722
 
I am not sure this is really needed - my memory is a bit dim regarding this now, I guess I should have added some comments...

The parking issue is a bit more challenging, I am getting closer to a solution but not yet there.

terjeio, 2020-05-12:
Do you have any form for debounce handling on the door switch? After spending nearly two days digging into this I still have problems making the code robust, and switch bouncing is the (remaining?) problem.

I have a couple of choices:
* add software debouncing at the driver level (I have too many drivers to fix...)
* try to add software debouncing in the core, but perhaps only for the restore phase
* ignore door reopening during the restore phase (not good)
* leave it to users to supply a clean signal (IMO not good either)

I will take a short break from this as I have some paid work pending, and I need to sleep a bit on it as well.

S2000Stefan, 2020-05-12:
> Coolant outputs are not working due to the pins beeing used for the JTAG interface.

Thanks for the information that the pins do not work, I had already thought. Don't worry about it anymore, I don't need them urgently, I just noticed, thought I'd report it.

> Do you have any form for debounce handling on the door switch?

No i don't use debounced switches, but it would be no problem to realize this i found here some information about it for me, maybe there is also something for you, unfortunately in german.
[https://www.mikrocontroller.net/articles/Entprellung](url)
Also some program code about it.

> I will take a short break from this as I have some paid work pending, and I need to sleep a bit on it as well.

Yes it is good to take a break if you have a problem that is really in front of you, then you usually get the best ideas, finish your important work first, the thing with the parking issue can wait.   ;)

S2000Stefan, 2020-05-15:
Update:
I have debounced the door switches according to this scheme and can report that it now works. After triggering the door switches, the spindle moves into parking position and switches off. After closing the door everything runs off again in reverse order and the milling job continues.
[https://github.com/gnea/grbl/wiki/Wiring-Limit-Switches](url)
(Normal closed improved noise filtering)

Unless I reopening during the restore phase there are still problems 
You know that, but who does that.

phil-barrett, 2020-05-30:
Sorry for the late comments here.  

The gnea wiki page suggests opto isolators but I'm not convinced they work 100% for debouncing in eneral.  That said, that's what I use on my various break out boards (though for isolation, debounce).  I suspect that, with the addition of grblHAL debounce, optos work pretty well. 

By the way, the link goes to grblHAL, not gnea, even though the text says gnea.  odd.

S2000Stefan, 2020-05-31:
Yes a break out board will be my next choice to debounce the switch pins. When my milling machine is finished mechanically. 
I'll take a look at your board. But at the moment use a STM32F103C8T6 Blue Pill.
I don't understand why the link doesn't work properly. You have to copy and paste the link, then it works.

S2000Stefan, 2020-10-04:
I think the issue can be closed since the problem was solved with some hardware.

S2000Stefan, 2020-11-08:
Unfortunately I have to bother you again about the safty door. 
After I changed my controller to UART and tested it on the real machine I noticed the following:
When activating the safty door the spindle goes off and moves to z-max, everything as it should be, first DOOR2 then DOOR1. After closing the door, the spindle moves only a short distance downwards (5mm), stops and after that the z-axis continues counting in the DRO window, although the z-axis is standing still. Display first DOOR2 then DOOR4. 

I also noticed that you can only run the spindle CW, no matter if you click CCW or use M4 to start the spindle, CW is always activated. Not that I need this I only noticed.

If I activate Alarm Substatus in $10 I don't get any feedback at startup because I force Homing on startup.
In earlier versions there was always alarm 11 at startup.

Everything refers to build 20201020, not the current version.
The one with the Safty door may not work for some build for a while, I haven't used it for a long time.

BTW that with the new STM32F303CCT6 controller may take some time, they are not available at the moment.  :(

terjeio, 2020-11-08:
> After closing the door, the spindle moves only a short distance downwards (5mm), stops and after that the z-axis continues counting in the DRO window, although the z-axis is standing still.

This is odd, somehow the Z motor must be disabled for this to happen. Will look into it tomorrow.

> I also noticed that you can only run the spindle CW, no matter if you click CCW or use M4 to start the spindle, CW is always activated. 

It is a state reporting bug in the driver, the pins are set correctly. Fix in next commit.

> If I activate Alarm Substatus in $10 I don't get any feedback at startup because I force Homing on startup.
In earlier versions there was always alarm 11 at startup.

This is due to a regression from when we added manual homing of axes. Fix in next commit.

terjeio, 2020-11-11:
I believe I have fixed the safety door code now, tricky stuff. Some of it was IMO bad code carried over from legacy grbl, some my own and from relying on legacy code beeing good. I have committed the changes to the [test branch](https://github.com/terjeio/grblHAL/tree/test) for now, test with care!

Note that opening and closing the door multiple times during retract/restore should now be handled correctly. I suspect legacy grbl does not, to be tested later.

S2000Stefan, 2020-11-11:
terjeio I am sorry to inform you that I cannot test the test branch.
The homing is not possible in this version, after $H or pressing the homing button the z-axis does not move. 
And I suppose that the controller is no longer responsive, I could only test it for a very short time.
But normal movement of the axes is possible.

phil-barrett, 2020-11-11:
I have seen $H fail.. (Homing in ioSender also does not work.). Sometimes you can individually home axes.  Sometimes not.  The problem sometimes disappears.  I have 2 customers that can not home at all and their setting look correct.  I have seen it on my little router but it went away and I have no idea what, if anything, I did.  So, I think there is a lurking intermittent bug somewhere.

terjeio, 2020-11-11:
@S2000Stefan grbllib.c was not updated as it should for some reason, should be ok now. Sorry about that.

@phil-barrett I would like to see the settings from the users that cannot home at all. Will it be possible to get them?

phil-barrett, 2020-11-11:
I have had a little trouble with email - lost a number of messages (lost
your mail address, for example).  Pretty sure I still have at least one,
will take some digging to find.

S2000Stefan, 2020-11-13:
Terjeio I think I owe you an apology.
After some tests yesterday and today I had to find out that the door switch is responsible for my problems, despite the upstream optocoupler.
I replaced all my hardware switches weeks ago and fitted them with optocouplers. Unfortunately, I did not test intensively whether all switches functioned properly, so I did not notice that the safty door no longer functioned properly.
Yesterday and today with the new test version and different settings and USB or UART connection, I noticed that the controller still does crazy things with the safty door. Now I assume that you have tested the whole thing and the error is on my site.

Long story short, today I replaced the new door switches with the old ones and the safty door works!?!
Actually I thought that the optocoupler filters out the interference coming from the switch, but I guess not???
Electronics is a book with seven seals for me. As I said, the old switches seem to work perfectly.
I will do more intensive tests tomorrow. 

But then I noticed a little beauty flaw. When the spindle stops at the safty door, the display in the ios transmitter stays at on. But as I said, this is only a small mistake.

The alarm substatus is now working again, thank you. :+1: 

Because I always did a reset with $RST=*   I noticed that everything is reset to default, only G30 remains.

terjeio, 2020-11-13:
> Terjeio I think I owe you an apology.

No you do not ;-) My part of safety door code was not good, hopefully it is better now. At least my testing says so.
And I found that the code carried over from grbl was not good either, that's a plus too.

> Actually I thought that the optocoupler filters out the interference coming from the switch, but I guess not???

Yes they do, but only up to a point. Simple filtering is not the same as proper debouncing where the switch state is rechecked after tens of milliseconds. And even that does not help if the switch is really bad. The parking code is also sensitive to fast changes as there are quite a bit of state transitions going on in the controller code. Hard to get right and hard to test.

> But then I noticed a little beauty flaw. When the spindle stops at the safty door, the display in the ios transmitter stays at on. 

I believe this is carried over from legacy grbl too - there are bits of code from that that I have never understood the reasoning behind. I'll check it out.

> Because I always did a reset with $RST=* I noticed that everything is reset to default, only G30 remains.

Ok, I'll check that too.

Thanks for your feeback.

S2000Stefan, 2020-11-14:
Update: 
So today I did some tests and rebuilt my hardware again. 
I removed the optocouplers from the safty door and put the old system back in that I linked above.
With the optocoupler the spindle didn't start once, after pressing the button about 100 times it happened once, could have been a coincidence or could be due to my hardware, who knows?
With the other system from above there were no problems even after several several times pressing the safety door.
I will leave it like this now.

One more thing I have to say about the safty door; when the door is opened, the acceleration upwards of the z-axis is good. When closing the door and accelerating downwards, the acceleration is associated with a slight bump, becoming more pronounced the higher the Parking fast rate ( $59) is set.  
Here is a short video to make it clearer, hope you can hear it:
[https://www.youtube.com/watch?v=Gbxg1h642TQ](url)

I forgot to tell you, the spindle display in the IOSender is now also displayed correctly, whether you click on it or use the M3 M4 command. :+1: 

One more little thing for the automatic tool change process. I think I have asked you this question before. 
With the $TPW command the probe will move down to the touchplate at the set speed in $344 (Tool change search seek rate) and back up from the touchplate at exactly the same speed, so you think you have connected this to it, and then move down to the second probe with the Tool change locate feed rate ($343). 
Would it be possible to use the $344 (Tool change search seek rate) times two or three to speed up the first retraction? 
It is not a must, only if it is very easy to implement.

terjeio, 2020-11-15:
> When closing the door and accelerating downwards, the acceleration is associated with a slight bump

I wonder if this is due to switching the motor on, can you try with setting $37 = 4?
BTW switching motors off when idle is bad for a homed machine? If I switch off the Z-axis motor on my router the spindle drops down thus losing position.

> Would it be possible to use the $344 (Tool change search seek rate) times two or three to speed up the first retraction? Would it be possible to use the $344 (Tool change search seek rate) times two or three to speed up the first retraction?

It would, but IIRC I landed on adding a setting for the pull-off rate? No done yet, it would be a little easier to multiply the $344 value by a factor but a new setting is better? I needto add a number of settings for auto squaring so could add this at the same time.

S2000Stefan, 2020-11-15:
> I wonder if this is due to switching the motor on, can you try with setting $37 = 4?
BTW switching motors off when idle is bad for a homed machine? If I switch off the Z-axis motor on my router the spindle drops down thus losing position.

O,k, I will try later. So far I have not had any problems with step losses when switching off my z-axis. But it would make sense to leave the holding current on with $37=4.

> It would, but IIRC I landed on adding a setting for the pull-off rate? No done yet, it would be a little easier to multiply the $344 value by a factor but a new setting is better? I needto add a number of settings for auto squaring so could add this at the same time.

Certainly a new setting is better but I think for such a small stroke of the z-axis. The latch distance is mostly only 0.5mm - 1mm, if it would be enough to insert a multiplier, i think it is much easier and sufficient, as i think. But of course I leave it up to you.  ;-)

S2000Stefan, 2020-11-15:
> I wonder if this is due to switching the motor on, can you try with setting $37 = 4?
BTW switching motors off when idle is bad for a homed machine? If I switch off the Z-axis motor on my router the spindle drops down thus losing position.

> O,k, I will try later. So far I have not had any problems with step losses when switching off my z-axis. But it would make sense to leave the holding current on with $37=4.

Just tested it, no difference. 
But I checked my power amplifiers (Leadshine DM556), they already have a holding current in idle position and are currently set to half, so no step losses at idle.
I should have known, you just get older.   :-)

terjeio, 2020-11-15:
Ok, if you enter a G1 command for the z-axis with the same feed rate and distance does that sound the same way?

And just to be sure, it is the noise heard just as the movement starts?

S2000Stefan, 2020-11-15:
Was just at the machine and tested with G1. But I have to say that you don't hear a beat even with F3000.
I have made two videos again.
The first one is with G1 F3000 and the second one is with Safty Door also with F3000, on both I deactivated the spindle so you can hear it better.
G1
[https://www.youtube.com/watch?v=y-QFdu0VY04](url)
Safty Door
[https://www.youtube.com/watch?v=YcCudnPQRTU](url)

terjeio, 2020-11-15:
I'll have to check switch debouncing in the F103 driver to see if that can be the cause. And perhaps I should ignore switch state changes during the fast retract/restore phase? Not easy this...

FYI when testing with my mini router (MSP432 and Trinamic drivers) I cannot detect anything out of order. It sounds nice and smooth.

terjeio, 2020-11-15:
I am getting older too... There is no debounce on the control signals, only on the limit signals.
First thing to do is to add debounce to the door switch. Tomorrow?

S2000Stefan, 2020-11-15:
Don't stress yourself with such small trivialities, you can still live with a small blow.
I just noticed it and wanted to tell you.
Think that you can use your valuable time for more important things.
My new controller, STM32F303CCT6, has been ordered and has been delivered, I hope it will come soon. 
I am sure that I will annoy you again. ;-)

terjeio, 2020-11-15:
> Don't stress yourself with such small trivialities, you can still live with a small blow.

Well, I have spent a lot of time on the door state handling so I am not going to let a triviality stop it functioning properly ;-)
So the door switch is going to get the debounce treatment - that is an easy thing to fix. A bit more work to do it for all drivers though, but that can wait.

The STM32F303 driver should not be too hard to make, with luck it can be a part of the STM32F4xx driver. I am ready to be annoyed...

terjeio, 2020-11-16:
[Test branch](https://github.com/terjeio/grblHAL/tree/test) updated with door switch debounce for STM32 drivers and a new setting for pull-off feed rate from tool length sensor among the changes.

There is one issue with debouncing that remains though, debouncing is not done on both switch close and switch open as it should for at least the door input. And switch handling could be improved in grblHAL in other ways too (compared to grbl) as the limitations related to GPIO interrupts in 8-bit Arduinos no longer apply. For another rainy day...

S2000Stefan, 2020-11-16:
O.k. could test the new version. But I have to tell you that the z-axis is still beating. But as I said, it is not that bad, you can live with it.

To the $TPW command:
After the first rehearsal the retreat is now as fast as you set it in $345. :+1: But now I have to note that after the latch rehearsal the speed is still very slow during the retraction. :-)
It also seems to me that the latch distance is now bigger than indicated in the probing tab? Maybe I am wrong and I will test it again later.
Could it be that there is still something missing: :-)
[image]

S2000Stefan, 2020-11-16:
Another thing I noticed earlier $61, Ignore door when idle, disturbs the operation of the safty door very much. If $61=0, the safty door process is correct and ok. 
If $61=1, the safety door process is disturbed, e.g. if you open the door the spindle will stop immediately without pulling out distance first. 
Even if a tool change is in progress, the tool TLO is set up with $TPW and you press start, the spindle returns to the last position before the tool change and increases the spindle speed by opening the door, the spindle stops and the milling program continues.

Example:
ok  G1X-28.000Y-25.000
ok  G0Z3.000
ok  M5
ok  M9
pending   T5M06(MSG,End Mill 3 mm Holz)
ok  M3S7000
ok  M07
ok  M08
ok  G4 P5           <<< Press safty door here
     G0X-10.683Y-20.003Z3.0

terjeio, 2020-11-16:
> But I have to tell you that the z-axis is still beating.

Ok, I'll check this out later when I my CNC machine simulator has evolved a bit more. With that I can perform more detailed diagnostics.

> It also seems to me that the latch distance is now bigger than indicated in the probing tab?

$TPW does not take any input from the sender. Parameters are from settings in the controller and the retract distance is "hardcoded" to 2mm via the TOOL_CHANGE_PROBE_RETRACT_DISTANCE #define in tool_change.c. The latch distance is hardcoded to TOOL_CHANGE_PROBE_RETRACT_DISTANCE + 2mm. I have to add another setting or two for these?

> Could it be that there is still something missing: :-)

You mean the sender does not know about $345? Details about settings are in the file "hal_setting_codes_en_US.txt" in the sender folder. This has to be updated when I add settings, I try to keep the controller repository in sync with changes so I do not need to make a new sender version... Latest version can be found [here](https://github.com/terjeio/grblHAL/tree/test/doc/csv), update "hal_setting_codes_en_US.txt" in the sender folder with taht and new settings will be recognized.

> If $61=1, the safety door process is disturbed, e.g. if you open the door the spindle will stop immediately without pulling out distance first.

This one I added for my CO2 laser, when I open the lid (door) I wanted the laser to switch off and not block jogging when idle as it is hard to position the workpiece when the lid is closed. When opened when a program is running I wanted the laser to switch off and the program to halt and no Z-motion (since the Z position is used for focusing). I have to go back an look at what to do when laser mode is not active - perhaps disregard this setting altogether? Or could a similar functionality be useful for a mill/router too?

S2000Stefan, 2020-11-16:
> $TPW does not take any input from the sender. Parameters are from settings in the controller and the retract distance is "hardcoded" to 2mm via the TOOL_CHANGE_PROBE_RETRACT_DISTANCE #define in [...]

No you don't need to, I now know how I can adjust the distances according to my wishes.

> You mean the sender does not know about $345? Details about settings are in the file "hal_setting_codes_en_US.txt" in the sender folder. This has to be updated when I add settings, I try to keep the [...]

O.k. thanks will update the file

> This one I added for my CO2 laser, when I open the lid (door) I wanted the laser to switch off and not block jogging when idle as it is hard to position the workpiece when the lid is closed. When [...]

Yes, this function is normally deactivated for me, but I only tested it with the function switched on and that's when I noticed it. I don't think that this is relevant for a milling machine, whereas it works perfectly for a laser.

S2000Stefan, 2020-11-22:
I believe the issues can be closed again. If everything works as it should and the small stroke of the z-axis is negligible.

closed 2020-11-22 by S2000Stefan