terjeio/grblHAL #49: Sigh, new version of Teensyduino breaks grblHAL when using PJRC USB library

phil-barrett, 2020-05-15:
Got my Teensy 4.1. Had to upgrade to the beta version of Teensyduino 1.52 (beta 6, almost released). Compiled ok for T4.1 but when I tried to connect got connection refused. Rebuilt with Arduino USB class library (USB_SERIAL_GRBL set to 1) - it connects and I can talk to grblHAL.

I have a bug report into PJRC. Hopefully it's something simple. Technically, it's a bug on their end - hopefully they will own it.

terjeio, 2020-05-17:
I have updated to 1.5.2 to check, but was stupid not to make a a copy of the previous version first...

There is no obvious changes in the source tree that I can identify as the reason for this.  Maybe something in the makefile? I will upgrade my laptop version of Arduino and 1.5.1 to be able to diff files not available in the [github repository](https://github.com/PaulStoffregen/cores/tree/master/teensy4).

Note that grblHAL uses the c (not cpp) versions of the USB library. This could be a lead to follow. A possible reason could be that PJRC has started to use a peripheral (timer?) that grblHAL later claims. 

My dislike for the Arduino framework only grows, this could most likely be solved quickly if real debugging was possible...

phil-barrett, 2020-05-17:
Not much response on the PJRC forum. I should have been more attentive to the C vs CPP version. 

And we agree "violently" about the lack of debugging support. I still think PaulS is in a good place to make that happen but he seems to be uninterested. An ICE or similar header takes up space.

terjeio, 2020-05-18:
> Technically, it's a bug on their end - hopefully they will own it.

Nope, it was on my end - my code referenced an uninitalized pointer. Strange that it worked before.

I guess that a memory exception resulted in the processor hanging in the (default?) exception handler, likely with all interrupts disabled. Easy to track down with a debugger...

phil-barrett, 2020-05-18:
glad to hear it was a simple fix. I have a lot of non-electronics stuff to do here but will update and test as soon as possible.

closed 2020-05-24 by terjeio