terjeio/grblHAL #51: ESP32 driver: positions not set to zero with UGS

terjeio, 2020-05-21:
Reported by @HuubBuis in a [grbl-Mega issue](https://github.com/gnea/grbl-Mega/issues/26#issuecomment-632255527).

I tested ESP32 and found 2 issues.

Positions are not set to zero, tested using UGS
G10 P0 L20 X0 Y0 Z0

Positions are not set to zero, tested using UGS
G92X0

terjeio, 2020-05-21:
It works for me, but only after setting `#define COMPATIBILITY_LEVEL 1` in grbl/config.h:

https://github.com/terjeio/grblHAL/blob/73604aa13c0b748a760d6f2e49c77830f38ce4f3/grbl/config.h#L41-L50

More info in the [Wiki](https://github.com/terjeio/grblHAL/wiki/Changes-from-grbl-1.1#workaround).

HuubBuis, 2020-05-21:
I have set  #define COMPATIBILITY_LEVEL 1 because at that level, the interface seems to be grbl compatible and UGS can be used for testing.
Will test both levels (0 and 1) again tomorrow and use a terminal.

HuubBuis, 2020-05-23:
I have tested compatibility_level 0,1 & 2 using a terminal (CoolTerm). G92 is working ok on all levels. I think I didn't set $10 to report work positions.
You may close this issue.

closed 2020-05-24 by terjeio