terjeio/grblHAL #53: ESP32 Complilation

ithinkido, 2020-05-25:
So it has been a while since I last complied grblHAL with ESP-IDF, but I remember that I managed this once before. This time I have not been able to get grblHAL to compile. I have followed the instruction https://docs.espressif.com/projects/esp-idf/en/stable/get-started/index.html#installation-step-by-step
Once that is done I downlad the zip, unpack it and copied the grbl root files to ESP32 drivers grbl folder, opened terminal , changed to the ESP32 directory and ran  idf.py build
The build works up to boot loader.bin then starts to go wrong with 

FAILED: grbl.elf 
: && /home/ubuntu/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++  -mlongcalls -Wno-frame-address  -nostdlib CMakeFiles/grbl.elf.dir/project_elf_src.c.obj CMakeFile..........
......
......
......
/home/ubuntu/esp/esp-idf/components/esp32/cpu_start.c:539: undefined reference to `app_main'
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
ninja failed with exit code 1

Just to check, I tried a hello world example and managed to get this to work, so I am thinking that the tool chain should be ok.
Is there something that I need to to different here for grblHAL ? or am I just out of practice.

jschoch, 2020-05-26:
i used cmake which will generate a Makefile and make and it worked just the other day.  There is a typo in bluetooth.c if you are using bluetooth.  you also have to set ESPPORT and ESPBAUD in your env to use make as well as `make menuconfig` if you want to make tweaks.  What IDF version are you using?  It may help if you post the full stack trace.

ithinkido, 2020-05-26:
OK, I have not tried cmake before, I can look at that. 

>
> What IDF version are you using?
>
I am recursive cloning 4.0.1 

I here is the full stack trace.
[stack trace.txt](https://github.com/terjeio/grblHAL/files/4681061/stack.trace.txt)

terjeio, 2020-05-26:
No backwards compatibility... The current configuration is for  IDF 3.2+ (but surely < 4).

I see _sdkconfig_ is changed by the build process, and there is no attempt at compiling the grblHAL source. I would guess that _CMakeLists.txt_ is read since there is an attempt at linking _grbl.elf_, thus hinting at a format change?

There may be breaking changes in the ESP32 libraries as well since _sdkconfig_ is changed, making it unlikely to be fixed by a few minutes of work. Upgrading to IDF v4 is thus going to be added to my todo list unless somebody else want the challenge...

ithinkido, 2020-05-26:
That was it. I tried with 3.2 and it has worked. Thanks

closed 2020-05-26 by ithinkido