terjeio/grblHAL #57: Arduino Ide compiler warning in v20200603 grblHAL-master/drivers/IMXRT1062/main/driver.c

hsh002, 2020-06-08:
WARNING
Dev tools       [Arduino ide 1.8.12]
Compiler        [arm-none-eabi-g++]
In file         [grblHAL-master/drivers/IMXRT1062/main/driver.c}
Warning         [warning: assignment from incompatible pointer type]
Line number     [1051]
Line text       [hal.probe_get_state = probeGetState;]

UPDATE 1
In file         [grblHAL-master/drivers/IMXRT1062/main/driver.c}
Change to       [lines 436..439]
to              static probe_state_t probeGetState(void)
                {
                  probe_state_t lprobe_state
                  lprobe_state.value = (((Probe.reg->DR & Probe.bit) != 0) ^ probe_invert);
                  return lprobe_state;
                }

UPDATE 2
In file         [grblHAL-master/grbl/grbl.h]
After line 62   #include "stream.h"
Insert          #include "probe.h"

QUESTION
 Working on 'teensy 4.1', waiting for additionnal 'PSRAM SMD chips' from China...
 For trace, i can only use 'Usb Serial' cause waiting for bottom soldering 'PSRAM SMD chips'.
 What is the best way to implement a trace on 'Usb serial' ?
 Is there a trace message format which is ignored by 'gcode sender clients' ?

Sorry for my poor english...
Thanks a lot for your job.
Have a nice day.

terjeio, 2020-06-08:
> WARNING
Dev tools [Arduino ide 1.8.12]
Compiler [arm-none-eabi-g++]
In file [grblHAL-master/drivers/IMXRT1062/main/driver.c}
Warning [warning: assignment from incompatible pointer type]
Line number [1051]
Line text [hal.probe_get_state = probeGetState;]

Update to the [latest driver commit](https://github.com/terjeio/grblHAL/commit/10819db219b1ea9406fa2c20e8add24f0b2a569f), this warning is from using the previous version.

> For trace, i can only use 'Usb Serial' cause waiting for bottom soldering 'PSRAM SMD chips'.

I do not understand this, T4.1 needs PSRAM for running grblHAL in some configurations?

> What is the best way to implement a trace on 'Usb serial' ?

`hal.stream.write("[MSG:my trace message]" ASCII_EOL);`

Note that calling this method from an interrupt context is not advisable and may crash the system.

> Is there a trace message format which is ignored by 'gcode sender clients' ?

Not that I know of, the [MSG:] format will likely show up in the UI. You may try to output a message encapsulated by [my message] or [TRACE:my message] but that could crash the sender.

A better method is perhaps to add hook up a USB <> serial converter to the UART pins and enable UART communication by calling  `serialInit()` in `driver_init()` and use `serialWriteS()` instead of `hal.stream.write()` to output the messages.

Note that there is an empty function `report_realtime_debug()` in vanilla grbl, but I do not know what this is supposed to do. I have removed it from grblHAL.

phil-barrett, 2020-06-11:
> I do not understand this, T4.1 needs PSRAM for running grblHAL in some configurations?
I think he wants to use PSRAM for writing a debug stream. Interesting idea, perhaps I should take a look at that.  If you haven't looked at the shipped T4.1, it has pads on the bottom for optional 2 memory chips.

closed 2020-06-19 by terjeio