terjeio/grblHAL #58: Build 20200603 crashes Grbl-GCode Sender A19

S2000Stefan, 2020-06-08:
I use a four-axis system with a blue pill STM32. When i use GrblHal Build 20200603 with Grbl-GCode-Sender Alpha 19 on three axes (XYZ) everything works as usual without problems.
But if I activate the rotation axis, the Grbl-GCode-Sender simply crashes after a short while.
To see in the picture below:
[image]
Unfortunately I can't give you any details yet to better isolate the error.
I have not yet identified a pattern that would explain the error.

terjeio, 2020-06-09:
Can you upload the file that fails? And your settings?

S2000Stefan, 2020-06-09:
Here the file as a link (10MB):

[https://www.dropbox.com/s/syeab7sfy32xj8o/Pfeffer%2054mm%232_Finishing.nc?dl=0](url)

Hope this works.

Settings file:
[https://www.dropbox.com/s/q2v9x0f4d90s10n/settings.txt?dl=0](url)

%
; grblHAL
; 1.1f(STM32F103C8).20200603
; [OPT:VNMPHTS+,35,1024,4,0]
; [NEWOPT:TC]
; [DRIVER VERSION:200528]
;
$0=4
$1=25
$2=0
$3=4
$4=0
$5=15
$6=1
$10=2047
$11=0.010
$12=0.002
$13=0
$14=15
$15=0
$16=0
$17=0
$18=0
$19=0
$20=1
$21=1
$22=55
$23=0
$24=20.000
$25=400.000
$26=25
$27=2.000
$28=0.100
$29=0
$30=24000.000
$31=0.000
$32=0
$33=5000
$34=0.0
$35=0.0
$36=100.0
$37=0
$39=1
$40=1
$41=1
$42=2
$43=1
$44=4
$45=3
$46=0
$47=0
$56=5.0
$57=150.0
$58=-5.0
$59=2000.0
$60=1
$61=0
$62=0
$63=2
$64=0
$65=0
$100=640.000
$101=640.000
$102=640.000
$103=66.667
$110=3500.000
$111=3500.000
$112=3500.000
$113=18000.000
$120=500.000
$121=500.000
$122=500.000
$123=3500.000
$130=513.000
$131=539.000
$132=160.000
$133=0.000
%

terjeio, 2020-06-09:
Thanks, download is ok - running tests now. The problem seems to be related to a 64 character boundary issue in the STM32 USB transfer implementation, so I will try with a FTDI interface to see if that fails too.

terjeio, 2020-06-09:
Serial via FTDI seems to be ok, at line 26000+ now. I will take a look at the 64 character workaround again...

Another issue is the A-axis DRO display. I wonder if it could be a good idea to limit the number of fractional digits to 1 (or not display them) when the axis limit is configured as infinite ($133 = 0) or when the program limit for the axis is large, this since I see that your file contains rather large A-axis values.

S2000Stefan, 2020-06-09:
Thanks for searching and finding the issues so quickly.
It just seemed strange to me that it worked with previous versions of grblhal.

> Another issue is the A-axis DRO display. I wonder if it could be a good idea to limit the number of fractional digits to 1 (or not display them) when the axis limit is configured as infinite ($133 = [...]

Yes the thing with the dro indicator of the a-axis is disturbing but just a beauty flaw, to limit the decimal place to 1 digit is a very good idea. As indicator ($133=0) should work.
The cam software generates a helix of the a-axis, which gives a very good surface, because the play of the axis is compensated.

May I also make a suggestion regarding the DRO display. Since I am already a bit older and usually stand a bit further away from the pc screen when milling, it would be an advantage if the dro display was bigger.    :)

terjeio, 2020-06-10:
> It just seemed strange to me that it worked with previous versions of grblhal.

Did it work with this file? It hangs for me when I revert to the previous version, and I suspect the sender crash when it runs out of memory. This is due to a "bug" in the sender, I assume grbl will never send a empty line - and this is what the STM driver does when it goes wrong. I have fixed this in the sender and that helps, but it is _not_ the solution as a empty line should never be sent in the first place.

I am not sure if this is a STM bug or a bug the grblHAL send function, I'll try with a Teensy 4 card to see if that suffers the same problem. It could well be the STM USB driver code as it seems to [have its own problems](https://community.st.com/s/question/0D50X00009XkfIBSAZ/stm32cube-usb-cdc-possible-bug-found-rx-tx-race-condition)...

@shaise : IMO you need to be aware of this for your [Black Pill port](https://github.com/terjeio/grblHAL/issues/55#).

shaise, 2020-06-10:
if it is the 64byte boundary issue, then the blackpill is same as bluepill. Did you change the workaround solution?

terjeio, 2020-06-10:
@shaise : Not yet sure what the reason is, could be a memory leak. I have working solutions that suggest so, more testing required.

BTW the Teensy 4 card did not fail.

S2000Stefan, 2020-06-10:
> Did it work with this file? It hangs for me when I revert to the previous version, and I suspect the sender crash when it runs out of memory.

Yeah, with this exact file. The file has also been milled three times on a real machine. Here is my version with which I am currently working and the file runs through without problems.
[https://www.dropbox.com/s/rytmqe4lgkqbp08/STM32F1xx.rar?dl=0](url)

terjeio, 2020-06-11:
This must be a STM USB driver/stack issue. I added a few missing parser state G-codes to the `$G` report in the latest build and this triggers the crash as it seemingly randomly corrupts the data to be transmitted. The corruption happens in the data already buffered by the USB driver from a `?` status request when a `$G` report is added, but only when the `$G` report is of a certain length and the `?` status report is > 64 bytes. Could well be when the combined length is 128 bytes?

The solution that seems to work is to add double buffering of transmit data in the grblHAL USB interface code, I have not been able to reproduce the issue with that in place.

I will commit this as a workaround when I have run a few more full length tests to completion.

S2000Stefan, 2020-06-12:
The program passed the stress test yesterday and today with flying colours. 
Thus I close the issue.
Thanks for the fast support.

closed 2020-06-12 by S2000Stefan