terjeio/grblHAL #60: Arduino Ide compiler warning in v20200603 grblHAL-master/drivers/IMXRT1062/main/driver.c

hsh002, 2020-06-12:
ArduinoIde compiler warning 'Unused data' line 146 "static qei_t qei = {0};"
This static is not defined protected.
Have a nice day.

terjeio, 2020-06-12:
Thanks, did not see that.

Do you know if there is a way to enable the Arduino IDE to give a warning/error summary at end of compilation? I am used to getting that when compiling using professional IDEs such as Eclipse.

And then my Arduino IDE [does not want to start](https://forum.arduino.cc/index.php?topic=471641.0) anymore...

hsh002, 2020-06-12:
Hi Terje,

My config :
-Windows 10 Pro 64  v1909 Os 18.363.900 from Winver.exe
-Arduino Ide 1.8.12 Windows ZIP file for non admin install
  -https://www.arduino.cc/download_handler.php?f=/arduino-1.8.12-windows.zip
-Teensyduino 1.52 installer
  -https://www.pjrc.com/teensy/td_152/TeensyduinoInstall.exe
 Compiler for Arm/Teensy
  -<ARDUINO-1.8.12-ROOT>/hardware/tools/arm/bin/arm-none-eabi-gcc-5.4.1.exe
 Compiler flag for warning (gcc & g++)
  -Wall
  You can set this flag in ArduinoIde->File->Preferences->Compiler warning:->All

 In case of new define in one 'board code parts'
  You must do two backwards steps compile  for THE board :
  1 with the new define set 
  1 with the new define unset

 In case of new define usage in 'grbl code parts'
   You must do two backwards steps compile  for EACH board :
  1 with the new define set 
  1 with the new define unset
... THE TITAN WORKS... LET THIS JOB DONE BY GITHUB USERS LIKE us...

What is your operating sytem for 'Arduino Ide' ?
Is it the same OS for Eclipse ?
What is your toolchain gcc/g++ compiler for eclipse ?
Do you need many toolchains for all the boards ?

Thanks for your job.

hsh002, 2020-06-13:
You can force the 'gcc/g++' compiler to stop project build on warnings :
 - Using the compiler command line switch :
    [**-Werror**]
 - See :
    [**https://gcc.gnu.org/onlinedocs/gcc/Warning-Options.html**]

Hack for
   ArduinoIde(1.8.12)/Teensyduino(1.52)  :
 - In file :
    [**<ARDUINO_IDE_ROOT>/hardware/teensy/avr/boards.txt**]
 -Replace every occurence :
    [**-Wno-error=narrowing**]
 -With :
    [**-Werror**]

Have a nice day...

terjeio, 2020-06-16:
Thanks for input.

> What is your operating sytem for 'Arduino Ide' ?

Win7

> Is it the same OS for Eclipse ?

Yes

> What is your toolchain gcc/g++ compiler for eclipse ?

I use the default compiler, I have never checked the version(s) provided. For Texas Instruments CCS I use their own compiler, the rest is AFAIK using gcc.

> Do you need many toolchains for all the boards ?

Yes, I have 4 instances of Eclipse based IDEs: CCS for Texas Instruments MCUs, STM32CubeIDE for STM32, MCUXpresso for NXP and generic version for ESP32. I use command line compiling/flashing for ESP32 (idf.py) as I have not yet been able to compile/download from Eclipse. For Cypress PSoc 5 I use PSoC Creator 4.2. Compared to these the Arduino IDE is primitive.

---

I need to look at how to organize the grblHAL repository - I am not happy with the way it is currently done. My subversion master is a lot cleaner as I have a separate project for each processor and use [externals](http://svnbook.red-bean.com/en/1.7/svn.advanced.externals.html) for the grbl core and plugins. Easy to check out and build - and no code duplication. It would be nice to have the github repository behave in the same way, select a processor and check out a complete project in one go...

terjeio, 2020-06-19:
added #if/#endif guard in my subversion master. Will commit to github later as this is only a warning.

closed 2020-06-19 by terjeio