terjeio/grblHAL #62: how to burn the microprocessor

MARIOBASZ, 2020-06-14:
Hi @terjeio, sometimes I am quite stupid ... can you tell me how to record the program? I can't find how.
I want to use Arduino Due or Esp32.
Huge job, very grateful.

closed 2020-06-15 by MARIOBASZ

terjeio, 2020-06-16:
> I am in the final stage of my ATC

Interesting, I am also tinkering with a ATC - making slow progress. grblHAL has entry points that can be used for adding code to control the ATC from T and M6 commands:

https://github.com/terjeio/grblHAL/blob/c9c9d2bfdd510d2651b252a7d0a6fda6ce015fff/grbl/hal.h#L179-L180

I need to come up with a [better organization](https://github.com/terjeio/grblHAL/issues/60#issuecomment-645019443) of the grblHAL project, or at least improve the [documentation](https://github.com/terjeio/grblHAL/wiki/Compiling-GrblHAL).

I hope you were able to program the processor of your choice since you already closed this issue.

MARIOBASZ, 2020-06-16:
hi @terjeio, thanks for caring.
Yes, I was able to load the firmware. Even out of the machine. The reference to documentation comes in handy! I had not found it.
If it does not bother you, surely I will ask you some questions later, as I said I am also limited in programming, even so I have managed to make some contribution to bCNC (only some plugins, halfway a complement to mill with two tools different diameter, useful in pcb) and some M code in grbl; Your work will surely be very useful to me in my project (one among others) of cnc with some different characteristics.
I do not know if I read correctly: Movement control in S? control jerks? I use a trochoidal strategy, this could do me great! (and investigate G5).
Hug.