terjeio/grblHAL #63: GrblHAL always goes into alarm state on MSP432.

ArghyaDas3301, 2020-06-16:
I'm using an MSP432P401R LaunchPad to run GrblHAL without anything connected to its GPIO pins. (No CNC_Boosterpack nor any limit switches etc.). And I'm using Universal G-code Sender to communicate with the MSP432.
1. I have come across a few problems which I'd really like some help with. 
The code compiles fine and I'm able to upload it to the MCU when I set **#define CNC_BOOSTERPACK       to 0** instead of 1 in driver.h. Because the problem seems to arise from the eeprom files when Boosterpack is enabled.

2. After that I've observed that I also had to set **#define ESTOP_ENABLE to  0**  instead of 1 in driver.h or UGS starts throwing an Alarm code 10 at me.

3. I've also had to set **#define COMPATIBILITY_LEVEL to 1** instead of 0 in config.h otherwise UGS doesn't even connect with the MSP432 and throws a "Grbl has not finished booting." when I try to send any commands.

4.after making the above changes the MSP is able to talk to UGS but is still always in Active Alarm mode and so G codes do not work but the '$' sign parameters can still be changed.
 This is where I'm stuck at right now. I also tried using GrblPanel and Openbuilds Controller and its the same story. It's always in a state of alarm and is locked out.

[image]

5.I've also noticed something strange about the $10 parameter. It shows a 510 by default which is odd because I believe in the documentation for grblHAL it says the default is supposed to be 1 . Here is the status log that  I get rn after making the previously mentioned changes.

**** Connected to COM7 @ 115200 baud ****
Grbl 1.1f ['$' for help]
>>> $$
[MSG:'$H'|'$X' to unlock]
$0 = 10    (Step pulse time, microseconds)
$1 = 25    (Step idle delay, milliseconds)
$2 = 0    (Step pulse invert, mask)
$3 = 0    (Step direction invert, mask)
$4 = 0    (Invert step enable pin, boolean)
$5 = 0    (Invert limit pins, boolean)
$6 = 0    (Invert probe pin, boolean)
$10 = 510    (Status report options, mask)
$11 = 0.010    (Junction deviation, millimeters)
$12 = 0.002    (Arc tolerance, millimeters)
$13 = 0    (Report in inches, boolean)
$14 = 0   
$15 = 0   
$16 = 0   
$17 = 0   
$18 = 0   
$19 = 0   
$20 = 0    (Soft limits enable, boolean)
$21 = 0    (Hard limits enable, boolean)
$22 = 4    (Homing cycle enable, boolean)
$23 = 0    (Homing direction invert, mask)
$24 = 25.000    (Homing locate feed rate, mm/min)
$25 = 500.000    (Homing search seek rate, mm/min)
$26 = 250    (Homing switch debounce delay, milliseconds)
$27 = 1.000    (Homing switch pull-off distance, millimeters)
$28 = 0.100   
$29 = 0   
$30 = 1000.    (Maximum spindle speed, RPM)
$31 = 0.    (Minimum spindle speed, RPM)
$32 = 0    (Laser-mode enable, boolean)
$33 = 5000.000   
$34 = 0.000   
$35 = 0.000   
$36 = 100.000   
$37 = 0   
$38 = 0   
$39 = 1   
$40 = 0   
$41 = 0   
$42 = 2   
$43 = 1   
$44 = 0   
$45 = 0   
$46 = 0   
$56 = 5.000   
$57 = 100.000   
$58 = -5.000   
$59 = 500.000   
$60 = 0   
$61 = 0   
$62 = 0   
$63 = 3   
$64 = 0   
$65 = 0   
$90 = 0.000   
$91 = 0.000   
$92 = 0.000   
$95 = 0.000   
$100 = 250.000    (X-axis travel resolution, step/mm)
$101 = 250.000    (Y-axis travel resolution, step/mm)
$102 = 250.000    (Z-axis travel resolution, step/mm)
$110 = 500.000    (X-axis maximum rate, mm/min)
$111 = 500.000    (Y-axis maximum rate, mm/min)
$112 = 500.000    (Z-axis maximum rate, mm/min)
$120 = 10.000    (X-axis acceleration, mm/sec^2)
$121 = 10.000    (Y-axis acceleration, mm/sec^2)
$122 = 10.000    (Z-axis acceleration, mm/sec^2)
$130 = 200.000    (X-axis maximum travel, millimeters)
$131 = 200.000    (Y-axis maximum travel, millimeters)
$132 = 200.000    (Z-axis maximum travel, millimeters)
ok
>>> $G
[GC:G0 G54 G17 G21 G90 G94 G49 G98 G50 M5 M9 T0 F0 S0.]
ok
>>> $X
[Error] An error was detected while sending '$X': (error:18) An unknown error has occurred. Streaming has been paused.
>>> G00 X10
[Error] An error was detected while sending 'G00 X10': (error:9) G-code commands are locked out during alarm or jog state. Streaming has been paused.

I have tried a bunch of things but can't seem to get out of the alarm state. 
Any help will be greatly appreciated.

terjeio, 2020-06-17:
1. My bad, default value for CNC_BOOSTERPACK should be 0.

2. This should perhaps be set to 0 when  CNC_BOOSTERPACK is 0...

3. Some senders don't like the grblHAL extensions - UGS is one of them. You may try [my sender](https://github.com/terjeio/Grbl-GCode-Sender) if you are using Windows.

4. grblHAL defaults to [normally closed (NC) switches](https://github.com/terjeio/grblHAL/wiki/Changes-from-grbl-1.1) and if the Reset input is asserted (not shorted to ground) error 18 is issued for safety reasons. Set $14=15 to invert most control pins or better connect the Reset pin to ground (P2.4). The default inversion can also be set by uncommenting (and possibly changing) the `INVERT_CONTROL_PIN_MASK` in [grbl\defaults.h](https://github.com/terjeio/grblHAL/blob/master/grbl/defaults.h).

5. Default value for $10 is dependent on the compatibility mode. grblHAL has a lot of [additional $-settings](https://github.com/terjeio/grblHAL/wiki/Additional-or-extended-settings) options over vanilla grbl.

Note that I have not added saving of settings to flash for this driver so settings will be reverted to default on a cold start or hard reset if an EEPROM is not present. You can edit [grbl/defaults.h](https://github.com/terjeio/grblHAL/blob/master/grbl/defaults.h) to match your machine in order to avoid issuing a bunch of $-commands on startup.

ArghyaDas3301, 2020-06-17:
Thank you so much! That was the issue indeed. I ended up uncommenting the invert option for reset and safety door bits and it solved the issue as well. 

I have tried to use the Gcode sender you developed but Vector3 library is missing on Nuget and I'm unable to build the project without it. I'm not sure how to import the library manually by downloading it from GitHub and adding it on Visual Studio. I'm very new to using Visual Studio.  I would love to use the Gcode sender you developed if I could get some help with successfully building/ running it. 

Lastly, I have one more question. I can see that GrblHAL has support for G96 command for constant surface speed and  MSP432 seems to be the only driver that supports CSS right now which is why I choose to use it. However, I've been trying to find the code snippet in the source files to see how the Spindle rpm is varied to maintain Constant surface speed but I couldn't find it. Could you direct me to the file where I can find the implementation for CSS? Also, Do I need any extra hardware like an encoder on my spindle and enable the Spindle PID to properly implement CSS?

I need G96 for facing operations using a lathe that I'm building. It needs to perform roughing & finishing operations on the front face of a large circular disk.

terjeio, 2020-06-17:
>... I could get some help with successfully building/ running it.

No need to build it yourself unless you want to modify it, binaries are available [here](https://github.com/terjeio/Grbl-GCode-Sender/releases).

> I can see that GrblHAL has support for G96 command for constant surface speed and MSP432 seems to be the only driver that supports CSS ...

CSS is not dependent on encoder input or spindle PID so G96 should work with all drivers. Set $32=2 to enable lathe mode for G96 to become available. You may want to add spindle linearization if actual speed deviates too much from the programmed speed.

> Could you direct me to the file where I can find the implementation for CSS?

https://github.com/terjeio/grblHAL/blob/c9c9d2bfdd510d2651b252a7d0a6fda6ce015fff/grbl/stepper.c#L868-L871

Search for `is_rpm_pos_adjusted` and `SpindleSpeedMode_CSS` in the source tree to find more.

ArghyaDas3301, 2020-06-17:
Thank you once again! This was extremely helpful. Really appreciate the quick response.

closed 2020-06-17 by ArghyaDas3301