terjeio/grblHAL #64: oddness with cycle/start button.

phil-barrett, 2020-06-17:
The cycle start button is not working  on my Teensy 4.1 system. 
- TSender sees the button press (S "led" goes off during press)
- I can pause a running program with the Feed Hold button but Cycle Start button has no effect
- I can resume program via TSender's Cycle Start button.
- If I press the Cycle Start button (the hw one), it puts the machine in Hold:0 state.

grblHAL software is current. I've enabled encoder input and set up 5 Axes.  I don't think those matter as I saw this on a version I build from the distribution with no changes.

$? and $I info
<Idle|WPos:205.528,97.524,3.000,0.000,0.000|Bf:35,1022|FS:0,0|Pn:XYZABHS>
ok
[VER:1.1f(IMXRT1062).20200603:]
[OPT:VNMSL,35,1024,5,0]
[NEWOPT:ES,TC]
[DRIVER VERSION:200606]
[DRIVER OPTIONS:USB.2]
ok

terjeio, 2020-06-17:
> TSender sees the button press (S "led" goes off during press)

The "led" should go on during press, also if the H "led" is on when pressed then the cycle start input has no effect. Invert your inputs with $14.

> If I press the Cycle Start button (the hw one) while running a program, it puts the machine in Hold:0 state.

As expected when the inversion setting is wrong. All "leds" should be off during normal operation.

phil-barrett, 2020-06-17:
ah. I keep forgetting that.

closed 2020-06-19 by terjeio