terjeio/grblHAL #65: Mask for limits

MARIOBASZ, 2020-06-17:
Trying to invert pin logic for limits with mask, I was assuming you could invert for all six axes, however it only works for X and Z.
$ 5 = 7 (111) reverses Z Y X
I assumed that 63 (111111) reversed the limits of the six axes
however, I cannot invest S R H
Is there a way to invest them?

phil-barrett, 2020-06-18:
Do you have 6 axes set up?You need to enable additional axes in grbl/config.h.  By default, there are only 3.  S, R and H are inputs for Cycle/Start, Reset and Feed/Hold. $14 inverts the control pins. Reset is 1, Hold is 2 and Start 4 so you would enter $14 = 7.

MARIOBASZ, 2020-06-18:
Thank you very much, phil-barret.
I will try as you told me.

closed 2020-06-18 by MARIOBASZ