terjeio/grblHAL #66: Gui does not finish loading

MARIOBASZ, 2020-06-18:
I have followed the indications (#65), as my due is not connected I have established $ 5 = 7 $ 14 = 6
If A0 connected to GND, bCNC warns "Not connected", however the axes move if I send G0 X10 (checked when sending "?", It returns change in M pos).
If I disconnect and reconnect with A0 open, it reports "error 18 [R]" and of course there is no movement.
I have tried with UGS plataform, to read that "error not connected" in bCNC is repeated, however there it warns "grbl has not finished booting"
Some help? Any ideas?
Note: if 14 = 7, Arduino Due continually reset (at least it seemed to me that problem from my little knowledge)
Will your sender work?
I have downloaded the directory, I do not know in which file to find to start the soft.

phil-barrett, 2020-06-18:
I suggest using Terje's sender ("TSender" https://github.com/terjeio/Grbl-GCode-Sender - use alpha 19) over older grbl G Code Senders. One thing it does is make it really easy to set all the various grbl configuration bits. Instead of "$14=6", you check boxes for the various bits. 
[image]

terjeio, 2020-06-18:
Some senders requires [COMPATIBILITY_LEVEL](https://github.com/terjeio/grblHAL/wiki/Changes-from-grbl-1.1#workaround) be set > 0 as they do not handle the grblHAL extensions well, my sender should work with all levels. My sender is started by running/double clicking `GCode Sender.exe`.

MARIOBASZ, 2020-06-18:
I have not found the .exe file, in which folder?
Will you believe that I repeated kindergarten?

phil-barrett, 2020-06-18:
OK, you did the "clone or download" link? What you want is the releases link which is above the green bar. That takes you to a page where you will see "Major refactoring of probing tab++". To the left of that is "v0.19 Alpha".   You are not the first person to be confused by this.

MARIOBASZ, 2020-06-18:
I'm in the right place, Grbl-gcode-Sender-master, from here to which file? (from CNC AppLaunch to readme, I assumed in Gcode Sender), I am two steps from the .exe file but the gps doesn't work for me

phil-barrett, 2020-06-18:
I'm sorry!  My bad.  Go to the releases page.  At the bottom of the "v0.19 Alpha"  section you will see "Assets 3". Click on that and you will see  GCode.Sender.Alpha19.zip

Here's the link [https://github.com/terjeio/Grbl-GCode-Sender/releases/download/v0.19-alpha/GCode.Sender.Alpha19.zip](url)

MARIOBASZ, 2020-06-18:
[image]

maybe I don't explain myself well, the translator translates it into angels
This folder Ok.
What path should I follow?

phil-barrett, 2020-06-18:
You downloaded the wrong zip file, the sender binary is not on your computer.  Go to terjeio/Grbl-GCode-Sender repository.  On that page click on "20 Releases".  On that page, find Assets 3 and click on that.  You will see a link to the zip file (GCode.Sender.Alpha19.zip). Copy the directory out of the zip file and look for "GCoder Sender.exe".

MARIOBASZ, 2020-06-18:
Found what I was looking for in the middle of the forest, behind a tree, under a stone at midnight.
Now it works.
I will try to collaborate.
Thank you.

phil-barrett, 2020-06-18:
divertidísimo!  glad you found it.

closed 2020-06-18 by MARIOBASZ