terjeio/grblHAL #71: Ethernet questions for iMXRT 1062 driver

phil-barrett, 2020-07-12:
I need to arrange a 1 mS timer to call enet_proc_input and enet_poll(). I looked at using driver_delay_ms but that seems like more overhead than necessary. Is there a better way to do that.  

From the lwip_websrc we have this in loop(){}:
```
void loop()
{
  static uint32_t last_ms;
  uint32_t ms;

  enet_proc_input();

  ms = millis();
  if (ms - last_ms > 100)
  {
    last_ms = ms;
    enet_poll();
  }
}
```

And this is in the T41.c wrapper for lwip.
```
void enet_proc_input(void)
{
    struct pbuf *p;

    if (!rx_callback)
    {
        if (rx_ready)
            rx_ready = 0;
        else
            return;
    }
    while ((p = enet_rx_next()) != NULL)
    {
        enet_input(p);
    }
}
void enet_poll()
{
    sys_check_timeouts();
    check_link_status();
}

```

terjeio, 2020-07-13:
There is, it should be added to the systick_isr.

Yesterday I added the required code to [driver.c](https://github.com/terjeio/grblHAL/tree/test/drivers/IMXRT1062/main/driver.c) (in the test branch), have you done the same?

My intention was to next add the needed glue code and settings handling in enet.c/enet.h in a similar way as the [MSP432E401Y](https://github.com/terjeio/grblHAL/tree/test/drivers/MSP432E401Y/ethernet) driver. Have you done this part already as well? If not do you want to give it a try?

phil-barrett, 2020-07-13:
No, I haven't added anything  yet. My approach is (was?) to simply get the lwip web sever running in the grblHAL context to make sure it would work.  Then to figure out basically what you've done. There is a large learning curve for me here. No surprise! 

I have been looking at the MSP432E401Y driver and started to get a glimmer of how it works.  I could probably figure it out but you could get it done 10X faster.  I would be happy to do testing and other tasks to help.

Phil

terjeio, 2020-07-13:
Ok then I'll do it, if I get the code to compile I can commit to the test branch for you to test.

terjeio, 2020-07-13:
Committed code for testing to the [test branch](https://github.com/terjeio/grblHAL/tree/test). Arduino does not let me compile for/flash to the T4.0 board so no testing whatsoever done on my part. Code compiles but does it run?

```
Using library teensy41_ethernet-master at version 2.0.2 in folder: C:\Users\terjeio\Documents\Arduino\libraries\teensy41_ethernet-master 
"C:\\Program Files (x86)\\Arduino\\hardware\\teensy/../tools/arm/bin/arm-none-eabi-size" -A "C:\\Users\\terjeio\\AppData\\Local\\Temp\\arduino_build_2786/main.ino.elf"
Sketch uses 121280 bytes (1%) of program storage space. Maximum is 8126464 bytes.
Global variables use 230068 bytes (43%) of dynamic memory, leaving 294220 bytes for local variables. Maximum is 524288 bytes.
```

phil-barrett, 2020-07-13:
It compiled cleanly.  will start testing.

You probably need to [upgrade teensyduino](https://www.pjrc.com/teensy/td_download.html) to the latest to get Teensy 4.1 support.  And add the lwip library from [paul's repository.](https://github.com/PaulStoffregen/teensy41_ethernet). Ignore all the apps - they are broken.

phil-barrett, 2020-07-13:
Hmmm, something is not right.  Teensy loader won't automatically load it.  Pressing the pm button on the T4.1 causes it to load the hex file but it seems to not want talk via USB.   Serial monitor connects but there is no response.  The MagJack LED is flickering so I think that means lwip is running.

It loaded ok the first time I tried to upload so it is probably the grblHAL that is breaking the ability to auto-upload. Should I try building for serial instead of USB and try that? May take a while as I have some wife-happiness hours coming due...

> Using precompiled core: C:\Users\philba\AppData\Local\Temp\arduino_cache_538183\core\core_823930f94f2ab1f89be54b87fba2d817.a
Linking everything together...
"C:\\Program Files (x86)\\Arduino\\hardware\\teensy/../tools/arm/bin/arm-none-eabi-gcc" -O2 -Wl,--gc-sections,--relax "-TC:\\Program Files (x86)\\Arduino\\hardware\\teensy\\avr\\cores\\teensy4/imxrt1062_t41.ld" -mthumb -mcpu=cortex-m7 -mfloat-abi=hard -mfpu=fpv5-d16 -o "C:\\Users\\philba\\AppData\\Local\\Temp\\arduino_build_356331/main.ino.elf" "C:\\Users\\philba\\AppData\\Local\\Temp\\arduino_build_356331\\sketch\\driver.c.o" "C:\\Users\\philba\\AppData\\Local\\Temp\\arduino_build_356331\\sketch\\enet.c.o" "C:\\Users\\philba\\AppData\\Local\\Temp\\arduino_build_356331\\sketch\\i2c.c.o" "C:\\Users\\philba\\AppData\\Local\\Temp\\arduino_build_356331\\sketch\\uart.c.o" "C:\\Users\\philba\\AppData\\Local\\Temp\\arduino_build_356331\\sketch\\usb_serial_pjrc.c.o" "C:\\Users\\philba\\AppData\\Local\\Temp\\arduino_build_356331\\sketch\\main.ino.cpp.o" "C:\\Users\\philba\\AppData\\Local\\Temp\\arduino_build_356331\\sketch\\usb_serial_ard.cpp.o" "C:\\Users\\philba\\AppData\\Local\\Temp\\arduino_build_356331\\sketch\\src\\eeprom\\eeprom_24AAxxx.c.o" "C:\\Users\\philba\\AppData\\Local\\Temp\\arduino_build_356331\\sketch\\src\\eeprom\\eeprom_24LC16B.c.o" "C:\\Users\\philba\\AppData\\Local\\Temp\\arduino_build_356331\\sketch\\src\\grbl\\coolant_control.c.o" "C:\\Users\\philba\\AppData\\Local\\Temp\\arduino_build_356331\\sketch\\src\\grbl\\corexy.c.o" "C:\\Users\\philba\\AppData\\Local\\Temp\\arduino_build_356331\\sketch\\src\\grbl\\eeprom_emulate.c.o" "C:\\Users\\philba\\AppData\\Local\\Temp\\arduino_build_356331\\sketch\\src\\grbl\\gcode.c.o" "C:\\Users\\philba\\AppData\\Local\\Temp\\arduino_build_356331\\sketch\\src\\grbl\\grbllib.c.o" "C:\\Users\\philba\\AppData\\Local\\Temp\\arduino_build_356331\\sketch\\src\\grbl\\limits.c.o" "C:\\Users\\philba\\AppData\\Local\\Temp\\arduino_build_356331\\sketch\\src\\grbl\\maslow.c.o" "C:\\Users\\philba\\AppData\\Local\\Temp\\arduino_build_356331\\sketch\\src\\grbl\\motion_control.c.o" "C:\\Users\\philba\\AppData\\Local\\Temp\\arduino_build_356331\\sketch\\src\\grbl\\nuts_bolts.c.o" "C:\\Users\\philba\\AppData\\Local\\Temp\\arduino_build_356331\\sketch\\src\\grbl\\override.c.o" "C:\\Users\\philba\\AppData\\Local\\Temp\\arduino_build_356331\\sketch\\src\\grbl\\planner.c.o" "C:\\Users\\philba\\AppData\\Local\\Temp\\arduino_build_356331\\sketch\\src\\grbl\\protocol.c.o" "C:\\Users\\philba\\AppData\\Local\\Temp\\arduino_build_356331\\sketch\\src\\grbl\\report.c.o" "C:\\Users\\philba\\AppData\\Local\\Temp\\arduino_build_356331\\sketch\\src\\grbl\\settings.c.o" "C:\\Users\\philba\\AppData\\Local\\Temp\\arduino_build_356331\\sketch\\src\\grbl\\sleep.c.o" "C:\\Users\\philba\\AppData\\Local\\Temp\\arduino_build_356331\\sketch\\src\\grbl\\spindle_control.c.o" "C:\\Users\\philba\\AppData\\Local\\Temp\\arduino_build_356331\\sketch\\src\\grbl\\state_machine.c.o" "C:\\Users\\philba\\AppData\\Local\\Temp\\arduino_build_356331\\sketch\\src\\grbl\\stepper.c.o" "C:\\Users\\philba\\AppData\\Local\\Temp\\arduino_build_356331\\sketch\\src\\grbl\\system.c.o" "C:\\Users\\philba\\AppData\\Local\\Temp\\arduino_build_356331\\sketch\\src\\grbl\\wall_plotter.c.o" "C:\\Users\\philba\\AppData\\Local\\Temp\\arduino_build_356331\\sketch\\src\\networking\\TCPStream.c.o" "C:\\Users\\philba\\AppData\\Local\\Temp\\arduino_build_356331\\sketch\\src\\networking\\WsStream.c.o" "C:\\Users\\philba\\AppData\\Local\\Temp\\arduino_build_356331\\sketch\\src\\networking\\base64.c.o" "C:\\Users\\philba\\AppData\\Local\\Temp\\arduino_build_356331\\sketch\\src\\networking\\multipartparser.c.o" "C:\\Users\\philba\\AppData\\Local\\Temp\\arduino_build_356331\\sketch\\src\\networking\\sha1.c.o" "C:\\Users\\philba\\AppData\\Local\\Temp\\arduino_build_356331\\sketch\\src\\networking\\strutils.c.o" "C:\\Users\\philba\\AppData\\Local\\Temp\\arduino_build_356331\\sketch\\src\\networking\\urldecode.c.o" "C:\\Users\\philba\\AppData\\Local\\Temp\\arduino_build_356331\\sketch\\src\\networking\\utils.c.o" "C:\\Users\\philba\\AppData\\Local\\Temp\\arduino_build_356331\\libraries\\lwip\\lwip_k6x.c.o" "C:\\Users\\philba\\AppData\\Local\\Temp\\arduino_build_356331\\libraries\\lwip\\lwip_t41.c.o" "C:\\Users\\philba\\AppData\\Local\\Temp\\arduino_build_356331\\libraries\\lwip\\api\\api_lib.c.o" "C:\\Users\\philba\\AppData\\Local\\Temp\\arduino_build_356331\\libraries\\lwip\\api\\api_msg.c.o" "C:\\Users\\philba\\AppData\\Local\\Temp\\arduino_build_356331\\libraries\\lwip\\api\\err.c.o" "C:\\Users\\philba\\AppData\\Local\\Temp\\arduino_build_356331\\libraries\\lwip\\api\\netbuf.c.o" "C:\\Users\\philba\\AppData\\Local\\Temp\\arduino_build_356331\\libraries\\lwip\\api\\netdb.c.o" "C:\\Users\\philba\\AppData\\Local\\Temp\\arduino_build_356331\\libraries\\lwip\\api\\netifapi.c.o" "C:\\Users\\philba\\AppData\\Local\\Temp\\arduino_build_356331\\lib [... cut ...]
C:\Program Files (x86)\Arduino\hardware\teensy/../tools/teensy_post_compile -file=main.ino -path=C:\Users\philba\AppData\Local\Temp\arduino_build_356331 -tools=C:\Program Files (x86)\Arduino\hardware\teensy/../tools -board=TEENSY41 -reboot -port=usb:0/140000/0/8 -portlabel=(null) -portprotocol=(null) 
An error occurred while uploading the sketch
Unable to open COM23 for reboot request
  Windows Error Info: Access is denied.
  more ideas... https://forum.pjrc.com/threads/40632?p=126667&viewfull=1#post126667
Teensy did not respond to a USB-based request to enter program mode.
Please press the PROGRAM MODE BUTTON on your Teensy to upload your sketch.

phil-barrett, 2020-07-13:
OK, further test.  I set 
```
#define ETHERNET_ENABLE    0 // Ethernet streaming.
#define TELNET_ENABLE    0 // Ethernet streaming.
```
At it came up.  

It also boots up if:
```
#define ETHERNET_ENABLE    1 // Ethernet streaming.
#define TELNET_ENABLE    0 // Ethernet streaming.
```
So, the issue is somewhere in telnet land.

phil-barrett, 2020-07-13:
Hmmm, TSender doesn't see it (connects but Settings:Grbl is greyed out). Serial terminal shows:

> [VER:1.1f(IMXRT1062).20200710:]
> [OPT:VNMZHT0SLW,35,1024,3,0]
> [NEWOPT:ES,ETH,TC]
> [DRIVER VERSION:200710]
> [DRIVER OPTIONS:USB.2 HUANYANG0]
> [IP:0.0.0.0]

And the response to $0 is error:3.

terjeio, 2020-07-13:
> You probably need to upgrade teensyduino to the latest to get Teensy 4.1 support.

Done but not sure if it is the latest, probably not. It does not matter because the programming app refuses to upload it.

> Hmmm, something is not right. Teensy loader won't automatically load it.

Could be that the processor is stuck in a panic handler? I've had this problem a couple of times earlier.

Ok, just saw your new comment. I am not sure about the timer callbacks working as they should so it could be due to that.
I have cleaned up the TM4C129 ethernet code a bit,and I have been running tests with telnet for the last few hours so the telnet code should be ok. It is with an older lwIP library though.

If you issue a $I do you get an IP address reported? 

```
[VER:1.1f(TM4C1294NCPDT FreeRTOS).20200710:]
[OPT:VNMSL,35,1024,3,0]
[NEWOPT:SD,ETH]
[DRIVER VERSION:200713]
[BOARD:CNC BoosterPack]
[IP:10.0.0.75]
```

What is your $301 setting? I have it set to 1, that is getting the IP address via DHCP.

In your latest comment:

Ip is not set...

terjeio, 2020-07-13:
> And the response to $0 is error:3.

It should be. What is the response to $$ (list all settings)?

phil-barrett, 2020-07-13:
Here is the results of $$.
> $0=0.0
> 
> $1=0
> 
> $2=10
> 
> $3=215
> 
> $4=35
> 
> $5=1
> 
> $6=0
> 
> $10=0
> 
> $11=36000.000
> 
> $12=36000.000
> 
> $13=0
> 
> $14=49992
> 
> $15=72
> 
> $16=128
> 
> $17=0
> 
> $18=128
> 
> $19=0
> 
> $20=1
> 
> $21=0
> 
> $22=26
> 
> $23=0
> 
> $24=0.000
> 
> $25=0.000
> 
> $26=0
> 
> $27=0.000
> 
> $28=36000.000
> 
> $29=0.0
> 
> $30=-200.
> 
> $31=-200.
> 
> $32=0
> 
> $33=-200.000
> 
> $34=0.002
> 
> $35=0.100
> 
> $36=0.000
> 
> $37=60
> 
> $39=0
> 
> $40=1
> 
> $41=0
> 
> $42=0
> 
> $43=0
> 
> $44=0
> 
> $45=25
> 
> $46=0
> 
> $56=0.000
> 
> $57=0.000
> 
> $58=500.000
> 
> $59=1.000
> 
> $60=0
> 
> $61=0
> 
> $62=0
> 
> $63=0
> 
> $64=0
> 
> $65=0
> 
> $70=255
> 
> $100=250.000
> 
> $101=250.000
> 
> $102=250.000
> 
> $110=250.000
> 
> $111=250.000
> 
> $112=500.000
> 
> $120=0.139
> 
> $121=0.139
> 
> $122=0.139
> 
> $130=-500.000
> 
> $131=-36000.000
> 
> $132=-36000.000
> 
> $300=GRBL
> 
> $301=1
> 
> ok
> 
>

terjeio, 2020-07-13:
Ok, something basic is wrong then - you should get an IP address from DHCP. How to debug this without a debugger...?

phil-barrett, 2020-07-13:
Note, this is with TELNET set to 0.  

Can we use Serial 1 while using USB?  Debug with printf sucks but ...

terjeio, 2020-07-13:
> Note, this is with TELNET set to 0.

That is ok, telnet should only be started when an IP address has been assigned anyway.

Can we use Serial 1 while using USB? Debug with printf sucks but ...

Yes, but perhaps better blink a LED instead?
Calling serialInit(115200); in driver_init() will enable Serial 1, use serialWriteS("some string") to write.

The most basic stuff to get lwIP running is keep its timer(s) ticking - I have not dug into the code to see how that is done.
In sys/arch/sys_arch.c there is a sys_now() function that reads a systick_millis_count variable. Is this beeing updated somewhere? By systick_isr_org(); in the systick_isr in driver.c? If so I see this only gets called when USB_SERIAL_GRBL == 2...

Does sys_now() get called by lwIP? Perhaps an idea to blink a led or output a character in sys_now() to see if the stack is alive?
If it get called is the systick_millis_count updated?

Do not forget your wife ;-)

terjeio, 2020-07-13:
> In sys/arch/sys_arch.c there is a sys_now() function that reads a systick_millis_count variable. Is this beeing updated somewhere? By systick_isr_org(); in the systick_isr in driver.c?

It is updated by a call to `systick_isr_org()` so this needs to be added to the systick_isr when ethernet is enabled, here is how I handle this now:
```
// Interrupt handler for 1 ms interval timer
static void systick_isr (void)
{
#if MODBUS_ENABLE
    modbus_poll();
#endif

#if ETHERNET_ENABLE
    uint32_t delay_cs = 100;

    systick_isr_org();

    if(!(--delay_cs)) {
        delay_cs = 100;
        grbl_enet_poll();
    }
#endif

#if QEI_ENABLE
    if(qei.debounce && !(--qei.debounce)) {
        if(qei.initial_debounce)
            encoder_debounce();
        else
            encoder_update();
    }
#endif

    if(grbl_delay.ms && !(--grbl_delay.ms)) {
        if(grbl_delay.callback) {
            grbl_delay.callback();
            grbl_delay.callback = NULL;
        }
    }
}
```

A first debug point is to check if `link_status_callback()` is called in enet.c and if `isLinkUp` is set true.

phil-barrett, 2020-07-13:
Garden work made wife happy. So back to enet.

I got your most recent note as I was going to send this.  Will follow that in a second but am sending this anyway.

Hooked up an LA. It looks like sys_now() gets called a few times early on. I set the output to high in driver_init().   and then twiddle it in sys_now().   The first image shows it being set high and then 842.2 uS later it drops. Zooming in on the drop part shows it gets called 7 times and then never again.  
Here's the sys_now() code.
```
#include <stdint.h>
#include <Arduino.h>
extern volatile uint32_t systick_millis_count;

// debug
uint32_t st_ms_count = 0;
// debug

/* Returns the current time in mS. This is needed for the LWIP timers */
uint32_t sys_now(void) {
	// debug
	if(st_ms_count++ & 0x1)digitalWrite(41,1);
	else digitalWrite(41,0);
	// debug

  return (uint32_t) systick_millis_count;
}

```
[image]
[image]

terjeio, 2020-07-13:
> Zooming in on the drop part shows it gets called 7 times and then never again.

Ok, it indicates that lwIP has died? You can still communicate with the controller?

phil-barrett, 2020-07-13:
Yes, it responds to $$.

I put the pin twiddling code in link_status_callback and the pin goes high and stays that way.  Looks like it never gets called.

terjeio, 2020-07-13:
Thinking aloud now, maybe a call to

`dhcp_start(netif_default);`

is required at the end of `grbl_enet_init()` in enet.c?

phil-barrett, 2020-07-13:
tried it.  $70 = 255

terjeio, 2020-07-13:
> tried it. $70 = 255

Not the same? I believe dhcp_start(netif_default); is the client side.

$70 is for services, that is server side protocols - not available for T4.

Do you have Skype?

terjeio, 2020-07-13:
Not the same - I see the ESP32 driver has both a client side and a server side. The server side is started in start_services(), client side in wifi_init(); So I believe it is needed when DHCP is in use, has to do with address lease renewals?

phil-barrett, 2020-07-13:
yes. though haven't used it much at all - mostly zoom these days. took a minute to set up. how do I connect to you?

terjeio, 2020-07-13:
My username is terjeio

phil-barrett, 2020-07-13:
My wife and I share an office and she has a business call at 1PM (now). Can I call at 2PM (23:00 your time)? Is that too late?

terjeio, 2020-07-13:
Not too late.

phil-barrett, 2020-07-13:
OK, will call then.

terjeio, 2020-07-13:
Maybe grbl_enet_poll() should be moved to the foreground, now it executes in an isr context. The supplied examples has enet_poll() in the loop() function.

This since T4 lwIP implementation does not have locking for critical code, I had to add some null macros to replace those in the protocol handlers...

Try with changing `execute_realtime()` in driver.c to this in the ETHERNET_ENABLE segment:

```
#if ETHERNET_ENABLE
    static uint32_t last_ms;
    uint32_t ms;

    enet_proc_input();

    ms = millis();
    if (ms - last_ms > 100)
    {
        last_ms = ms;
        grbl_enet_poll();
    }
#endif 
```
and comment out `grbl_enet_poll()` in the systic_isr handler.

phil-barrett, 2020-07-13:
OK, tried that.  No change - TSender still has Settings:Grbl greyed out. I can see execute_realtime being called every 450 nS.

My wife's call with her client is running late and I have to go somewhere around 3:40 so maybe we should do it tomorrow.   I will be available from 8AM tomorrow which is 17:00 your time.

terjeio, 2020-07-13:
Ok, hopefully we can solve this tomorrow.

phil-barrett, 2020-07-14:
This is a list of things I've tried and various observations.

hooked up serial1 printed out a test message. Here's what I got on the terminal:
> Debug grblHAL
> 
> GrblHAL 1.1f ['$' for help]

netif link is up. I added a check for the cycle start button push in execute_realtime and printed link status. When I push start - it shows the link is up. I took this from lwip_websrv - it's in setup() and loops waiting for netif_is_link_up() to return zero before exiting. I probably should debounce the button (get about 50 messages) but it's just a test.
```
	if(digitalRead(17) == 0) {
		sprintf(dbg_buf,"elink status: %s\n\r",netif_is_link_up(netif_default) ? "up" : "down");
		serialWriteS(dbg_buf);
	}

```
Note that it takes some time (3-5 seconds) for the link to come up - pressing start too soon gets a "down" output. I also noticed that if you never call enet_proc_input (or grbl_enet_poll), the link stays down.

If I run TSender, I get idle messages on the serial port but no idle messages on console with verbose checked. Also 
> Idle|WPos:0.000,0.000,0.000|WCS:G54|A:|Sc:|H:0,0|T:0>

only happens once at the start.

Printed ip addr from netif_default - got 0:0:0:0.  I don't know if that's the right netif to test against, though. Wish I had a packet sniffer... could see if dhcp happened.

Getting tired, you are probably just getting up.

By the way, if teensyduino isn't letting you load, have you pressed the button on the T4.1? That should force loading to happen. I find I need to do that about half the time. Not sure what the problem is, pretty annoying.

terjeio, 2020-07-14:
> Wish I had a packet sniffer... could see if dhcp happened.

Did you add `dhcp_start(netif_default)` at the end of `enet_init()`?

> By the way, if teensyduino isn't letting you load, have you pressed the button on the T4.1?

It is nearly a month since I ordered one, not yet here :(
So no button to press...

Strange that the sender seems to crash the controller. If you connect with a terminal after exiting the sender is it still responsive?

phil-barrett, 2020-07-14:
OK, got my coffee, resembling human...

> Did you add dhcp_start(netif_default) at the end of enet_init()?

You meant grbl_enet_init()?  Yes, 
```
bool grbl_enet_init (network_settings_t *network)
{
    *IPAddress = '\0';

#if NETWORK_IPMODE_STATIC

    if(driver_settings.network.ip_mode == IpMode_Static)
        enet_init((ip_addr_t *)&driver_settings.network.ip, (ip_addr_t *)&driver_settings.network.mask, (ip_addr_t *)&driver_settings.network.gateway);
    else 
#endif
    enet_init(NULL, NULL, NULL);

    netif_set_status_callback(netif_default, netif_status_callback);
    netif_set_link_callback(netif_default, link_status_callback);
    netif_set_up(netif_default);
#if LWIP_NETIF_HOSTNAME
    netif_set_hostname(netif_default, network->hostname);
#endif
    dhcp_start(netif_default);

    return true;
}
```

phil-barrett, 2020-07-14:
Sorry, missed the last question.

> Strange that the sender seems to crash the controller. If you connect with a terminal after exiting the sender is it still responsive?
Yes, I can get output on the serial monitor and, if I connect  the sender again, it spits out idle messages and responds to $i.

terjeio, 2020-07-14:
Ok, Skype soon then. 10 past the hour (17:10 here) is ok for me.

phil-barrett, 2020-07-14:
trying now.

terjeio, 2020-07-14:
Did you add me as a contact? No request showing up here..

phil-barrett, 2020-07-14:
By the way, in looking for info on stack size, it looks like the sp is set to the top of the first 512K bank of RAM (RAM1 in the chat at the bottom of [this page](https://www.pjrc.com/store/teensy40.html)) and expands downward.  Note that all grblHAL code and data fit into this bank.  So the max stack size is the 512K - (code + initialized variables + uninitialized variables). I am guessing that mallocs happen out of a heap that is in uninit vars (aka BSS)

terjeio, 2020-07-14:
I am pretty sure we had the IP stack up and running at one point with telnet enabled and `TCPStreamListen()` and `TCPStreamPoll()` commented out. Could it be that changing (commenting in) `MEMP_NUM_PBUF` and `PBUF_POOL_SIZE` in lwipopts.h is the reason it failed at our last attempt? If so then it could perhaps be due to a memory issue? 
But then maybe not as there should be plenty memory available and IIRC lwIP manages its memory allocation from a preallocated chunk of `MEM_SIZE` (defined in lwipopts.h - and taken from the heap?). `MEM_SIZE` for the [TI MCUs](https://github.com/terjeio/grblHAL/blob/master/drivers/MSP432E401Y/lwipopts.h) are 32K vs 24K for the T4.1.

I checked tsender for the websocket error message - it is due to websocket support not beeing compiled in in the release version. Will fix that in the next release.

terjeio, 2020-08-07:
Got my T4.1 today and are now able to connect via both telnet and websocket protocols. However, there are some issues to be resolved:

Since the IP stack is run in the foreground, there is no RTOS in the Arduino framework, larger stream transmit buffers needs to be assigned. The $$ output is getting large, > 512 bytes that is the current size, and this message hangs grblHAL. I'll like to be able to override the default size for networking only - should be fairly easy.

Then there is output stream duplication, e.g. the real time messages are sent to the serial output when a network client is connected. This is intended for MPG/DRO pendants and what not. When the serial output is sent out via USB and there is no client connected grblHAL hangs after a short while, I assume when the USB transmit buffer is full. Either this duplication has to be disabled or a reliable way to detect USB client connect/disconnect has to be found. UART output should be ok - but I have not yet tested this.

@phil-barrett Are you able to do help with testing by downloading from my subversion repository? I have made large changes to how configuration is done, this for a global config file and/or for symbols defined in the IDE. I am not yet ready to commit this to github...

phil-barrett, 2020-08-07:
> @phil-barrett Are you able to do help with testing by downloading from my subversion repository? I have made large changes to how configuration is done, this for a global config file and/or for symbols defined in the IDE. I am not yet ready to commit this to github...

Yes, of course.  I will make some time today.

phil-barrett, 2020-08-07:
Setting it up now (see question via email).  

Are the buffers coming from RAM2? (i.e. via malloc?)  There is a barely used 512K of memory there.

phil-barrett, 2020-08-07:
Some related thoughts on USB.  The Teensy 4.1 has a second USB channel available in Host mode. I want to look into using it to interface to USB based MPGs.  There are several kinds for sale that are less than $100 USD - relatively inexpensive as CNC goes. I think availability of off-the-shelf solutions would increase the desirability of grblHAL. I assume a wrapper is needed for this?

A side question, should this be a separate issue?

terjeio, 2020-08-07:
> Are the buffers coming from RAM2? (i.e. via malloc?) There is a barely used 512K of memory there.

They are statically allocated, from structs defined in stream.h. Since this is common for all drivers I do not want to increase the size there. Could have done it by setting a symbol in the IDE project properties. But the Arduino IDE does not even remember which processor/board a project is for...

> A side question, should this be a separate issue?

I believe so, it has nothing to do with ethernet? What kind of USB devices are they? Presenting themself as a keyboard/mouse (HID) or a serial port?

phil-barrett, 2020-08-07:
Ok, got the source, built and loaded.  I can see the teensy on my network 192.168.86.104 - can ping it ok.  Just running TSender causes it to sit there with "Waiting for Controller (COM23:115200,N,8,1)...

Starting up TSender in terminal window with 
> "GCode Sender.exe" -port 192.168.86.104:23 

It runs but doesn't appear to connect (no DRO values showing) and it exits after a little poking around (timeout?).

also tried

> "GCode Sender.exe" -port ws://192.168.86.104:80

and it didn't even run.

This is using the beta GCode Sender, if that matters.

I also tried using putty to connect via telnet.  Not sure if the teensy was listening (I got echo of what I typed, probably local echo) but then putty exits after about 30 seconds (probably timeout).

phil-barrett, 2020-08-07:
Interesting. I noticed that GRBL had disconnected from the network.  Power cycled it. Tried websockets:
> "GCode Sender.exe" -port ws://192.168.86.104:80

This brought up TSender with the DRO showing numbers but the Settings: Grbl Tab is blank and jog doesn't do anything.  GRBL remains on the network though.

Closing TSender and restarting it using Telnet seems to cause it to drop off the network - ping shows host unreachable.

phil-barrett, 2020-08-07:
OK, strange behavior going on.  I power cycled  the teensy and restarted TSender using telnet.  Now I get the DRO showing numbers and jogging works.

terjeio, 2020-08-07:
Seems that I have fixed the buffer size issue by moving some #includes around and overriding the default size. Output stream duplication is deactivated for USB streams and reenabled on network disconnect. I have also added some missing includes since changing the way configuration is done adds less includes per default. These changes are now committed to the subversion repository.

One (final?) issue is how often to poll the IP stack, currently it is done every 100 ms - it should be reduced if keyboard jogging is erratic? It is handled in `grbl_enet_poll()` in enet.c.

phil-barrett, 2020-08-07:
Updated, built and uploaded it.  It is working with telnet now. Running a big test GCode job now.  will test out websocket mode next.  Which mode do you recommend people use?

phil-barrett, 2020-08-08:
I was able to get multi-hour runs in for both websocket and telnet interface. Seems to be pretty solid now.

terjeio, 2020-08-08:
I am 6 hours+ into a websocket job. Telnet is the preferred protocol because there is no protocol overhead. Websocket is aimed at web-based senders.

terjeio, 2020-08-08:
Next is to add SD-card support, a http daemon and then [ESP-3D-webui](https://github.com/luc-github/ESP3D-webui)? SD-card support is a given, not sure about the rest. BTW I added a 2K FRAM to your board, it worked out-of-the-box with the EEPROM plugin.

einencool, 2020-08-08:
Although I didn't use the ethernet adapter and the option is commented out, I always get an error message while compiling.

In which folder do I have to put the "lwipopts.h" file?

C:\Users\xxxx\AppData\Local\Temp\arduino_build_739742\sketch\enet.c:27:22: fatal error: lwipopts.h: No such file or directory

compilation terminated.

Fehler beim Kompilieren für das Board Teensy 4.1.

terjeio, 2020-08-08:
@einencool  Can you try with deleting enet.c - it is not needed when ethernet is not in use. I will fix this later.

einencool, 2020-08-08:
Hello @terjeio 
Sadly then the next Message occurs.
But no rush, its no Problem, I'm working with the Alarm Connection and have to tram the Spindle, so I really didn't need it essentially today :-)

In file included from C:\Users\csowa\AppData\Local\Temp\arduino_build_671108\sketch\src\networking\TCPStream.c:45:0:

C:\Users\xxxx\AppData\Local\Temp\arduino_build_671108\sketch\src\networking\networking.h:24:22: fatal error: lwip/opt.h: No such file or directory

compilation terminated.

Fehler beim Kompilieren für das Board Teensy 4.1.

terjeio, 2020-08-08:
@einencool  You have added the networking plugin folder to main/src? Delete it as well.

Networking requires a 3rd party lwip library - will add documentation for that later as well. First we need to verify that networking is ok...

einencool, 2020-08-08:
Yes after the first errors I added the folder. 
I deleted it, and this failure seems to be gone, now I have another issue with the homing section, but I think I'll get it to work in the next hours, Thank you for this fast help :-)

phil-barrett, 2020-08-08:
Concentric circles as arcs completed in 7:30 overnight.  It was close to the USB time (less than 10 minutes longer). 

Will spend this AM testing but I think we are good.

phil-barrett, 2020-08-08:
It doesn't look like the connection method (websocket vs telnet) is reported in $i. Perhaps this would be a useful piece of data.

terjeio, 2020-08-08:
I termintated my websocket test at 8h+ Latest subversion commit has improved include handling in order to avoid compilation errors. Also added connection status to $I, could be useful for debugging.

Trying to get SD card working but no luck yet, [this library](https://github.com/WMXZ-EU/uSDFS) works with a test program but not grblHAL for some strange reason. Lack of debugger is a royal pain...

IMO you should add a pin mapping file for your board and perhaps also a matching .c file for the optional GPIO pins available?

[19 more replies]

closed 2020-09-04 by terjeio